ghdl/ghdl #1120: Crash when assigning to aggregate using port with initial value

bradleyharden, 2020-01-29:
What happened to the issue template? It's not showing up for me.

The following reproducer crashes for me with the LLVM backend. My system is Ubuntu 18.04. I'm on master at a4fdd199.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  generic(
    BITS  : positive;
    ZEROS : unsigned(BITS - 1 downto 0) := (others => '0'));
  port(
    min : in  u_unsigned(BITS - 1 downto 0) := ZEROS);
end entity;

architecture rtl of test is

begin

  process
    variable sum : unsigned(BITS - 2 downto 0);
    variable carry : std_ulogic;
  begin
    (carry, sum) := min;
    wait;
  end process;

end architecture;
```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.35-2194-ga4fdd199) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/home/hardebj1/tmp/tmp/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o ghdl_reproducer.o ghdl_reproducer.vhd                                                                       
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:307
Call stack traceback locations:
0x7f8cf47562d1 0x5cd2f8 0x59b344 0x5b16c2 0x5b17d7 0x5b5cd1 0x5bb149 0x643ca5 0x587a81 0x64a2c4 0x64bba0 0x64d499 0x408693 0x7f8cf3801b95 0x407998 0xfffffffffffffffe                                             
******************************************************************
```

eine, 2020-01-29:
> What happened to the issue template? It's not showing up for me.

Maybe you used not the default button?

- https://github.com/ghdl/ghdl/issues/new/choose
- https://github.com/ghdl/ghdl/issues/new?template=bug_report.md
- https://github.com/ghdl/ghdl/issues/new?template=feature_request.md

but https://github.com/ghdl/ghdl/issues/new

bradleyharden, 2020-01-29:
Yes, I think that's what happened.

diogratia, 2020-01-30:
I found this an interesting bug. First the use of the generic BITS in declaring the ZEROS is specifically allowed in -2008 6.5.6 Interface lists, 6.5.6.1 General".... A name that denotes an interface declaration in a generic interface list may appear in an interface declaration within the interface list containing the denoted interface declaration.

This was illegal in earlier revisions of the standard (4.3.2.1 interface lists "A name that denotes an interface object may not appear in any interface declaration within the interface list containing the denoted interface object except to declare this object. ").

Analyzing the entity under an earlier revision sheds light on the problem:

```console
ghdl -a test.vhdl
test.vhdl:8:22:error: generic "bits" is not visible here
test.vhdl:10:15:error: no declaration for "u_unsigned"
test.vhdl:13:14:error: entity 'test' was not analysed
ghdl:error: compilation error
```
using `--std=08` and commenting out the process statement allows test.vhdl to be analyzed elaborated and run despite the type mark  `u_unsigned` for port `min` having no visible declaration.

With the process statement present no semantic check of the type mark in the declaration of `m` has occurred before the  system assertion failure.

(Note the change in 6.5.6.1 is not referenced in Annex E Changes from IEEE Std 1076-2002.)

bradleyharden, 2020-01-30:
Thanks for taking a look. I knew something about it was odd, because it was hard to isolate and build a small reproducer. Changing several different and seemingly disparate parts of the code would eliminate the bug.

One question though, you said

> using --std=08 and commenting out the process statement allows test.vhdl to be analyzed elaborated and run despite the type mark u_unsigned for port min having no visible declaration.

But if you're using `std=08`, then isn't u_unsigned visible in `numeric_std`?

diogratia, 2020-01-30:
So it is:

```vhdl
subtype U_UNSIGNED is UNRESOLVED_UNSIGNED;
```

So where is the problem?

Commenting out the assignment:
```vhdl
--  (carry, sum) := min;
```
allows the example to analyze, elaborate and simulate (while doing nothing).  It's definitely an issue with assignment where the target is an aggregate.

9.3.3.3 Array aggregates  

>**_For an aggregate of a one-dimensional array type, each choice shall specify values of the index type, and the expression of each element association shall be of either the element type or the type of the aggregate._** If the type of the expression of an element association is the type of the aggregate, then either the element association shall be positional or the choice shall be a discrete range.

The type of the aggregate is taken from the right hand side of the assignment. And the association is by matching elements:

>For a positional association with an expression of the element type of the aggregate, the expression specifies one element of the aggregate value. For a positional association with an expression of the type of the aggregate, the expression specifies a number of matching elements (see 9.2.3) of the aggregate value given by the length of the value of the expression.

9.2.3 Relational operators
>...  
>... For two one-dimensional array values, matching elements are those (if any) whose index values match in the following sense: the left bounds of the index ranges are defined to match; if two elements match, the elements immediately to their right are also defined to match. ...

(A fancy way of saying the elements match left to right.) 

9.3.3.1 General

>The type of an aggregate shall be determinable solely from the context in which the aggregate appears, excluding the aggregate itself but using the fact that the type of the aggregate shall be a composite type. The type of an aggregate in turn determines the required type for each of its elements.

12.5 The context of overload resolution

>...Each of the following constructs is a complete context:  
>>— A declaration  
>>— A specification  
>>— A statement  
>>... 
>  
>When considering possible interpretations of a complete context, the only rules considered are the syntax rules, the scope and visibility rules, and the rules of the form as follows:  
>> ...
>>d) The rules that require the type of an aggregate or string literal to be determinable solely from the enclosing complete context. Similarly, the rules that require that the meaning of the prefix of an attribute must be determinable independently of the attribute designator and independently of the fact that it is the prefix of an attribute.  
>> ...

So that tells us the aggregate assignment with both an element of the element type (of u_unsigned) and the type of the aggregate (u_unsigned) is causing ghdl to fail. (sum and min are of subtypes of the same base type).

10.6.2 Simple variable assignments  
10.6.2.1 General  

> ...  
>If the target of the variable assignment statement is in the form of an aggregate, then the type of the aggregate shall be determinable from the context, excluding the aggregate itself but including the fact that the type of the aggregate shall be a composite type. The base type of the expression on the right-hand side shall be the same as the base type of the aggregate. Furthermore, the expression in each element association of the aggregate shall be a locally static name that denotes a variable. This form of variable assignment assigns each subelement or slice of the right-hand side value to the variable named as the corresponding subelement or slice of the aggregate.  
>  
>If the target of a variable assignment statement is in the form of an aggregate, and if the locally static name in an element association of that aggregate denotes a given variable or denotes another variable of which the given variable is a subelement or slice, then the element association is said to _identify_ the given variable as a target of the assignment statement. It is an error if a given variable is identified as a target by more than one element association in such an aggregate. Furthermore, it is an error if an element association in such an aggregate contains an **others** choice, or if the element association contains a choice that is a discrete range and an expression of a type other than the aggregate type.

You could note Tristan fixed the identical issue for simple signal assignment in #676 (the two preceding paragraphs have the same meaning as the corresponding paragraphs in 10.5.2.1.

(The index values for the aggregate  element association are derived from the index type of the base type unresolved_unsigned from the left bound of the index type.)

We can demonstrate the #676 fix is implemented:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  generic(
    BITS  : positive;
    ZEROS : unsigned(BITS - 1 downto 0) := (others => '0'));
[... 70 lines cut ...]
>>— Otherwise, the direction of the index range of the aggregate is that of the index subtype of the base type of the aggregate.  
>  
>The bounds of an array that does not have an others choice are determined as follows. For an aggregate that has named associations, the leftmost and rightmost bounds are determined by the direction of the index range of the aggregate and the smallest and largest choices given. **_For a positional aggregate, the leftmost bound is determined by the applicable index constraint if the agg [... cut ...] subelement or slice of the right-hand side value.

As are signals:

10.5.2.2 Executing a simple assignment statement

>...It is also an error if the value of any value expression in the waveform does not belong to the subtype of the target.

Here that means the target has a different number of elements (the lengths don't match).

A valid bug.

And that about fills up the number of things I can juggle in my head at the same time these days (the declaration of subtype u_unsigned not withstanding).

bradleyharden, 2020-01-30:
@diogratia, thanks for digging in. I was already fairly certain that this example is valid VHDL-2008. It is a stripped down version of code that works fine in Modelsim. I'm trying to take that project and get it to run in GHDL, filing bug reports along the way.

commit a87153e4 (2020-02-02): trans-chap8: handle vector in aggregate target for variable assignments.

Fix #1120
closed 2020-02-02 by tgingold