ghdl/ghdl #1123: Function in port map crashes during simultation

malte0811, 2020-01-30:
**Description**
Using a function in a port map crashes ghdl when trying to run a simulation.

**Expected behaviour**
Run correctly if the code is valid (I think it is), or give a clear error (during analysis/elaboration) if it isn't.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity A is
port(x: in std_ulogic_vector(4 downto 0));
end entity;

architecture test of A is
begin
end architecture;

entity B is
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture test of B is
    function to_vector(signal d: unsigned(4 downto 0)) return std_ulogic_vector is
    begin
        return std_ulogic_vector(d);
    end function;

    signal s: unsigned(4 downto 0) := (others => '0');
begin
    test: entity work.A
    port map(
        x => to_vector(s)
    );
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a test.vhd
ghdl -r b
```

**Context**
Please, provide the following information:

- OS: Debian GNU/Linux bullseye/sid
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `c83bce97cdb629fc52d5b9f73f9d236c79122ae4`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1548-gc83bce97) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/media/data/VHDL/breakingghdl/
Command line:
ghdl -r b
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:1232
Call stack traceback locations:
0x5592e1ba40f8 0x5592e1bbb0c4 0x5592e1d8e136 0x5592e1d8e84d 0x5592e1d76513 0x5592e1de1fa8 0x5592e1d4dbf5 0x5592e1d4ad07 0x5592e1d58aa4 0x5592e1de66f5 0x5592e1d2d708 0x5592e1cc44f6 0x5592e1de9ea0 0x5592e1ad5583 0x7f2028decbb9 0x5592e1ad4108 0xfffffffffffffffe
******************************************************************
```

eine, 2020-01-30:
Although the function seems useless, because it might be replaced with a direct cast, I think that it is valid VHDL 2008. It feels equivalent to `foo => std_ulogic_vector(to_unsigned(s, 5))`, which would make sense if `s` was of type `integer`.

eine, 2020-01-30:
Interestingly, the following variant works:

```vhdl
--:file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity A is port(x: in std_ulogic_vector(4 downto 0)); end entity;
architecture test of A is begin end architecture;

entity B is end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture test of B is
    signal s: integer := 10;
begin
    test: entity work.A
    port map( x => std_ulogic_vector(to_unsigned(s, 5)) );
end architecture;
```

```sh
# ghdl -a test.vhd
test.vhd:19:31:error: actual expression must be globally static
ghdl:error: compilation error

# ghdl -a --std=08 test.vhd 
# ghdl -e --std=08 B
# ghdl -r --std=08 B
```

However, with the MWE posted by the OP, I get:

```sh
# ghdl -a --std=08 test.vhd 

Execution terminated by unhandled exception
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f7e8e75983e 0x7f7e9004754b 0x7f7e901d7a08 0x7f7e901d93f4 0x7f7e8fa48666 0x7f7e8fa488c1 0x7f7e8fa48c89 0x7f7e9090996d 0x7f7e909097b9 0x64ee67 0x409e1d 0x7f7e8e746099 0x409128 0xfffffffffffffffe
ghdl:error: compilation error
```

bradleyharden, 2020-01-30:
I'm not an expert on the LRM, but I think the functions here have to be locally static. Now that the `std_logic_1164` and `numeric_std` packages are considered officially part of the standard, their type conversion functions are considered locally static. However, I don't know if custom functions can ever be considered locally static, even if they only use functions from packages in the standard. I think that might be the source of the problem here.

Paebbels, 2020-01-30:
In the given code, the line

    x => to_vector(s)

is translated to:
* a signal assignment to an implicit signal `anonymous`:  
  `anonymous <= to_vector(...)`
* a port association to that implicit signal `anonymous`:  
  `x => anonymous`

> If the actual part of a given association element for a formal signal port of a block is the reserved word inertial followed by an expression, or is an expression that is not globally static, then the given association element is equivalent to association of the port with an anonymous signal implicitly declared in the declarative region that immediately encloses the block. The signal has the same subtype as the formal signal port and is the target of an implicit concurrent signal assignment statement of the form
>
>    anonymous <= E;
>
> where E is the expression in the actual part of the given association element. The concurrent signal
assignment statement occurs in the same statement part as the block.

tgingold, 2020-02-01:
The crash is due to the use of interface signal.  If you replace:
```
 function to_vector(signal d: unsigned(4 downto 0)) return std_ulogic_vector is
```
with 
```
 function to_vector(d: unsigned(4 downto 0)) return std_ulogic_vector is
```
it works correctly.

I think this is an issue in the LRM.  I don't see why an interface signal would be allowed.

commit 6ed3a692 (2020-02-01): vhdl-sem_assocs: check that interface of function conv is a constant.

Fix #1123
closed 2020-02-01 by tgingold