ghdl/ghdl #1127: Slicing vector from array results in `SYSTEM.ASSERTIONS.ASSERT_FAILURE` raised

suoto, 2020-02-09:
**Description**

Slicing vector from array results in `SYSTEM.ASSERTIONS.ASSERT_FAILURE` raised

**Expected behaviour**

Source should be synthesized

**How to reproduce?**

```vhd :file: slice_from_array.vhd
library ieee;
use ieee.std_logic_1164.all;

entity foo is
  port (
    addr : in integer;
    nibble0 : out std_logic_vector(3 downto 0);
    nibble1 : out std_logic_vector(3 downto 0)
  );
end foo;

architecture foo of foo is

  type data_array_t is array (3 downto 0) of std_logic_vector(7 downto 0);
  signal data_buffer : data_array_t;

begin

  nibble0 <= data_buffer(addr)(3 downto 0);
  nibble1 <= data_buffer(addr)(7 downto 4);

end foo;
```

```sh :image: ghdl/syth:beta
ghdl --synth rtl/foo.vhd  -e foo
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `v0.36-1576-g76e2f6b3` (from `ghdl/synth:beta`)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
root@f75417222768:/src# ghdl --synth rtl/foo.vhd  -e foo
rtl/foo.vhd:35:10:warning: signal "data_buffer" is never assigned and has no default value

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1576-g76e2f6b3) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/src/
Command line:
ghdl --synth rtl/foo.vhd -e foo
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:596
Call stack traceback locations:
0x7f9caad254f1 0x563269302b6f 0x5632692fd22a 0x563269318f78 0x5632693192fc 0x563269471364 0x5632694713d8 0x56326946ae1d 0x5632694c7eff 0x563269410916 0x563269572a9b 0x5632691da949 0x7f9caa90d099 0x5632691d91a8 0xfffffffffffffffe
******************************************************************
```

**Additional context**

If I change `data_buffer` to a constant (e.g. `constant data_buffer : data_array_t := (others => (others => '0'));`) the problem does not happen. Tried some intermediate steps between `data_buffer(addr)` and `nibble[01]` but couldn't get any of those working.

If it helps, the original file is https://github.com/phase4ground/dvb_fpga/blob/master/rtl/axi_stream_master_adapter.vhd.

tgingold, 2020-02-10:
Pass with current version.

closed 2020-02-10 by tgingold

suoto, 2020-02-11:
When trying to synth the file mentioned I still got issues (built ghdl from source from 29ff7ac2)

A slightly different version of the file breaks:

```vhd :file: slice_from_array_v2.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  generic ( TDATA_WIDTH : integer := 8);
  port (
    clk        : in std_logic;
    wr_field_0 : in std_logic_vector(TDATA_WIDTH - 1 downto 0);
    wr_field_1 : in std_logic;
    wr_en      : in std_logic;

    rd_field_0 : out std_logic_vector(TDATA_WIDTH - 1 downto 0);
    rd_field_1 : out std_logic
  );
end foo;

architecture foo of foo is

  type data_array_t is array (3 downto 0) of std_logic_vector(TDATA_WIDTH downto 0);
  signal data_buffer : data_array_t;

  signal addr : unsigned(3 downto 0) := (others => '0');

begin

  rd_field_0 <= data_buffer(to_integer(addr))(TDATA_WIDTH - 1 downto 0);
  rd_field_1 <= data_buffer(to_integer(addr))(TDATA_WIDTH);

  process(clk)
  begin
    if rising_edge(clk) then
      addr <= addr + 1;

      if wr_en = '1' then
        data_buffer(to_integer(addr)) <= wr_field_0 & wr_field_1;
      end if;

    end if;
  end process;

end foo;
```

```
+ ghdl --synth foo.vhd -e foo
foo.vhd:21:10:note: found RAM "data_buffer", width: 9 bits, depth: 4

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/project/
Command line:
ghdl --synth foo.vhd -e foo
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:616
Call stack traceback locations:
0x7f4c536c64f1 0x55d846dd8e2f 0x55d846de247a 0x55d846e64b12 0x55d846e655ad 0x55d846f4575e 0x55d846f457e0 0x55d846f3f217 0x55d846f9c955 0x55d846ee4d10 0x55d8470474f1 0x55d846cb098d 0x7f4c532ae099 0x55d846caf1a8 0xfffffffffffffffe
******************************************************************
```

commit 77e18685 (2020-02-11): netlists-memories: handle split memories.  Fix #1127