ghdl/ghdl #1128: Signal cannot be declared using subtype of unconstrained signal in port map

bradleyharden, 2020-02-09:
If I leave an entity port unconstrained, and then try to use the subtype of that port to declare a signal in the architecture, I get an error. 

```
subtype_bug.vhd:12:10:error: declaration of signal "copy" with unconstrained array subtype "unsigned" is not allowed
```

I do not think this is correct, because Modelsim accepts this without error. Here is an example.

```vhd :file: subtype_bug.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
  port(
    input : in unsigned);
end entity;

architecture rtl of test is
  signal copy : input'subtype;
begin
end architecture;
```

```sh :image: ghdl/ghdl:ubuntu18-llvm-5.0
ghdl -a --std=08 subtype_bug.vhd
```

**Context**
- OS: Ubuntu 18.04
- Origin:
  - [x] Built from sources: 76e2f6b
- Backend: LLVM

tgingold, 2020-02-26:
Unfortunately, there isn't a simple fix for this issue. It requires a little bit of thinking.

bradleyharden, 2020-02-26:
This would be a nice-to-have feature, but it's easy to work around.

I'm trying to take an existing VHDL-2008 codebase that works in Modelsim and get it working in GHDL by filing bug reports. If this one is too much work, I can change the code. But maybe you could take a look at #1129? Is that one easier?

closed 2020-08-04 by tgingold

JimLewis, 2021-05-13:
@bradleyharden   Thanks for posting your test case.   I was able to use it to create some good test cases for another similar issue.

commit 05bbe09a (2020-08-04): testsuite/gna: add a test for #1128.

Close #1128