ghdl/ghdl #1129: Setting port to constant using to_unsigned causes crash

bradleyharden, 2020-02-09:
If I try to set an input port to a constant using the `to_unsigned` function, GHDL crashes. However, if I first declare the constant, and then set the port equal to it, there is no crash. I believe this is valid VHDL-2008, because Modelsim accepts it without issue.

```vhd :file: constant_bug.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
  generic(
    BITS : positive);
  port(
    input : in unsigned(BITS - 1 downto 0));
end entity;

architecture rtl of ent is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
  constant MAX : positive := 7;
  signal input : u_unsigned(MAX - 1 downto 0);
  constant CONST : unsigned := to_unsigned(MAX, input);
begin
  ent : entity work.ent
    generic map(
      BITS => 8)
    port map(
      input => to_unsigned(MAX, input));
      --input => CONST);
end architecture;
```

```sh :image: ghdl/ghdl:ubuntu18-llvm-5.0
ghdl -a --std=08 constant_bug.vhd
```

**Context**
- OS: Ubuntu 18.04
- Origin:
  - [x] Built from sources: 76e2f6b3
- Backend: LLVM

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1576-g76e2f6b3) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/home/hardebj1/tmp/tmp/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o constant_bug.o constant_bug.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:555
Call stack traceback locations:
0x4a1230 0x4a16a6 0x4a1997 0x4aae59 0x6228a0 0x6322a8 0x632385 0x6192e8 0x5e2a26 0x5e2d32 0x5e3329 0x5e3860 0x5f4c9c 0x5f3a87 0x5ad68d 0x5ad844 0x5ae0aa 0x5aed5a 0x5b285e 0x5b2991 0x5b6e8b 0x5bc303 0x644e5f 0x588699 0x64b47e 0x64cd5a 0x64e653 0x408693 0x7fcd9c12fb95 0x407998 0xfffffffffffffffe
******************************************************************
```

bradleyharden, 2020-02-26:
I just realized that my original code here didn't make sense. I corrected the problems, but it had no effect on the bug.

diogratia, 2020-02-29:
Another test case that passes:
```vhdl
architecture rtl of test is
    constant BITS:  positive := 8;
    constant MAX:   positive := 2 ** BITS - 1;
    signal input:   u_unsigned(BITS - 1 downto 0);
    -- constant CONST:  unsigned := to_unsigned(MAX, input);
begin
ent:
    entity work.ent
        generic map (
            BITS => BITS
        )
        port map (
            -- input => to_unsigned(MAX, input) -- CRASH
            -- input => CONST                   -- WORKS
            input => to_unsigned(MAX, input'length)  -- WORKS 
        );
end architecture;
```
using the other function to_unsigned.

So what's the difference in the use of `input` in the two?

IEEE Std 1076-2008:

9.4.3 Globally static primaries  

>An expression is said to be globally static if and only if every operator in the expression denotes a pure function and every primary in the expression is a globally static primary, where a globally static primary is a primary that, if it denotes an object or a function, does not denote a dynamically elaborated named entity (see 14.6) and is one of the following:  
>>...
>> i) A function call whose function name denotes a pure function and whose actual parameters are each globally static expressions
>>...

The function call to `to_unsigned` with the second parameter a signal of type u_unsigned isn't a globally static expression while the one with the  `input'length` parameter is. This has consequence in the port clause:

6.5.6.3 Port clauses

>If the actual part of a given association element for a formal port of a block is the reserved word inertial followed by an expression, or is an expression that is not globally static, then the given association element is equivalent to association of the port with an anonymous signal implicitly declared in the declarative region that immediately encloses the block. The signal has the same subtype as the formal port and is the target of an implicit concurrent signal assignment statement of the form  
>  
>_anonymous_ <= E;  
>  
>where E is the expression in the actual part of the given association element. The concurrent signal assignment statement occurs in the same statement part as the block.

This is where the crash occurs. The equivalent anonymous signal declaration or assignment failed during analysis.

(And about here you you might wonder if it were worthwhile to have a warning of an implicitly declared signal as the result of an actual that's a non-static expression in a port clause association.)

bradleyharden, 2020-02-29:
@diogratia,  I don't think such a warning would be useful. I can't really imagine a situation where that warning would point out a mismatch between my intent and the actual behavior. Can you?

diogratia, 2020-02-29:
If you actually had authored the source for this test case would you have intended assigning a constant require the declaration of a signal?

bradleyharden, 2020-03-02:
Yes, that's what I intended. This was a minimal example that came from a real use case.

I had already defined the equivalent of `MAX` and used it more than once. I wanted to convert `MAX` to `unsigned` and apply it to a port as a constant. Since I was only using the `unsigned` version of `MAX` once, I didn't see any value in declaring it as its own constant. The new constant would be entirely redundant and would only serve to perform a type conversion. To me, it's cleaner and simpler to use a type conversion in the port map. That seems like the ideal use case for this feature of VHDL-2008.

I don't see why declaring an anonymous signal would be a problem. Other than adding a delta cycle in simulation, I don't see how it would have any effect in simulation or synthesis.

tgingold, 2020-03-02:
The problem is that the anonymous signal subtype is defined by the formal of the instance.
This requires a special handling that is not yet implemented.

commit ae3e6932 (2021-09-10): testsuite/gna: add a test for #1129

close #1129
closed 2021-09-10 by tgingold