ghdl/ghdl #1131: Can't mix generics when constraining record

cogsy23, 2020-02-10:
Here's a pretty wacky bug I've been trying to narrow down for a few hours.

When I define a record type with unconstrained arrays, and then constrain that type with a mixture of constants and generics I get a GHDL crash.

```
signal BusInst : BusT(
    A(63 downto 0),
    F(3 downto 0)
);
```
works fine.

```
signal BusInst : BusT(
    A(BUS_WIDTH-1 downto 0),
    F(OTHER_WIDTH-1 downto 0)
);
```
works fine.

just not this example:

DUT.vhd
```
library IEEE;
use IEEE.Std_Logic_1164.all;

entity DUT is
	generic (
		BUS_WIDTH : integer := 64;
		OTHER_WIDTH : integer := 4
	);
	port (
		Clk : in std_logic;
		Reset : in std_logic
	);
end entity;

architecture Behavioural of DUT is
	type BusT is record
		A : std_logic_vector;
		F : std_logic_vector;
	end record;

	signal BusInst : BusT(
		A(BUS_WIDTH-1 downto 0),
		F(3 downto 0)
	);
begin

end architecture;
```

```
$ ghdl -i --std=08 --work=work DUT.vhd
$ ghdl -m --std=08 --work=work dut
$ ghdl -r --std=08 --work=work dut --wave=wave.ghw --ieee-asserts=disable-at-0

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36 (tarball) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
In directory: /home/benc/instruments/
Command line:
ghdl -r --std=08 --work=work dut --wave=wave.ghw --ieee-asserts=disable-at-0
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-waves.adb:850 access check failed
Call stack traceback locations:
0x5637ed32609a 0x5637ed32611a 0x5637ed3220bc 0x5637ed322deb 0x5637ed322a84 0x5637ed322e04 0x5637ed322c80 0x5637ed322ffd 0x5637ed32229b 0x5637ed32624d 0x5637ed32638b 0x5637ed326569 0x5637ed3267da 0x5637ed32691f 0x5637ed329036 0x5637ed32946c 0x5637ed2ae9eb 0x5637ed32a232 0x5637ed32a37d 0x5637ed579508 0x5637ed4bfb81 0x5637ed44983c 0x5637ed579dee 0x5637ed27f6c3 0x7f1210a62b95 0x5637ed27e4a8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-02-10:
I suppose this is related to --wave.  Try without it.

cogsy23, 2020-02-10:
Yup, works without --wave.

Xiretza, 2020-03-06:
Might be related to #756?

commit 73d3e2bd (2020-08-04): grt: handle more unbounded types in disp_rti and ghw.

Fix #1131
closed 2020-08-04 by tgingold