ghdl/ghdl #1134: Predef integer division.

donnie-j, 2020-02-12:
Excellent progress, thanks Tristan! :)

Last thing stopping the J-Core J2 datapath from synthesizing.  I get a crash that the below patch fixes.  I notice that a few others are selectively the same way, I suspect it might be that way to avoid emitting division or exponentiation in the netlist?  In any event, this trips up synthesis.

```
diff --git a/src/synth/synth-static_oper.adb b/src/synth/synth-static_oper.adb
index be485799..3bf2a1b9 100644
--- a/src/synth/synth-static_oper.adb
+++ b/src/synth/synth-static_oper.adb
@@ -331,7 +331,8 @@ package body Synth.Static_Oper is
                Res_Typ);
          when Iir_Predefined_Integer_Div =>
             return Create_Value_Discrete
-              (Left.Scal / Right.Scal, Res_Typ);
+              (Get_Static_Discrete (Left) / Get_Static_Discrete (Right),
+               Res_Typ);
          when Iir_Predefined_Integer_Mod =>
             return Create_Value_Discrete
               (Get_Static_Discrete (Left) mod Get_Static_Discrete (Right),
```

There are a few places where a generic is divided by a constant in the affected RTL, but I've not narrowed down a simple reproducer for this, yet.

(In the unrelated FYI dept, something is causing yosys to throw an assert importing our instruction decoder with ghdl, but ghdl --synth produces a VHDL netlist without error... I'll see if I can isolate it)

tgingold, 2020-02-12:
Thanks!

What is the assertion generated by yosys ?

closed 2020-02-12 by tgingold

donnie-j, 2020-02-12:
> Thanks!
> 
> What is the assertion generated by yosys ?

You're welcome.

Here's how to reproduce the assert, and the snapshot of the test 
directory.  This is all opensource (BSD) you can use it for test cases 
as you like.

Cheers,
J

jeff@JeffNB:~/work/tools/jcore-j1c$
jeff@JeffNB:~/work/tools/jcore-j1c$ ./ghdl_lattice.sh
rm: cannot remove '*.o': No such file or directory
components_pkg.vhd:378:41:warning: declaration of "rotate" hides 
enumeration literal rotate [-Whide]
decode_core.vhd:113:3:warning: declaration of "decode_core" hides 
entity "decode_core" [-Whide]
decode_core.vhd:66:14:warning: declaration of "op" hides port "op" 
[-Whide]
decode_core.vhd:98:14:warning: declaration of "op" hides port "op" 
[-Whide]
datapath.vhd:207:2:warning: declaration of "datapath" hides entity 
"datapath" [-Whide]
datapath.vhd:73:25:warning: declaration of "sr" hides signal "sr" 
[-Whide]
datapath.vhd:87:28:warning: declaration of "mem" hides port "mem" 
[-Whide]
datapath.vhd:121:28:warning: declaration of "instr" hides port "instr" 
[-Whide]
mult.vhd:19:3:warning: declaration of "mult" hides entity "mult" 
[-Whide]
LED: Write 00000000 at 0 fs
LED: Write 000000FF at 1791638000 fs
LED: Write 00000011 at 2624958000 fs
LED: Write 0000004F at 3291614000 fs
LED: Write 0000002F at 69957214000 fs
LED: Write 00000012 at 110539898000 fs
LED: Write 00000021 at 156039170000 fs
LED: Write 00000022 at 187372002000 fs
LED: Write 00000031 at 241871130000 fs
LED: Write 00000032 at 269204026000 fs
LED: Write 00000033 at 285953758000 fs
LED: Write 00000041 at 374285678000 fs
LED: Write 00000042 at 396868650000 fs
LED: Write 00000043 at 419451622000 fs
LED: Write 00000044 at 443034578000 fs
LED: Write 00000045 at 466450870000 fs
LED: Write 00000046 at 489867162000 fs
LED: Write 00000047 at 591615534000 fs
LED: Write 00000051 at 611948542000 fs
LED: Write 00000061 at 697947166000 fs
LED: Write 00000062 at 786279086000 fs
LED: Write 0000007F at 790195690000 fs
LED: Write 000000C0 at 794112294000 fs
CPU tests passed
GDB Stub for HS-2J0 SH2 ROM
revision: changeset:   44:ccfd5949190e
build: Tue Oct  8 15:16:19 EDT 2019
rodata string
ghdl:info: simulation stopped by --stop-time @1999968ns
jeff@JeffNB:~/work/tools/jcore-j1c$
jeff@JeffNB:~/work/tools/jcore-j1c$ ghdl -e decode
jeff@JeffNB:~/work/tools/jcore-j1c$
jeff@JeffNB:~/work/tools/jcore-j1c$ yosys -m ghdl

 
/----------------------------------------------------------------------------\
 |                                                                      
      |
 |  yosys -- Yosys Open SYnthesis Suite                                 
      |
 |                                                                      
      |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>     
      |
 |                                                                      
      |
 |  Permission to use, copy, modify, and/or distribute this software 
for any  |
 |  purpose with or without fee is hereby granted, provided that the 
above    |
 |  copyright notice and this permission notice appear in all copies.   
      |
 |                                                                      
      |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL 
WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF    
      |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE 
FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY 
DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN 
AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING 
OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.      
      |
 |                                                                      
      |
 
\----------------------------------------------------------------------------/

 Yosys 0.9+1706 (git sha1 81e6b040, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC 
-Os)

yosys> ghdl decode
1. Executing GHDL.
decode.vhd:58:12:warning: no assignment for offsets 1:24
decode_core.vhd:167:61:warning: comparing non-numeric vector is 
unexpected
decode_core.vhd:56:10:warning: signal "instr_state_sel" is never 
assigned and has no default value
decode_core.vhd:92:38:note: gate i338 cannot be part of a memory
Importing module decode.
Importing module decode_core_0_fc179eeb1358558d0b64a188805b6cb1773c394a.
Importing module decode_table.
ERROR: Found error in internal cell \decode.\75 ($mux) at 
kernel/rtlil.cc:807:
  cell $mux \75
    parameter \WIDTH 14
    connect \Y $auto$ghdl.cc:522:import_module$24
    connect \S $auto$ghdl.cc:522:import_module$7
    connect \B 14'00000000000000
    connect \A $auto$ghdl.cc:522:import_module$22
  end
jeff@JeffNB:~/work/tools/jcore-j1c$

tgingold, 2020-02-12:
Yes, something is going wrong but I haven't yet found what.

tgingold, 2020-02-12:
If you declare the ports in the component in the same order as in the entity it works.

donnie-j, 2020-02-13:
On Feb 13, 2020, at 04:03, tgingold <notifications@github.com> wrote:
> 
> If you declare the ports in the component in the same order as in the entity it works.

confirmed.  using that work around/correction to decode_core gets us to the point where the whole of the ‘cpu’ entity will import!

Yosys is pretty happy, but then abc throws an assert... one step at a time :)

tgingold, 2020-02-13:
Ouch.  Try to write as verilog, and to re-import.  At least that will allow you to post a testcase.

donnie-j, 2020-02-14:
On 金, 2月 14, 2020 at 2:37 午前, tgingold 
<notifications@github.com> wrote:
> Ouch.  Try to write as verilog, and to re-import.  At least that will 
> allow you to post a testcase.

Actually, d47da51f fixes that.  Must have been more places where we 
have subtle out of order, but still legal, ports.  Also some strange 
warnings went away, probably related.

We now synthesize and map for Ice40 and Xilinx series 7 ghdl->yosys.  
Excellent, thanks for making it happen :)

J.

tgingold, 2020-02-14:
Great!  Thank for the status report.

commit 5bcc890f (2020-02-12): synth-static_oper: handle more division operands.  Fix #1134

From donnie-j