ghdl/ghdl #1135: Incorrect error during elaboration: "several sources for unresolved signal"

MJoergen, 2020-02-12:
**Description**
When running the simulator (using ghdl -r) the following error is reported:
```
for signal: .test(synthesis).stages(6).state
./test:error: several sources for unresolved signal
./test:error: error during elaboration
```

**Expected behaviour**
The code does NOT contain multiple drivers. This is verified by synthesizing using Xilinx Vivado and running in hardware.
The simulator should run the code without generating this error.

**How to reproduce?**
I've reduced the design to the following single entity:

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity test is
end entity test;

architecture synthesis of test is

   signal clk_i     : std_logic;
   signal rst_i     : std_logic;

   type STATE_ADSR_t is (ATTACK_ST, DECAY_ST, SUSTAIN_ST, RELEASE_ST);

   type stage_t is record
      state        : STATE_ADSR_t;
   end record stage_t;
   
   constant C_STAGE_DEFAULT : stage_t := (
      state        => RELEASE_ST
   );

   type stages_t is array (0 to 32) of stage_t; -- Stage 32 is the same device as stage 0.
   signal stages : stages_t := (others => C_STAGE_DEFAULT);

begin

   -----------------------------
   -- Generate clock and reset
   -----------------------------

   -- Generate cpu clock @ 8.33 MHz
   proc_clk : process
   begin
      clk_i <= '1', '0' after 60 ns;
      wait for 120 ns;
   end process proc_clk;

   -- Generate cpu reset
   proc_rst : process
   begin
      rst_i <= '1', '0' after 500 ns;
      wait;
   end process proc_rst;

   p_stage6 : process (clk_i)
   begin
      if rising_edge(clk_i) then
         stages(6).state <= stages(5).state;
      end if;
   end process p_stage6;

   gen_stages : for i in 1 to 32 generate
      p_stages : process (clk_i)
      begin
         if rising_edge(clk_i) then
            if i >= 7 then
               stages(i).state <= stages(i-1).state;
            end if;
         end if;
      end process p_stages;
   end generate gen_stages;

end architecture synthesis;
```

```sh
ghdl -i --ieee=synopsys --std=08 --work=work test.vhd
ghdl -m --ieee=synopsys --std=08 -frelaxed-rules test
ghdl -r test --assert-level=error --wave=test.ghw --stop-time=100us
```

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04 LTS
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X ] Built from sources: `commit 2388da390b41e5fe3aef8c7b7e7a2a610944f362`

**Additional context**
"ghdl -v" outputs the following
```
GHDL 0.36-dev (20181129) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

eine, 2020-02-12:
I'm not sure using `i` as the condition inside the process is a good idea...

MJoergen, 2020-02-12:
I've tried replacing the generate with a loop instead. The following gives the same result:
```
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         for i in 1 to 32 loop
            if i >= 7 then
               stages(i).state <= stages(i-1).state;
            end if;
         end loop;
      end if;
   end process p_stages;
```

eine, 2020-02-12:
Which is equivalent to:

```vhdl
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         for i in 1 to 32 loop
            if i >= 7 then
               stages(i).state <= stages(i-1).state;
			else
               stages(i).state <= stages(i).state;
            end if;
         end loop;
      end if;
   end process p_stages;
```

Hence, you are assigning each `stages(i)` 32 times.

Should you want to generate a shift-register, you can either use `if generate` inside the `for generate`, or change the `for loop` to match the range (without the condition inside the process).

MJoergen, 2020-02-12:
Ok, I suppose this might be an issue with my understanding of the VHDL language. I tried the following, which still gives the same error:
```
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         for i in 7 to 32 loop
            stages(i).state <= stages(i-1).state;
         end loop;
      end if;
   end process p_stages;
```
In this last example I wouldn't expect any error, because i=6 does never appear in the loop.

MJoergen, 2020-02-12:
However, the following does NOT produce this error:
```
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         stages(7).state <= stages(6).state;
      end if;
   end process p_stages;
```
So the behaviour seems to be related to the use of the for loop.

eine, 2020-02-12:
I would have written it as in https://github.com/ghdl/ghdl/issues/1135#issuecomment-585171184. I think that should be valid for both simulation and synthesis.

*ping* @Paebbels

MJoergen, 2020-02-12:
Well, it appears to be the enumeration, rather than the record. I tried with
```
   type stage_t is record
      state        : std_logic;
   end record stage_t;
```
and the error then disappears.

Xiretza, 2020-02-12:
`std_logic` is a resolved type, so multiple assignments aren't a simulation error - the equivalent unresolved `std_ulogic` should again cause an error.

eine, 2020-02-12:
@Xiretza, however, in https://github.com/ghdl/ghdl/issues/1135#issuecomment-585171184 there are not multiple assignments to the signals inside the record, are there?

Xiretza, 2020-02-12:
Oh, sorry, I guess I phrased that wrong. With the current ghdl, `std_ulogic` will probably cause an error just like the custom enumerated type, even though it probably shouldn't.

tgingold, 2020-02-12:
ghdl is correct here.  Just use `gen_stages : for i in 7 to 32 generate`.

The synthesizer can remove unused code even if it creates a driver according to the standard.

MJoergen, 2020-02-12:
@tgingold : Thank you for your reply. However, I'm still trying to understand, why you think ghdl is correct. In other words, what am I misunderstanding.

In the following example, ghdl produces an error (correctly, according to you):
```
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         for i in 7 to 32 loop
            stages(i).state <= stages(i-1).state;
         end loop;
      end if;
   end process p_stages;
```

However, in this example ghdl does NOT produce an error:
```
   p_stages : process (clk_i)
   begin
      if rising_edge(clk_i) then
         stages(7).state <= stages(6).state;
      end if;
   end process p_stages;
```
I fail to see, why the first example should generate an error, and the second example should not.

tgingold, 2020-02-12:
A driver is added for the longest static prefixes.

So for:

>          for i in 7 to 32 loop
>             stages(i).state <= stages(i-1).state;
>          end loop;

A driver is created for `stages`.

But here:

>       if rising_edge(clk_i) then
>          stages(7).state <= stages(6).state;
>       end if;

a driver is created for `stages(7).state`.

MJoergen, 2020-02-12:
@tgingold : Thank you for the explanation. That makes sense. I learned something new today :-)

I guess this is a non-issue, and can be closed.

closed 2020-02-12 by eine