ghdl/ghdl #1136: GHDL Synth + Yosys results in ERROR: wire not found for $edge

suoto, 2020-02-14:
**Description**

I'm trying to use GHDL Synth + Yosys to build a design but I'm getting `ERROR: wire not found for $edge` when running `ghdl file1 file2 -e name_of_the_design_unit` (no further warnings or errors before this). Running `ghdl --synth` outside Yosys completes fine. I'm using the latest version of both (at least at the time of writing).

As I see, this could be an issue with either GHDL, Yosys or the way I wrote the code.

It seems the problem is with this exact line: https://github.com/phase4ground/dvb_fpga/blob/edb25f5293a7f741092a5a3ff4a2d362407b61f8/rtl/axi_bit_interleaver.vhd#L183

Replacing that with either

```vhd
  wrdata_a  => std_logic_vector(to_unsigned(column, DATA_WIDTH)),
```
or
```vhd
  wrdata_a  => (others => '0'),
```

seems to work just fine.

I've attached the output of `Yosys -X` here: [yosys.log](https://github.com/ghdl/ghdl/files/4202110/yosys.log)

**Expected behaviour**

Pointing to problematic line on the VHDL side would be great, although I'm not sure how feasible that is :)

**How to reproduce?**

```
git clone --recurse-submodules --branch=add_dvbs2_tx_top_level https://github.com/phase4ground/dvb_fpga
cd dvb_fpga
yosys -m ghdl dvbs2_tx.ys
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 47da51fc

eine, 2020-02-14:
> Pointing to problematic line on the VHDL side would be great, although I'm not sure how feasible that is :)

Ref #898

tgingold, 2020-02-14:
I am worried by the line 393:
```
ram_wr_data(wr_column_cnt_i) <= wr_data_mux(to_integer(wr_remainder));
```
where you copy from RAM to RAM.  Maybe it would work if you use an intermediate signal.

suoto, 2020-02-14:
While trying out some alternatives, I found out that adding `ram_wr_data <= (others => (others => 'U'));` (see [patch.txt](https://github.com/ghdl/ghdl/files/4204531/patch.txt)) to the reset branch seems to solve this issue.

FTR, other alternatives I tried that had no effect were

1. Indexing `ram_wr_data` with the same conversion
1. `ram_wr_data(to_integer(<the/actual/index>)`
1. Converting the index to integer and storing it on a variable
1. Converting the index to integer and storing it on a signal
1. Instead of  `ram_wr_data(wr_column_cnt_i) <= wr_data_mux(to_integer(wr_remainder));`, I created an intermediate signal for `wr_data_mux(to_integer(wr_remainder))`
1. In the previous setting, even if don't assign anything to the intermediate signal the result is the same

Please let me know if you'd like me to try something else!

tgingold, 2020-02-14:
I can reproduce the issue.  But I haven't yet found the cause.

tgingold, 2020-02-16:
I think this is now fixed.

suoto, 2020-02-17:
Tried on ghdl e4aa89cd (and Yosys cd60f079) and seems to have worked indeed, thank you! (I'll leave for you to close this as I don't know if you plan to work on it further).

Just an FYI, I'm not yet testing the builds in real hardware; when I say "it works", I mean that the synth process gets past the point it was failing before. I noticed that `ghdl --synth` produces a VHDL, would it add anything if I ran some tests on that?

tgingold, 2020-02-17:
You can do a post-synthesis simulation, but that's not very useful in general.

closed 2020-02-17 by tgingold