ghdl/ghdl #1137: Vector indexed with variable in process is assigned XX

marzoul, 2020-02-14:
**Description**
Vector-to-vector copy wrongly assigns "XX" instead of "00" in destination.

**How to reproduce?**
Compile with VHDL 2008, also tested with 2002 by rewriting sensitivity to "all", same issue.
See testcase file :
[testbench.vhd.gz](https://github.com/ghdl/ghdl/files/4205963/testbench.vhd.gz)

The testcase implements 4 different variants of a normally simple std_logic vector-to-vector assignment.
Destination is initialized to "11", source is "00". One variant fails to copy, gets "XX" instead of "00".
The source of the issue seems to be the initialization/assignment of the variable used as vector. 

**Context**

OS is Archlinux.
GHDL built from sources.

GHDL 0.37-dev (v0.36-1603-g47da51fc) [Dunoon edition]
 Compiled with GNAT Version: 9.2.0

Issue happens with all backends (gcc, llvm, mcode).

tgingold, 2020-02-14:
ghdl is correct here.  In gen2 you have two processes that drive data_dst2, so it results in conflicts as indicated by 'X'.

marzoul, 2020-02-14:
This can't be the only explanation, at least because gen4 works and there are 2 processes driving data_dst4.
- Identical in gen4 and gen2, the 2 "potentially-conflicting" processes drive non-overlapping ranges in destination.
- The only difference between gen2 and gen4 is how variable idx is assigned. Or maybe the type of "idx", variable or constant.

This issue has driven me mad for several days in a much much larger design that I managed to narrow down to that testcase... I was used to this working in synthesis (Vivado).

marzoul, 2020-02-14:
Additional detail not visible in the testcase:
Change the source init to "01", you will see that there is no X assigned for indexes where data being written is identical to current value of destination (destination data stays "1" there).
The behavior of GHDL lacks consistency:
- There are some X generated because the vector/array as a whole has 2 driving processes
- But the X are only generated where driven data do change the value of the array
- And multi-driver is OK if it's concurrent assignment in multiple generates with no process involved

So GHDL detects multiple drivers and also checks if data would change.
At least a warning would save time and nerves ;-)
Better, even if such constructs are no allowed in VHDL for obsure reasons (but I don't see one that would be acceptable), then maybe GHDL could still allow it as relaxed rules ?

I understand that synthesis tools could have trouble handling assignments in arrays where addresses are the result of a computation and can't be known statically. But this should not be an issue for simulation and it is visibly not an issue for GHDL already because it does handle when assigned value doesn't change the destination.

tgingold, 2020-02-15:
In gen4, one process drivers data_dst4(0) and the other data_dst4(1) because the index is static (a constant).  In gem2 the index is not static so drivers are created for data_dst2.

tgingold, 2020-02-15:
Because if two drivers drives the same value there is no conflict.

GHDL strictly follows the VHDL standard.  You are using resolved signals.
If you don't want to have resolved signals, use std_ulogic/std_ulogic_vector instead of std_logic/std_logic_vector.
(Really, just try on your testbench.vhd example).

marzoul, 2020-02-15:
I see... but I want to use resolved signals exactly to detect early I am creating physical multi-driven wires, so that I don't discover this weeks later in synthesis. Resolved signals may also be mandatory per company coding rules for the sake of regression tests or style or whatever reason.
Here I was certain that I was not creating multi-driver but simulation was saying that I was, so this was strongly misleading. And also impossible to understand because for all common sense it is purely "coding style" related.

In my small example I can rewrite easily, only if I am aware that VHDL rules will hit me there.
It's just that, in very complex projects, the coding style of gen2, with multiple processes driving different ranges or an array, may be what is clearer to write and review.

Would it be possible to, either (or both xD):
- emit a warning at some compilation step, about multiple processes driving an array with non-static indexes ?
- make simulation not emit XX because indeed accessed indexes do not conflict ? Under an option of course, if such behavior is not pure VHDL standard.

Regards

tgingold, 2020-02-15:
> I see... but I want to use resolved signals exactly to detect early I am creating physical multi-driven wires, so that I don't discover this weeks later in synthesis.

I don't understand this sentence.  If you want to detect early multi-driver wires, just use *non* resolved signals.

> Resolved signals may also be mandatory per company coding rules for the sake of regression tests or style or whatever reason.

You should complain that the company coding rules aren't code as they prevent from early error detection.  In vhdl08, std_ulogic_vector are compatible with std_logic_vector so there aren't any good reason not to use std_ulogic_vector.

> Would it be possible to, either (or both xD):
> 
> 	• emit a warning at some compilation step, about multiple processes driving an array with non-static indexes ?

I tried that before, but there are too many false positive as soon as you have if-generate statements.

> 	• make simulation not emit XX because indeed accessed indexes do not conflict ? Under an option of course, if such behavior is not pure VHDL standard.

Humm, not sure about that.  But an option could be added to not allow multi-driven signals.

marzoul, 2020-02-16:
The root cause is that "multi-driver wire" is, for the HW designer that I am, a per-unique-wire concept, whereas for VHDL it is more a per-declaration-statement concept. I don't know that, it's super disappointing, I always thought that resolution was done for all individual bits...

Then I guess std_ulogic is not appropriate for tristates and in case of tristates involved then the coding style of gen2 is also not appropriate ?

eine, 2020-02-17:
@marzoul, would it be possible to use two signals, each of them being a subset that contains the range to be used in a process only? Then, the output of both signals can be aggregated in a concurrent statement.

marzoul, 2020-02-17:
@eine, there are many workarounds but this one is not an option because the driven indexes are the result of a computation that depends on externally-provided matrices of constants.
I have converted my code to generate-only statements where that computation is placed at initialization of constants. Not great for code readability, but works :-)
I am now closing the issue because that was actually a lack of vhdl expertise from me.

closed 2020-02-17 by marzoul