ghdl/ghdl #1138: port width constraint by function

Gabriel-Degret, 2020-02-16:
crc_pkg.vhd :

    library ieee;
        use ieee.std_logic_1164.all;
        use ieee.numeric_std.all;
        use ieee.math_real.all;
        
    package crc_pkg is
    
      type crcParam_t is record
        selLen  : integer; 
        poly    : std_ulogic_vector;
        iniVect : std_ulogic_vector; 
        refIn   : boolean;   
        refOut  : boolean;    
        xorOut  : std_ulogic_vector;
      end record;
      
      pure function getCrc32Param( stdCrc : string 
      ;                            datLen : integer   
      ) return crcParam_t;
      
    end crc_pkg;
    
    package body crc_pkg is
    
      pure function getCrc32Param( stdCrc : string
      ;                            datLen : integer                
      ) return crcParam_t is
        variable crcParam_v : crcParam_t( poly    ( 31 downto 0)  
                                        , iniVect ( 31 downto 0)
                                        , xorOut  ( 31 downto 0) );
      begin
        if ( "CRC-32/CCITT-FALSE" = stdCrc ) then
          crcParam_v.selLen  := datLen / 8  ; 
          crcParam_v.poly    := X"04C11DB7" ;
          crcParam_v.iniVect := X"FFFFFFFF" ; 
          crcParam_v.refIn   := true        ;
          crcParam_v.refOut  := true        ;
          crcParam_v.xorOut  := X"FFFFFFFF" ;
        else   
         crcParam_v.selLen  := datLen / 8  ; 
         crcParam_v.poly    := X"000000AF" ;
         crcParam_v.iniVect := X"00000000" ; 
         crcParam_v.refIn   := false       ;
         crcParam_v.refOut  := false       ;
         crcParam_v.xorOut  := X"00000000" ;
          assert false report 
          " Standard crc not implemented Yet."
          severity failure;
        end if;
        return crcParam_v;
      end function;
      
    end package body crc_pkg;

wbCrc_syn.vhd:

    library ieee;
        use ieee.std_logic_1164.all;
        use ieee.numeric_std.all;
        use IEEE.MATH_REAL.all;

    library ghdl_test;
        use ghdl_test.crc_pkg.all;

    entity wbCrc_syn is 
        generic(  yes_G        : std_logic :='1'
                ; fClkSys_Hz_G : integer   := 100e6
                ; isSimul_G    : boolean   := false
                ; wbDataWidth_G : integer   :=  8   
                ; wbAddrWidth_G : integer   := 32  
                ; wbTgaWidth_G  : integer   :=  1 
                ; wbTgcWidth_G  : integer   :=  1 
                ; wbTgdWidth_G  : integer   :=  1 
                ; crcDefault_G  : String    := "CRC-32/CCITT-FALSE"   
               );
        port( clk_i         : in    std_ulogic; 
        
              rx_is_dat_i   : in    std_ulogic_vector( wbDataWidth_G     - 1 downto 0 );
              rx_is_adr_i   : in    std_ulogic_vector( wbAddrWidth_G     - 1 downto 0 );
              rx_is_sel_i   : in    std_ulogic_vector( wbDataWidth_G / 8 - 1 downto 0 );
              rx_is_loc_i   : in    std_ulogic;
              rx_is_cyc_i   : in    std_ulogic;
              rx_is_stb_i   : in    std_ulogic;
              rx_is_we_i    : in    std_ulogic;
              rx_os_ack_o   :   out std_ulogic;
              rx_os_stl_o   :   out std_ulogic;
              rx_os_err_o   :   out std_ulogic;
              
              tx_os_cyc_o  :    out std_ulogic;
              tx_os_stb_o  :    out std_ulogic;
              tx_os_we_o   :    out std_ulogic;
              tx_os_dat_o  :    out std_ulogic_vector( getCrc32Param( crcDefault_G , 8 ).poly'length - 1 downto 0 );
              tx_os_cti_o  :    out std_ulogic_vector(2 downto 0) ;
              tx_os_sel_o  :    out std_ulogic_vector( wbDataWidth_G / 8 - 1 downto 0 );
              tx_os_loc_o  :    out std_ulogic;
              tx_is_ack_i  : in     std_ulogic
            );
    end wbCrc_syn;
    architecture rtl of wbCrc_syn  is  
    begin 

    end rtl;

get this error:

    ******************** GHDL Bug occurred ***************************
    Please report this bug on https://github.com/ghdl/ghdl/issues
    GHDL release: 0.36 (v0.36) [Dunoon edition]
    Compiled with GNAT Version: 7.3.0
    Target: i686-w64-mingw32
    In directory: F:\vhdl\ghdl_test\port_width_by_func\
    Command line:
    F:\vhdl\sim\ghdl\0.36-mingw32-mcode\bin\ghdl.exe -a --std=08 --work=ghdl_test .\crc_pkg.vhd .\wb_crc_syn.vhd
    Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
    Exception information:
    raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Parent
    Call stack traceback locations:
    0x5b2986 0x5b7cea 0x5caeec 0x5c2b32 0x5caf43 0x5bfe03 0x5c04f7 0x5c0910 0x5bcbd4 0x5bcf50 0x5bedab 0x5bf27b 0x5ccc83 0x5cdb7c 0x5d89e3 0x5df291 0x5a2a12 0x5f387f 0x589224 0x69b94f 0x402635 0x401389 0x76c86357 0x76fc7b72 0x76fc7b42
    ******************************************************************

tgingold, 2020-03-01:
The crash is due to
```
getCrc32Param( crcDefault_G , 8 ).poly'length
```

I am not sure you can apply 'length to the result of a function call, because it neither denotes an array subtype nor is an array object.

The list of objects is in LRM08 6.4.1

tgingold, 2020-03-01:
If you are using `-frelaxed`, this will be allowed (without a crash).  Otherwise it is an error.

commit b8ea7696 (2020-03-01): vhdl: a function call is not an object.  Fix #1138.

Report a warning (or an error if not relaxed) when a non-object name
is used for an array attribute.
Also consider subtype attribute as a type name.
closed 2020-03-01 by tgingold