ghdl/ghdl #1139: synth: unsupported IIR_KIND_ANONYMOUS_SIGNAL_DECLARATION with configurations

Xiretza, 2020-02-18:
**Description**
When combining configurations with anonymous signals in port maps, synthesis crashes.

**Expected behaviour**
The design is synthesized correctly.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity child is
	port (
		x : in std_logic;
		y : out std_logic
	);
end;

architecture a of child is
begin
	y <= x;
end;

-----------------------------------

library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		a : in std_logic;
		b : in std_logic;
		q : out std_logic
	);
end;

architecture a of ent is
	component child_comp is
		port (
			x : in std_logic;
			y : out std_logic
		);
	end component;
begin
	child_inst: child_comp
		port map (
			x => a and b,
			y => q
		);
end;

-----------------------------------

configuration conf of ent is
	for a
		for child_inst : child_comp
			use entity work.child;
		end for;
	end for;
end configuration;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e conf
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 9ee1a4a1

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
iterator_decl: cannot handle IIR_KIND_ANONYMOUS_SIGNAL_DECLARATION (ent.vhd:39:31)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1610-g9ee1a4a1) [Dunoon edition]
Compiled with GNAT Version: 9.2.1 20200130
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth --std=08 ent.vhd -e conf
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x564353369968 0x5643533d2104 0x5643533d2140 0x5643533d4bca 0x5643533d4ea8 0x564353450809 0x56435344f311 0x564353455872 0x56435340a853 0x56435340b763 0x5643534994bf 0x56435349b03c 0x5643534c14a6 0x5643534c1f56 0x564353401e8c 0x564353575283 0x5643531bbda8 0x7ff6a8466021 0x5643531ba58c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-02-25:
I was not able to reproduce the error.  I have added the reproducer as a testcase.

closed 2020-02-25 by tgingold

Xiretza, 2020-02-25:
Confirmed, my local project fails with 9ee1a4a, but works with master (ab2e4632).