ghdl/ghdl #1141: Error trying to exec 'cc1' when building GHDL from sources after installing gnat community.

talonmyburgh, 2020-02-18:
Hi guys, new to this tool but trying to use it since I need it to run VUnit simulations (since XSim is not supported). I'm using PopOS! 18.04 LTs on a Dell Inspiron machine. I cloned the ghdl master branch. The output of running `git rev-parse origin/master` is:

`939221749e7f1228ed959d1007a3fd3e3b4b9146`

I installed gnat from [adacore](https://www.adacore.com/community) and tried doing the example mentioned [here](https://ghdl.readthedocs.io/en/latest/getting/mcode.html).

When I type `make` I get the following:

`gcc -c -g -DWITH_GNAT_RUN_TIME -o jumps.o /home/talon/Desktop/ghdl/build/../src/grt/config/jumps.c gcc: error trying to exec 'cc1': execvp: No such file or directory ../src/grt/Makefile.inc:134: recipe for target 'jumps.o' failed make: *** [jumps.o] Error 1`

I am trying to build using mcode, but may not be doing this right. I suspect this error has something to do with the installing of ada altering my gcc version to a version not containing cc1 ... but I don't know how to rectify this. If I run `which gcc` I get:
`/opt/gps/bin/gcc` - which is the gcc that came with the ada install, see when I type `gcc -v` I get:
`Using built-in specs. COLLECT_GCC=gcc OFFLOAD_TARGET_NAMES=nvptx-none Target: x86_64-pc-linux-gnu Configured with: ../src/configure --enable-languages=ada,c,c++ --enable-dual-exceptions --enable-offload-targets=nvptx-none --enable-_cxa_atexit --enable-threads=posix --with-bugurl=URL:mailto:report@adacore.com --disable-nls --without-libiconv-prefix --disable-libstdcxx-pch --disable-libada --enable-checking=release --disable-multilib --with-mpfr=/it/sbx/a2c2/x86_64-linux/mpfr_stable-c/install --with-gmp=/it/sbx/a2c2/x86_64-linux/gmp_stable-c/install --with-mpc=/it/sbx/a2c2/x86_64-linux/mpc_stable-c/install --with-build-time-tools=/it/sbx/a2c2/x86_64-linux/gcc-c/build/buildtools/bin --prefix=/it/sbx/a2c2/x86_64-linux/gcc-c/pkg --build=x86_64-pc-linux-gnu Thread model: posix gcc version 8.3.1 20190518 (for GNAT Community 2019 20190517) (GCC)` 

I'm not getting ada to compile a test adb file. If I run `gprbuild greet.adb` I get the following error:

`using project file /opt/gps/share/gpr/_default.gpr Compile [Ada] greet.adb gcc: error trying to exec 'gnat1': execvp: No such file or directory gprbuild: *** compilation phase failed`

And even if I try to compile a simple c file now, I get an error. Running `gcc hi.c` gives:
`gcc: error trying to exec 'cc1': execvp: Not a directory`

Should I not have installed ada from where I did? How do I proceed? 
Thank you.

**EDIT** PopOS does have gnat in it's package manager but [here](https://ghdl.readthedocs.io/en/latest/getting/index.html) it mentioned I could get ada either from the site or by installing the gcc-ada package etc - so I got the full install from the adacore site.

closed 2020-02-18 by talonmyburgh

eine, 2020-02-19:
> I installed gnat from [adacore](https://www.adacore.com/community) and tried doing the example mentioned [here](https://ghdl.readthedocs.io/en/latest/getting/mcode.html).

> PopOS does have gnat in it's package manager but [here](https://ghdl.readthedocs.io/en/latest/getting/index.html) it mentioned I could get ada either from the site or by installing the gcc-ada package etc - so I got the full install from the adacore site.

Note that the details in that page are related to:

> https://ghdl.readthedocs.io/en/latest/getting/index.html
> Since GHDL is written in Ada, independently of the code generator you use, the a compiler is required. Most GNU/Linux package managers provide a package named gcc-ada or gcc-gnat. Alternatively, GNU Ada compiler, GNAT GPL, can be downloaded anonymously from libre.adacore.com (2014, or later; for x86, 32 or 64 bits). Then, untar and run the doinstall script.

Hence, the suggested approach is to install package `gcc-ada`/`gcc-gnat`/`gnat`/`gnat-*` through the package manager of your distribution. When it is not available, using AdaCore's distribution is an alternative. However, as it happened to you, it might not work straightaway.
 
> I am trying to build using mcode, but may not be doing this right. I suspect this error has something to do with the installing of ada altering my gcc version to a version not containing cc1 ... but [...]

It seems that the content you installed from AdaCore is badly mixed with whatever you had already, or whatever you installed afterwards. If you are using AdaCore's distribution, you should ensure that all the tools which are used are from that distribution. Hence, you need to be careful with envvars and symbolic links that are pointing to other resources, by default.

It might also be that AdaCore's distribution does not work on your system. If it is dynamically linked, it might depend on some version of the standard library that does not match your host. You might try with an older or earlier version. Nonetheless, as said, the package manager is a safer option.

> **EDIT:** I fixed this by just uninstalling the gnat community version I installed from [here](https://www.adacore.com/community) and instead just used the PopOS gnat package. Other errors have arisen but as far as this issue is concerned, I seem to have solved it.

Glad you could fix it! Are those other errors related to the build or usage of GHDL?

talonmyburgh, 2020-02-19:
Thanks @eine,

I am quite puzzled by why the adacore install messed with my symbolic links etc and why it appeared to contain an outdated gcc version (or at least one that didn't contain c11), but since I won't be programming in ada much and dont need the gps IDE, I'm fine with just using the gnat package that is in my package manager.

My new errors are related to the building of GHDL. After configuring with mcode and then running `make` I get a similar error to #831 see below:
`
x86_64-linux-gnu-gnatlink-7 ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o chkstk.o jumps.o times.o grt-cstdio.o grt-cgnatrts.o grt-cvpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=/home/talon/Desktop/ghdl/build/../src/grt/grt.ver -Wl,--export-dynamic
/usr/bin/ld: jumps.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: grt-cstdio.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: grt-cvpi.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
x86_64-linux-gnu-gnatlink-7: error when calling /usr/bin/x86_64-linux-gnu-gcc-7
gnatmake: *** link failed.
Makefile:186: recipe for target 'ghdl_mcode' failed
make: *** [ghdl_mcode] Error 4
`
***Note that the above is just the last section of the output containing the error, I'll add all the output if need be.***

I see by the reply @eine gave in #831 that for a host of situations this error arises. I then thought to try build using llvm (I installed LLVM version `1:6.0-41~exp5~ubuntu1` and clang version `1:6.0-41~exp5~ubuntu1`). I configure it with `../configure --with-llvm-config --prefix=PREFIX` and run `make` and get the following error:

`x86_64-linux-gnu-gnatlink-7 ghdl_llvm.ali -g grt-cstdio.o
/usr/bin/ld: grt-cstdio.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
x86_64-linux-gnu-gnatlink-7: error when calling /usr/bin/x86_64-linux-gnu-gcc-7
gnatmake: *** link failed.
Makefile:338: recipe for target 'ghdl_llvm' failed
make: *** [ghdl_llvm] Error 4
`
I'm hesitant, but shall I now try build with gcc? If so, I can't find the script to install the necessary dependencies mentioned [here](https://ghdl.readthedocs.io/en/latest/getting/GCC.html#build-gcc).

Thanks very much.
Ps. sorry if this ought to be in a new issue.

eine, 2020-02-19:
Using configure option [`--default-pic`](https://github.com/ghdl/ghdl/blob/master/configure#L113) should fix both builds (mcode or LLVM). There is some logic to automatically detect when it is required, but it seems that it is not working here.

> I'm hesitant, but shall I now try build with gcc?

Unless you need to debug or you really need code coverage (which is not a first-class feature), you don't want to try GCC. It is much slower to build, the artifacts are larger and it does not provide any significant advantage compared to LLVM.

BTW, we provide ready-to-use docker images: https://github.com/ghdl/docker. You can use those to try different backends without actually building all of them.

talonmyburgh, 2020-02-19:
Okay, I tried using that configure option in the mcode build as follows:

`../configure --prefix=PREFIX --default-pic`

but still got the same error as earlier. I did not know about these docker images so I think I'll give these a try instead :dancer:
Thanks.

eine, 2020-02-19:
Try cleaning the build/work dir. If some of those files were already compiled, make won't build them again, even if they are wrong. Also, `PREFIX` should be `$PREFIX`. I assume that's just a typo here.

> I did not know about these docker images so I think I'll give these a try instead

Do not hesitate to ask, should you need help to understand which to use.

talonmyburgh, 2020-02-19:
Ahah! Cleaning the build directory/inserting the `$` sign seemed to do it. 

Please note that [here](https://ghdl.readthedocs.io/en/latest/getting/mcode.html) there is no `$` sign in front of `PREFIX`. I used that command directly. 

Thanks so much @eine

eine, 2020-02-19:
> Please note that [here](https://ghdl.readthedocs.io/en/latest/getting/mcode.html) there is no `$` sign in front of `PREFIX`. I used that command directly.

In that case `PREFIX` is a keyword, which is expected to be replaced by the user with `/usr/local`, `/opt/ghdl`, whatever. Instead, `$PREFIX`, as used here, is an environment variable. I see that it can be misleading. Anyway, if you were not aware of this, it probably means that it is empty, so you can just remove that argument and use `../configure --default-pic`.

> Thanks so much @eine

I’m happy to help