ghdl/ghdl #1145: textio.readline: DOS line endings cause additional lines

eschmidscs, 2020-02-22:
**Description**
Reading a text file with readline causes spurious empty lines to be "read" if the text file has DOS style line endings.
The issue is definitely connected to the line ending. Converting the file to \n only removes the problem.
The issue is somehow connected to the line length, but not in terms of characters only. I did not figure out what the correct criteria are.

**Expected behaviour**
Only the lines in the files should be read.

**How to reproduce?**
The testbench below provides an example. It just outputs all lines of the also provided file. Note that the testbench outputs an empty line that is not in the file.

```vhd :file: tb_ghdl_test.vhd
library std;
use std.textio.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_ghdl_test is
end entity tb_ghdl_test;

architecture tb of tb_ghdl_test is
begin
  p_test : process is
    file f_in : text;
    variable v_line     : line;
  begin
    file_open(f_in, "ghdl_test.csv", read_mode);

    while not endfile(f_in) loop
      readline(f_in, v_line);
      report v_line.all;
    end loop;

    wait;
  end process;
end architecture;
```

[ghdl_test.txt](https://github.com/ghdl/ghdl/files/4239215/ghdl_test.txt)

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a tb_ghdl_test.vhd
ghdl --elab-run tb_ghdl-test
```

**Context**
Please, provide the following information:

- OS: debian bullseye/sid
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `1f466e4d2ed515d426edfd9c5c3dcf2ed4cc25a8`

**Additional context**
Expected output:
```
tb_ghdl_test.vhd:21:7:@0ms:(report note): 114,-76,-317,-192,-243,261,99,37,10,-267,-429,-77,133,-117,-760,-363,-294,586,179,-62,-414,-188,-196,351,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -141,-91,-22,441,448,180,-38,-141,-331,211,214,633,-268,-77,207,1347,999,-216,-241,-206,-62,687,664,74,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -92,161,541,265,507,-808,-36,173,813,-509,-59,-682,17,545,1826,-875,-893,-1493,-38,308,380,-571,-341,-482,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 181,88,676,-1484,-1635,-413,25,-83,-1470,235,461,876,500,-217,-675,-1357,-1164,1060,-213,8,-18,1511,1070,-455,1,2,1,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -278,154,2919,650,-35,-2073,-190,-389,-2376,-110,550,2684,-58,-51,856,1775,860,-395,-848,-813,-1922,3408,3586,1359,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 86,578,1208,-1260,-635,-1180,-338,488,1747,642,1115,-2076,-171,855,3564,-1449,-1067,-3147,-90,908,2039,-1723,-1117,-1933,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 467,-198,-806,-1601,-1615,907,332,177,-266,-652,-1178,-91,630,-361,-2452,-1816,-1546,2127,333,-226,-1315,484,219,678,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -123,-185,-735,561,741,788,15,-154,-48,323,199,468,-427,-146,64,1835,1469,-203,-187,-129,377,313,217,-171,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -80,92,508,404,401,-661,-35,61,330,-363,-83,-252,32,332,1070,-463,-522,-898,-9,114,-122,-372,-153,88,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 87,29,118,-570,-630,6,36,25,-284,21,-47,78,156,-150,-650,-415,-271,633,46,-49,-96,312,214,-45,1,2,0,0,0
```

Broken output:
```
tb_ghdl_test.vhd:21:7:@0ms:(report note): 114,-76,-317,-192,-243,261,99,37,10,-267,-429,-77,133,-117,-760,-363,-294,586,179,-62,-414,-188,-196,351,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -141,-91,-22,441,448,180,-38,-141,-331,211,214,633,-268,-77,207,1347,999,-216,-241,-206,-62,687,664,74,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -92,161,541,265,507,-808,-36,173,813,-509,-59,-682,17,545,1826,-875,-893,-1493,-38,308,380,-571,-341,-482,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 181,88,676,-1484,-1635,-413,25,-83,-1470,235,461,876,500,-217,-675,-1357,-1164,1060,-213,8,-18,1511,1070,-455,1,2,1,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -278,154,2919,650,-35,-2073,-190,-389,-2376,-110,550,2684,-58,-51,856,1775,860,-395,-848,-813,-1922,3408,3586,1359,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 86,578,1208,-1260,-635,-1180,-338,488,1747,642,1115,-2076,-171,855,3564,-1449,-1067,-3147,-90,908,2039,-1723,-1117,-1933,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 467,-198,-806,-1601,-1615,907,332,177,-266,-652,-1178,-91,630,-361,-2452,-1816,-1546,2127,333,-226,-1315,484,219,678,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note):
tb_ghdl_test.vhd:21:7:@0ms:(report note): -123,-185,-735,561,741,788,15,-154,-48,323,199,468,-427,-146,64,1835,1469,-203,-187,-129,377,313,217,-171,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): -80,92,508,404,401,-661,-35,61,330,-363,-83,-252,32,332,1070,-463,-522,-898,-9,114,-122,-372,-153,88,1,2,0,0,0
tb_ghdl_test.vhd:21:7:@0ms:(report note): 87,29,118,-570,-630,6,36,25,-284,21,-47,78,156,-150,-650,-415,-271,633,46,-49,-96,312,214,-45,1,2,0,0,0
```

eschmidscs, 2020-02-22:
How does that work with the MWE? Can I check if that is working?
I guess the attached file is not found?

diogratia, 2020-02-24:
I'll skip ahead and  supply the answer. When a 128 character fetch with a procedure call to `untruncated_text_read ` in procedure `std.textio.readline` has a right bound straddling a CR+LF pair the resulting empty buffer is appended to line `old_l`.

The save to `old_l` needs to be protected against `len = '0' after the LF feed character is deleted.

This was found by instrumenting a copy of `readline` then protecting against appending a 0 length buffer:

```vhdl
use std.textio.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_ghdl_test is
end entity tb_ghdl_test;

architecture tb of tb_ghdl_test is
    procedure untruncated_text_read                              --!V87
      (file f : text; str : out string; len : out natural);      --!V87

    -- attribute foreign of untruncated_text_read : procedure is "GHDL intrinsic";
    
    attribute foreign of untruncated_text_read : procedure is
    "VHPIDIRECT std__textio__untruncated_text_read";

    procedure untruncated_text_read
    --    (variable f : text; str : out string; len : out natural) is    --V87
      (file f : text; str : out string; len : out natural) is    --!V87
    begin
      assert false report "must not be called" severity failure;
    end untruncated_text_read;
    procedure readline (file f: text; l: inout line) --!V87
      is
        variable len, nlen, posn : natural;
        variable nl, old_l : line;
        variable str : string (1 to 128);
        variable is_eol : boolean;
      begin
        --  LRM93 14.3
        --  If parameter L contains a non-null access value at the start of the
        --  call, the object designated by that value is deallocated before the
        --  new object is created.

        if l /= null then
          deallocate (l);
         --  report "l deallocated.";
        end if;

        --  End of file is not expected.  The user should check endfile before
        --  calling readline.
        assert not endfile (f)
          report "eof in std.textio.readline" severity failure;

        -- We read the input in 128-byte chunks.
        -- We keep reading until we reach a newline or there is no more input.
        -- The loop invariant is that old_l is allocated and contains the
        -- previous chunks read, and posn = old_l.all'length.
        posn := 0;
        loop
          untruncated_text_read (f, str, len);
          -- report "str len = " & integer'image(len);
          exit when len = 0;  -- ORIGINAL
          if str (len) = LF or str (len) = CR then
            --  LRM 14.3
            --  The representation of the line does not contain the representation
            --  of the end of the line.
            is_eol := true;
            len := len - 1;
            --  End of line is any of LF/CR/CR+LF/LF+CR.
            if len > 0 and (str (len) = LF or str (len) = CR) then
[... 49 lines cut ...]
/usr/local/bin/ghdl -r  tb_ghdl_test
```
produces:

```console
tb_ghdl_test.vhdl:98:7:@0ms:(report note): 114,-76,-317,-192,-243,261,99,37,10,-267,-429,-77,133,-117,-760,-363,-294,586,179,-62,-414,-188,-196,351,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): -141,-91,-22,441,448,180,-38,-141,-331,211,214,633,-268,-77,207,1347,999,-216,-241,-206,-62,687,664,74,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): -92,161,541,265,507,-808,-36,173,813,-509,-59,-682,17,545,1826,-875,-893,-1493,-38,308,380,-571,-341,-482,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): 181,88,676,-1484,-1635,-413,25,-83,-1470,235,461,876,500,-217,-675,-1357,-1164,1060,-213,8,-18,1511,1070,-455,1,2,1,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): -278,154,2919,650,-35,-2073,-190,-389,-2376,-110,550,2684,-58,-51,856,1775,860,-395,-848,-813,-1922,3408,3586,1359,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): 86,578,1208,-1260,-635,-1180,-338,488,1747,642,1115,-2076,-171,855,3564,-1449,-1067,-3147,-90,908,2039,-1723,-1117,-1933,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): 467,-198,-806,-1601,-1615,907,332,177,-266,-652,-1178,-91,630,-361,-2452,-1816,-1546,2127,333,-226,-1315,484,219,678,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): -123,-185,-735,561,741,788,15,-154,-48,323,199,468,-427,-146,64,1835,1469,-203,-187,-129,377,313,217,-171,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): -80,92,508,404,401,-661,-35,61,330,-363,-83,-252,32,332,1070,-463,-522,-898,-9,114,-122,-372,-153,88,1,2,0,0,0
tb_ghdl_test.vhdl:98:7:@0ms:(report note): 87,29,118,-570,-630,6,36,25,-284,21,-47,78,156,-150,-650,-415,-271,633,46,-49,-96,312,214,-45,1,2,0,0,0
```
with no zero length buffer showing up.

Note this example was done on a MacOS llvm back end version of ghdl installed in PREFIX /usr/local.

The idea was to use a local copy of the procedure `readline` that could be modified. Because procedure `untruncated_text_read` is only found in the textio package body it was replicated here.

It was quite an adventure discovering how to link `untruncated_text_read` to `std__textio__untruncated_text_read`. The essential element is searching for the word _untruncated_ in directory `ghdl/src/grt`.

(And yes, I use the .vhdl suffix for VHDL files. The red reverse video block is an artifact of the highlighting used by github for VHDL.)

tgingold, 2020-02-25:
@diogratia Your change points out the issue but it will skip empty lines.  I think the end-of-line should be handled in untruncated_text_read.

diogratia, 2020-02-25:
>Your change points out the issue but it will skip empty lines.

It is possible to detect and correctly handle a CR+LF straddling the boundary between two successive calls to untruncated_text_read here. Doing both CR+LF and LF+CR requires knowledge of the order of the two format effectors.  (Single format effector end of line representations don't have this problem.) 

A Fix for CR+LF straddling:
```vhdl
    procedure untruncated_text_read
    --    (variable f : text; str : out string; len : out natural) is    --V87
      (file f : text; str : out string; len : out natural) is    --!V87
    begin
      assert false report "must not be called" severity failure;
    end untruncated_text_read;
    procedure readline (file f: text; l: inout line) --!V87
      is
        variable len, nlen, posn : natural;
        variable nl, old_l : line;
        variable str : string (1 to 128);
        variable is_eol : boolean;
        variable cr_last:   boolean;  -- ADDED
      begin
        --  LRM93 14.3
        --  If parameter L contains a non-null access value at the start of the
        --  call, the object designated by that value is deallocated before the
        --  new object is created.

        if l /= null then
          deallocate (l);
        end if;

        --  End of file is not expected.  The user should check endfile before
        --  calling readline.
        assert not endfile (f)
          report "eof in std.textio.readline" severity failure;

        -- We read the input in 128-byte chunks.
        -- We keep reading until we reach a newline or there is no more input.
        -- The loop invariant is that old_l is allocated and contains the
        -- previous chunks read, and posn = old_l.all'length.
        posn := 0;
        loop
          untruncated_text_read (f, str, len);
          exit when len = 0;
          cr_last := false;                            -- ADDED
          if len > 0 and str (len) = CR then
              cr_last := true;
          end if;                                      -- END ADDED
          if str (len) = LF or str (len) = CR then
            --  LRM 14.3
            --  The representation of the line does not contain the representation
            --  of the end of the line.
            is_eol := true and not cr_last;  -- ADDED cr_last qunalification
            len := len - 1;
            --  End of line is any of LF/CR/CR+LF/LF+CR.
            if len > 0 and (str (len) = LF or str (len) = CR) then
              len := len - 1;
            end if;
          elsif endfile (f) then
            is_eol := true;               -- cr_last qualification not needed
          else
            is_eol := false;
          end if;
            l := new string (1 to posn + len);
            if old_l /= null then
              l (1 to posn) := old_l (1 to posn);
              deallocate (old_l);
            end if;
            l (posn + 1 to posn + len) := str (1 to len);
            exit when is_eol;
            posn := posn + len;
            old_l := l;
        end loop;
      end readline;
```
This is done without memory across calls to procedure readline by suppressing  is_eol when the last character of str is a CR, delaying the exit until the end of line is completely removed.

```console
/usr/local/bin/ghdl -r  tb_ghdl_test
tb_ghdl_test.vhdl:96:7:@0ms:(report note): 114,-76,-317,-192,-243,261,99,37,10,-267,-429,-77,133,-117,-760,-363,-294,586,179,-62,-414,-188,-196,351,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): -141,-91,-22,441,448,180,-38,-141,-331,211,214,633,-268,-77,207,1347,999,-216,-241,-206,-62,687,664,74,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): -92,161,541,265,507,-808,-36,173,813,-509,-59,-682,17,545,1826,-875,-893,-1493,-38,308,380,-571,-341,-482,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): 181,88,676,-1484,-1635,-413,25,-83,-1470,235,461,876,500,-217,-675,-1357,-1164,1060,-213,8,-18,1511,1070,-455,1,2,1,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): -278,154,2919,650,-35,-2073,-190,-389,-2376,-110,550,2684,-58,-51,856,1775,860,-395,-848,-813,-1922,3408,3586,1359,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): 86,578,1208,-1260,-635,-1180,-338,488,1747,642,1115,-2076,-171,855,3564,-1449,-1067,-3147,-90,908,2039,-1723,-1117,-1933,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): 467,-198,-806,-1601,-1615,907,332,177,-266,-652,-1178,-91,630,-361,-2452,-1816,-1546,2127,333,-226,-1315,484,219,678,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): -123,-185,-735,561,741,788,15,-154,-48,323,199,468,-427,-146,64,1835,1469,-203,-187,-129,377,313,217,-171,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): -80,92,508,404,401,-661,-35,61,330,-363,-83,-252,32,332,1070,-463,-522,-898,-9,114,-122,-372,-153,88,1,2,0,0,0
tb_ghdl_test.vhdl:96:7:@0ms:(report note): 87,29,118,-570,-630,6,36,25,-284,21,-47,78,156,-150,-650,-415,-271,633,46,-49,-96,312,214,-45,1,2,0,0,0
```

If you conveyed the representation of end of line in textio-body.vhdl  you could replace the CR evaluation with the first character of end of line representation or single character non end of line representation, there'd only be two values for the four modes:
```vhdl
            --  End of line is any of LF/CR/CR+LF/LF+CR.
```

Which of the two could be tied to OS (platform).  There are only two of real interest and the code above covers them both already.  You could hide the detail in simulation kernel (grt) space by doing it in untruncated_text_read.

You have to admit the ghdl_test.txt file makes a great test article.

closed 2020-02-25 by tgingold

tgingold, 2020-02-26:
That works.  But at the end, I have modified untruncated_text_read to handle end of lines.  That simplifies the vhdl code.

Tristan.

diogratia, 2020-02-26:
I saw the switch to fgetc. Gives more freedom In fixing the problem. You still have unused line access nl and natural nlen. I was wondering about LF+CR end of line platforms. Someone  was talking about ghdl on a Raspberry Pi. I take it that would have been Linux.

commit 83449709 (2020-02-25): vhdl: handle CR+LF for readline in grt.  Fix #1145

Previously CR+LF was handled in std.textio.readline.  But that
doesn't work if CR is at position 128 because we would need to read the
next character.  Now untruncated_text_read handles CR/CR+LF/LF and
calls ungetc if needed.