ghdl/ghdl #1148: cannot handle IIR_PREDEFINED_UNIVERSAL_R_I_DIV

stevenbell, 2020-03-02:
I'm trying to convert a time delta into a frequency as a real (rather than an integer) using the following bit of code:

    report "Resulting frequency is: " & to_string(1.0 / (dt / 1000 ms)) & " Hz";

This causes GHDL to choke with the error

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/tmp/vhdlweb/TuKVVrmksRHHaFCGfeZNVkhTF/tone440/testing/
Command line:
/home/steven/tools/ghdl/ghdl_mcode -e --std=08 mwe
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:39
Call stack traceback locations:
0x5628764b0287 0x5628766049a3 0x56287660a4cb 0x562876602fe3 0x56287660a638 0x5628766003d0 0x5628765ffde8 0x5628765fff20 0x5628765ffd04 0x5628765fff13 0x5628765ffe71 0x5628765ff1a7 0x562876602b9f 0x56287660a4cb 0x5628765c8161 0x5628765c83ab 0x5628765d7b6a 0x5628765d7ce7 0x5628765db13b 0x5628765dfa62 0x56287664b550 0x5628765b6d91 0x5628765b3ea3 0x5628765c1c40 0x56287664fc9d 0x562876598d92 0x56287652c62a 0x562876653448 0x562876339ee7 0x7f4afff7cb95 0x562876338a28 0xfffffffffffffffe
******************************************************************
```
Ubuntu 18.04, GHDL built from source release 0.37.

MWE below:

```vhd :file: testbench.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use std.textio.all;

entity mwe is
end mwe;

architecture test of mwe is

begin

  process
    variable starttime : time := 1120 us;
    variable endtime : time := 2031 us;
    variable dt : time;

  begin
    dt := endtime - starttime;
    report "Resulting frequency is: " & to_string(1.0 / (dt / 1000 ms)) & " Hz";

    wait;
  end process;
end test;

```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 testbench.vhd
ghdl -e --std=08 mwe
ghdl -r --std=08 mwe --stop-time=10ms
```

stevenbell, 2020-03-02:
The workaround is to just use a type cast:

    to_string(1.0e6 / real(dt / 1 us)

tgingold, 2020-03-02:
The original crash is fixed, but the process is not synthesizable as is.

commit 319c6380 (2020-03-02): synth: fix conversion of net to constant for physical values.  Fix #1148
closed 2020-03-02 by tgingold