ghdl/ghdl #1149: Allow excluding signals in wave options file

Xiretza, 2020-03-03:
**Is your feature request related to a problem? Please describe.**
With issues in the GHW generator like #1067, as well as just not wanting huge signals in a wave dump, it would be useful to exclude exactly some signals from the output waveform using `--read-wave-opt=`. However the current wave options file format only allows *adding* signals.

**Describe the solution you'd like**
A syntax like in `gitignore(5)`, where a line starting with `!` *removes* some signal(s) that were previously added.

**Describe alternatives you've considered**
Manually adding all but one signal in a module works, but is tedious and needs to be updated whenever any signals are added or their names change.

tgingold, 2020-03-04:
There is certainly work to do on wave dumps: handling unbounded types, better filters, better compression...

bahm-lgs, 2021-05-04:
I have just started using ghdl and yes, this feature is something I was also missing already. I like the idea of having a gitignore-style syntax. However, my first idea was: it would also be nice, being able to exclude a whole component with simply a VHDL attribute in the sources...

umarcor, 2021-05-04:
Ref https://github.com/ghdl/ghdl/issues/1281#issuecomment-623308860.

JumpyLionnn, 2026-03-12:
are there any updates on this issue? I think it would be an extremely useful feature

tgingold, 2026-03-13:
There is `--read-wave-opt` and `--write-wave-opt`.