ghdl/ghdl #1156: Use GHDL as a VHDL parser

jagjordi, 2020-03-09:
**Is your feature request related to a problem? Please describe.**
I am wondering if it is possible to use GHDL to parse a file and analyze its content and exportit to some JSON/YAML/XML or whatever format so it can be read by other tools
**Describe the solution you'd like**
Lets say I have a file that is describing an entity, I would like to then parse it with GHDL and get a file that lists things like ports, generics, processes, etc.
**Describe alternatives you've considered**
There are other tools that can do this, pyVHDLParser, VHDLParser, hdlparser... but I have not been able to sucssesfuly use it as I want.
**Additional context**
Since GHDL is internaly parsing the vhdl files to generate the design model and synthesise it so I guess it would be somehow easy to implement (if it does not exist yet) the feature I descibed

eine, 2020-03-10:
Hi @jagjordi! Have a look at the discussion in jeremiah-c-leary/vhdl-style-guide#312. See also #111 and https://ghdl.readthedocs.io/en/latest/internals/AST.html.

tgingold, 2020-03-10:
There are two solutions with ghdl:

* you could use --file-to-xml, but I plan to deprecate it.

* You can use the python interface.  In testsuite/python/units01, there is an example on how to dump design units name.

hackfin, 2020-03-10:
Will the XML feature remain anyway? I have quite some dependencies on it..
See also https://hackfin.gitlab.io/xhdl/, this is one example on how to convert the GHDL XML output to SVG, etc.

jagjordi, 2020-03-10:
Thank you @tgingold I found the units01 example very usefull!. Is there a exhaustive documentation for libghdl somewhere? I was not able to find any.

tgingold, 2020-03-10:
Currently there is no plan to remove it.  You have nice examples of how to use it!

tgingold, 2020-03-10:
The python is a think interface to the code, which is documented in the .ads files.

The main file is vhdl-nodes.ads

There is also a few paragraph in the ghdl manual.

closed 2020-03-30 by tgingold

pidgeon777, 2020-11-22:
> Will the XML feature remain anyway? I have quite some dependencies on it..
See also https://hackfin.gitlab.io/xhdl/, this is one example on how to convert the GHDL XML output to SVG, etc.

@hackfin I tried your webpage but I don't know how to generate a schematic starting from the XML output file. Also something seems to be broken in your webpage because no schematic is shown.

I read that **File API & FileReader available**. Is there a way to obtain them?

Also, you might be interested in this discussion:

https://github.com/ghdl/ghdl/issues/1519

umarcor, 2021-01-20:
> Also something seems to be broken in your webpage because no schematic is shown.

@pidgeon777 it seems to work for me. Just opening that URL:

[image]

umarcor, 2023-06-22:
@Nabav are you asking about `testsuite/python/units01`?

Nabav, 2023-06-22:
Exactly

umarcor, 2023-06-22:
The Python interface evolved into https://github.com/ghdl/ghdl/tree/master/pyGHDL (which can be installed as explained in https://ghdl.github.io/ghdl/quick_start/python/index.html) and the corresponding tests are located in https://github.com/ghdl/ghdl/tree/master/testsuite/pyunit.

Nowadays, the closest example to the file you are looking for would be the README of pyVHDLModel: https://github.com/vhdl/pyVHDLModel#list-all-entities-with-generics-and-ports. That's the foundation of https://umarcor.github.io/osvb/apis/project/OSVDE.html, which is a single Python script combining pyGHDL-pyVHDLModel-tkinter: https://github.com/umarcor/osvb/blob/main/mods/pyOSVDE/main.py. OSVDE is a proof of concept to showcase the capabilities of pyGHDL and pyVHDLModel. The same foundation is to be used in https://github.com/VHDL/VHDLDomain.

For completeness, if you really want to read/see what the `units01` was, check https://github.com/ghdl/ghdl/pull/1449. That's the most meaningful dialogue we had since I tried to use/update the Python interface in the second half of 2020, until @Paebbels stepped in and pyGHDL was created in January 2021. `units01` contained two scripts:

- https://github.com/umarcor/ghdl/blob/73184decbb9ef0159e3b1bb586d158112a7a027b/testsuite/python/units01/show_ports.py
- https://github.com/umarcor/ghdl/blob/73184decbb9ef0159e3b1bb586d158112a7a027b/testsuite/python/units01/show_units.py

So, those scripts use(d) pyGHDL.libghdl directly, while the "pythonic" approach is to use pyGHDL.dom.