ghdl/ghdl #1162: synth: SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1340

bwimm, 2020-03-19:
**Description**
A zero bit counter fails to synthesize

**How to reproduce?**

```vhd :file: issue.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity counter is
	generic(
		LEN    : positive := 1
	);
	port(
		clk           : in std_ulogic;
		reset_n       : in std_ulogic
	);
end counter;

architecture behav of counter is
	signal c : integer range 0 to LEN-1;
begin
	process(clk, reset_n)
	begin
		if reset_n = '0' then
			c  <= 0;
		elsif rising_edge(clk) then
			if c = LEN-1 then
				c <= 0;
			else
				c <= c + 1;
			end if;
		end if;
	end process;
end architecture;
```

```
ghdl --synth --std=08 issue.vhdl -e counter
```

**Context**

- OS: Ubuntu 19.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `2d50c68e`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-111-g2d50c68e) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/home/user/Desktop/tmp/
Command line:
ghdl --synth --std=08 issue.vhd -e counter
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1340
Call stack traceback locations:
0x7f54284a0521 0x559ca2bb11af 0x559ca2c3696f 0x559ca2c36f8f 0x559ca2c373c9 0x559ca2c3a3ad 0x559ca2c3a670 0x559ca2d25967 0x559ca2d25b0d 0x559ca2d1629e 0x559ca2d163ca 0x559ca2d0fa26 0x559ca2d71b67 0x559ca2cb51e3 0x559ca2e1c6fb 0x559ca2a7b98d 0x7f542806ab69 0x559ca2a7a1a8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Setting the generic LEN to 2 does not fail.

eine, 2020-03-19:
@bwimm, what do you expect to be synthesised for/from `c`?

bwimm, 2020-03-19:
In this case probably nothing.
In the original code it's used in other processes where I would expect it to turn into a constant/wire.

closed 2020-03-19 by tgingold

eine, 2020-03-19:
Should it be turned into a constant/wire, which value would you expect it to hold? Note that, AFAIK, this is currently undefined behaviour in the LRM. Context: http://grouper.ieee.org/groups/1076/email/msg01110.html

bwimm, 2020-03-19:
@eine Interesting, I didn't know.
tbh. I expected something like ```signal c : integer range A to A;``` to turn into a constant A and not 0.
I'll also have to double check my code why it's a null range at all...

commit 804e222c (2020-03-19): netlists-builders: allow null net for all dffs.  Fix #1162