ghdl/ghdl #1168: synth: unhandled call to ieee function "minimum"

bwimm, 2020-03-21:
**Description**
"minimum()" is not synthesized

**How to reproduce?**

```vhd :file: issue.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
	generic(
		LEN : positive := 32
	);
	port(
		input_a :  in unsigned(LEN-1 downto 0);
		input_b :  in unsigned(LEN-1 downto 0);
		output  : out unsigned(LEN-1 downto 0)
	);
end bug;

architecture behav of bug is
begin
	output <= minimum(input_a, input_b);
end architecture;
```

```
ghdl --synth --std=08 issue.vhdl -e bug
```

**Output**
```
issue.vhd:18:18:error: unhandled call to ieee function "minimum"
issue.vhd:12:16:warning: no assignment for port "output"
```

**Context**
Please, provide the following information:

- OS: Ubuntu 19.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `28cf5741`

bwimm, 2020-03-21:
Still trying to get my project through ghdl and I really appreciate how quickly my issues were fixed, thanks :)

tgingold, 2020-03-21:
Thank you for all the issues.  I try to do my best.  Some issues aren't very difficult, some others require more work.

Right now, min/max aren't handled by the yosys plugin.

closed 2020-03-21 by tgingold

Xiretza, 2020-03-21:
I can add them to ghdlsynth if you haven't done it already.

tgingold, 2020-03-21:
Yes, that would be very nice!

commit 134c6ed9 (2020-03-21): synth: handle numeric_std minimum/maximum.  Fix #1168