ghdl/ghdl #1174: Synthesis output formats (generators): VHDL, EDIF, Verilog...

eine, 2020-03-27:
One of the new features of GHDL (which I am most excited about) is the "implicit standard version downgrade" that is achieved with `ghdl --synth`. I.e., we can take any modern and complex VHDL codebase with 2008 features and GHDL will provide a plain old 1993 netlist. Vendor tools with poor VHDL support will probably choke on 2008 sources but they will gracefully accept the 1993 netlist.

However, in essence, the VHDL netlist generated by `ghdl --synth` is equivalent to Yosys' C API (which is used by ghdlsynth-beta). Hence, there are currently two synthesis generators/elaborators/backends. Some other tools in the EDA industry might benefit from additional output formats. This issue is to gather ideas and discuss such other formats:

# Verilog

I believe it is already possible to generate Verilog from Yosys' output. However, Yosys' does not only do synthesis, but also technology mapping.

- Would it make sense to generate some Verilog sources which are equivalent to Yosys' input instead of the output?
- What's the use case for desiring synthesised but not technology mapped Verilog sources? The use case for VHDL is poor language support.
- Is it worth creating en explicit backend instead of using some external tool to convert the current output (VHDL netlist)?
- Would this make it easier to merge VHDL and Verilog sources through components/black boxes?

# EDIF

AFAIK, there is no "standard" EDIF. I think it would be great to generate an EDIF which is generic enough to be accepted by any open source or vendor tool. But maybe that's not possible and GHDL would need to generate *flavours* of EDIF. See https://j.mp/openfpga-diagram. OTOH, the generation of EDIF flavours from a single source is a feature that can be externalised (to be shared with other FOSS projects).

Ref #957, #908

anhangaba, 2020-03-28:
Maybe GHDL could generate [FIRRTL](https://github.com/freechipsproject/firrtl/)?

eine, 2020-03-28:
Last time I checked (see SymbiFlow/ideas#19), FIRRTL was tailored towards the subset of hardware designs that is targeted by Chisel, which is apparently much narrower than the type of designs that can be described in VHDL. Another project with a similar objective is [fabianschuiki/llhd](https://github.com/fabianschuiki/llhd) (written in Rust).

@anhangaba, would you mind providing brief discussions about advantages/disadvantages of targeting FIRRTL and/or btor2? I'll be happy to copy them to the first message in this issue.

tgingold, 2020-03-29:
From what I understand, EDIF is too low-level (there is no portable way to represent a memory).  EDIF is really a netlist format, without predefined cells.

RTLIL looks higher level and I think it could be generated by ghdl synth. Would that be useful ?  I am not sure.

tmeissner, 2020-03-29:
BTOR2 can be generated using Yosys, it uses that format for boolector. I don't know if it is useful to implement that in GHDL.

anhangaba, 2020-03-30:
I was just throwing suggestions, I don't have a dog in that fight.

I reckon it is already possible to produce most netlists format using the synth output and the python interface. I think more examples of the python interface would be more useful than new outputs formats (maybe a synth output AST to FIRRTL-lo?).

About the BTOR2, it could be used to integrate formal verification to GHDL simulation.

alemuller, 2020-09-28:
# How to convert vhdl to other formats

I made a list of "How to convert vhdl to ..." in hopes it gets indexed by the search engines. All commands are similar, they are listed so common search phrases get indexed.

The format translation can be done with the yosys's `write_*`commands. Therefore, in order to them to work,  [ghdl](https://github.com/ghdl/ghdl), [yosys](https://github.com/YosysHQ/yosys) and [ghdl-yosys-plugin](https://github.com/ghdl/ghdl-yosys-plugin) have to be installed (make sure they are updated).

In the following examples, ghdl is being used, in what I believe, is the most straightforward way. See [ghdl-yosys-plugin#usage](https://github.com/ghdl/ghdl-yosys-plugin#usage) for a workflow more adequate for larger projects.

### Convert VHDL to verilog:
```
 yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_verilog filename.v'
```
### Convert VHDL to EDIF:
```
 yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_edif filename.edif'
```
### Convert VHDL to SMT (SMT-LIBv2)
```
 yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_smt2 filename.smt2'
```
### Convert VHDL to BTOR2:
```
 yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_btor filename.btor'
```
### Convert VHDL to FIRRTL:
```
 yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_firrtl filename.firrtl'
```

For a comprehensive list of supported output formats, check out the [yosys documentation](http://www.clifford.at/yosys/documentation.html).  They included aiger, blif, ilang, intersynth, json, c, smv, spice.

eine, 2020-09-28:
@alemuller, that looks awesome! Would you like to add it (or have it added) to https://ghdl.github.io/ghdl/using/Synthesis.html#yosys-plugin?

Xiretza, 2020-09-28:
@alemuller, how about also posting this as a self-answer on [EE SE](https://electronics.stackexchange.com/) or [stackoverflow](https://stackoverflow.com/)? Those are usually indexed and presented pretty well by several search engines.

alemuller, 2020-09-28:
Thanks for the suggestions, I will do both latter.

eine, 2020-10-09:
Yosys' `write_*` commands are now explained in https://ghdl.github.io/ghdl/using/Synthesis.html#convert-v-hdl-to-other-formats.

Nevertheless, note that the purpose of this issue was to discuss output formats generated by GHDL only, not through Yosys.

closed 2021-09-14 by tgingold

umarcor, 2021-09-29:
FTR, https://chipsalliance.org/blog/2021/09/09/symbiflow-fpga-interchange-format-to-enable-interoperable-fpga-tooling/.