ghdl/ghdl #1180: synth: synth_expression_with_type: cannot handle IIR_KIND_LOW_TYPE_ATTRIBUTE

bwimm, 2020-03-30:
**Description**
Using a generic as the range limit fails to synthesize.
Replacing C_HIGH with 5 in the subtype definition makes it synthesize.

**Expected behaviour**
Successful synthesis

**How to reproduce?**

```vhd :file: issue.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
	generic(
		C_HIGH : natural := 5
	);
	port(
		reset_n : in std_ulogic;
		clk     : in std_ulogic
	);
end bug;

architecture behav of bug is
	subtype st is natural range 0 to C_HIGH;
	signal c : st;
begin
	process(clk, reset_n)
	begin
		if reset_n = '0' then
			c <= st'low;
		elsif rising_edge(clk) then
			if c = st'high then
				c <= st'low;
			else
				c <= c + 1;
			end if;
		end if;
	end process;
end architecture;
```

```
ghdl --synth --std=08 issue.vhdl -e bug
```

Output:
```
synth_expression_with_type: cannot handle IIR_KIND_LOW_TYPE_ATTRIBUTE (issue.vhd:22:32)
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `42b91517`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-194-g42b91517) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/home/user/Desktop/tmp/
Command line:
ghdl --synth --std=08 issue.vhd -e bug
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x55ff05f75d92 0x55ff06099c76 0x55ff0606d56c 0x55ff06070b11 0x55ff0607ad0f 0x55ff060711a2 0x55ff0607acf2 0x55ff0607b5d5 0x55ff0607cf9f 0x55ff0607d159 0x55ff0606cb7d 0x55ff0606ccce 0x55ff06066626 0x55ff060cb6fb 0x55ff0600bb19 0x55ff0617628f 0x55ff05dcf9a3 0x7f99d3486b69 0x55ff05dce1a8 0xfffffffffffffffe
******************************************************************
```

bwimm, 2020-03-30:
Probably a duplicate of issue #1098

tgingold, 2020-03-30:
Yes, but with a testcase!

commit 9261f29d (2020-04-01): synth: handle low/high type attributes.  Fix #1180
closed 2020-04-01 by tgingold