ghdl/ghdl #1181: synth: synth-static_oper.adb:605 discriminant check failed

bwimm, 2020-03-30:
**Description**
using a variable after assignment fails to synthesize.

**Expected behaviour**
Successful synthesis

**How to reproduce?**

```vhd :file: issue.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
	port(
		dummy       : in std_ulogic
	);
end bug;

architecture behav of bug is
begin
	process(all)
		variable index : integer;
	begin
		index := 10;
		if index > 3 then
		--	index := index-1;
		end if;
	end process;

end architecture;
```

```
ghdl --synth --std=08 issue.vhd -e bug
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `42b91517`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-194-g42b91517) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/home/user/Desktop/tmp/
Command line:
ghdl --synth --std=08 issue.vhd -e bug
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-static_oper.adb:605 discriminant check failed
Call stack traceback locations:
0x55f69ac47aad 0x55f69ac3d7d4 0x55f69ac56fc6 0x55f69ac57cc5 0x55f69ac2f00d 0x55f69ac38cf2 0x55f69ac395d5 0x55f69ac3af9f 0x55f69ac3b159 0x55f69ac2ab7d 0x55f69ac2acce 0x55f69ac24626 0x55f69ac896fb 0x55f69abc9b19 0x55f69ad3428f 0x55f69a98d9a3 0x7f7bc740eb69 0x55f69a98c1a8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Might be related to issue #1176

anhangaba, 2020-03-30:
That is exactly how I found #1176 in the real code, but with slv instead of integer.

tgingold, 2020-03-30:
Yes, the handling of static values is currently too simple.

commit d0af178e (2020-04-09): synth: use memtyp in synth-static_oper.  Fix #1181
closed 2020-04-09 by tgingold