ghdl/ghdl #1182: evaluation of expression defined with generics constant on aggregate

anhangaba, 2020-03-30:
The code works with `--synth` but not with `-a`. The code also works if `N` is defined in a constant declaration.

```vhd :file: issue.vhd
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    generic (N : natural := 16);
       port (foo : out std_logic_vector(2*N-1 downto 0));
end issue;

architecture rtl of issue is
    signal bar : std_logic_vector(N-1 downto 0);
begin
    foo <= (bar,others=>'0');
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 issue.vhd
```

```
eval_pos: cannot handle IIR_KIND_SUBSTRACTION_OPERATOR (issue.vhdl:10:36)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-194-g42b91517) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
***
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/ghdl1 --ghdl--std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -quiet -o issue.s issue.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x6c543a 0x6ed94e 0x6eddf1 0x7c0dbb 0x7c12a5 0x7c2161 0x7bc1eb 0x7c2c8d 0x78b785 0x7972df 0x797b95 0x797675 0x7a0fdc 0x788f8e 0x7650a5 0x756604 0x7541cc 0x664d1b 0xbc219f 0x65b453 0x669c0c 0x7ca6eb 0x659301 0x7f4e5a80e151 0x66024c 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/ghdl1 terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x6c543a 0x6ed94e 0x6eddf1 0x7c0dbb 0x7c12a5 0x7c2161 0x7bc1eb 0x7c2c8d 0x78b785 0x7972df 0x797b95 0x797675 0x7a0fdc 0x788f8e 0x7650a5 0x756604 0x7541cc 0x664d1b 0xbc219f 0x65b453 0x669c0c 0x7ca6eb 0x659301 0x7f4e5a80e151 0x66024c 0xfffffffffffffffe
ghdl:error: compilation error
```

anhangaba, 2020-04-16:
Another example that throws an error instead the Bug Report.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    generic (constant N : integer := 3);
    port (foo : in  std_logic;
          bar : out std_logic_vector(7 downto 0));
end issue;

architecture beh of issue is
begin
    bar <= (N=>foo, others=>'0');
end architecture;
```

It throws the error:
```
issue.vhdl:12:12:error: not static choice exclude others choice
issue.vhdl:12:12:error: non-locally static choice for an aggregate is allowed only if only choice
```
I this would fit the spec criterea:
> As the default expression for a generic constant declared to be of a fully constrained array subtype

eine, 2020-04-16:
@anhangaba, it seems that you can work around it as follows:

```vhd
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    generic ( constant N : integer := 3 );
    port (foo : in  std_logic;
          bar : out std_logic_vector(7 downto 0));
end issue;

architecture beh of issue is
  constant M : integer := 3;
begin
    bar <= (M=>foo, others=>'0');
end architecture;
```

tgingold, 2020-04-27:
There are two different issues in this issue.  The first one is the crash during analysis.

The second one is the diagnostic, which is correct.

anhangaba, 2020-04-27:
I've read the specification of static expression and now I understand why a generic constant isn't allowed (isn't a locally static primary).

anhangaba, 2020-05-04:
The original example don't crash anymore. The issue seems to be fixed.

closed 2020-05-04 by anhangaba