ghdl/ghdl #1188: Compiling Xilinx primitives - unisim

wtatarski, 2020-04-03:
**Description**
I'm trying to include precompiled Xilinx Vivado primitives from unisim. I'm getting the following errors:
```
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: no declaration for "pll_lockg" in package "vcomponents"
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: target is not a signal name
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1209:27:error: no declaration for "gsr"
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/IDELAYE2.vhd:198:30:error: no declaration for "gsr"
```

**Expected behaviour**
I expect that there should be no errors in Vivado source files relating to primitives.

**How to reproduce?**
Source code which I'm using is available here: https://github.com/hamsternz/Artix-7-HDMI-processing/tree/master/src

Xilinx primitives were precomplied with the command:
```
/usr/local/lib/ghdl/vendors/compile-xilinx-vivado.sh --all --src /media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/
```
and after that I'm trying to launch synthesis with precompiled primitives
```
$ ghdl --synth -Pxilinx-vivado/unisim/v93 --vendor-library=unisim -fsynopsys -fexplicit src/*.vhd -e hdmi_design
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: no declaration for "pll_lockg" in package "vcomponents"
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1186:28:error: target is not a signal name
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/MMCME2_ADV.vhd:1209:27:error: no declaration for "gsr"
/media/olddrive/home/wojtek/xilinx2018.3/Vivado/2018.3/data/vhdl/src/unisims/primitive/IDELAYE2.vhd:198:30:error: no declaration for "gsr"
```

When I remove `-Pxilinx-vivado/unisim/v93` from ghdl command I'm getting the following log:
```
$ ghdl --synth --vendor-library=unisim -fsynopsys -fexplicit src/*.vhd -e hdmi_design

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-105-gf7991678) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/media/newdrive/prjxray/Artix-7-HDMI-processing/
Command line:
ghdl --synth --vendor-library=unisim -fsynopsys -fexplicit src/alignment_detect.vhd src/audio_meters.vhd src/audio_to_db.vhd src/conversion_to_RGB.vhd src/deserialiser_1_to_10.vhd src/detect_interlace.vhd src/dvid_output.vhd src/edge_enhance.vhd src/edid_rom.vhd src/expand_422_to_444.vhd src/extract_audio_samples.vhd src/extract_video_infopacket_data.vhd src/guidelines.vhd src/hdmi_design.vhd src/hdmi_input.vhd src/hdmi_io.vhd src/input_channel.vhd src/line_delay.vhd src/pixel_processing.vhd src/serialiser_10_to_1.vhd src/symbol_dump.vhd src/tmds_decoder.vhd src/tmds_encoder.vhd -e hdmi_design
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : files_map.adb:83
Call stack traceback locations:
0x55f1b23337c1 0x55f1b2335d29 0x55f1b233fe1f 0x55f1b23de16c 0x55f1b2465db5 0x55f1b252d83e 0x55f1b252e6b5 0x55f1b2471edb 0x55f1b25d92bb 0x55f1b223890d 0x7f70cb0f8b95 0x55f1b2237128 0xfffffffffffffffe
******************************************************************

```

**Context**

- OS: Ubuntu 18.04
- Origin:
```
ghdl --version
GHDL 1.0-dev (v0.37.0-105-gf7991678) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.
```

eine, 2020-04-04:
@wtatarski, I think that the vendor libraries that you are trying to use are meant for simulation only:

> https://www.xilinx.com/support/answers/64052.html
> The UNISIM library is used in functional simulation and behavioral simulation when the RTL instantiates device primitives.

However, the command you used is `ghdl --synth`. What are you trying to achieve?

tgingold, 2020-04-04:
The declarations from vcomponents are needed to instantiate the primitives.  That part is OK.

I think for synthesis, only unisim_vcomp.vhd is needed (and maybe unisim_vpkg.vhd).
Then the analysis stops at components when --vendor-library=unisim is used: `ghdl --synth` doesn't try to instantiate the entities for the components.

tgingold, 2020-04-04:
Unfortunately, I am not able to reproduce the same crash.
I have just analyzed `unisim_VCOMP.vhd` (that's faster than analyzing the whole unisim and enough).

I got a first crash because in `edid_rom.vhd` the inout port `sdat_raw` has a default value.  This is not yet supported, so I commented out the default value.

Then I have:
```
 $GHDL --synth -P../../unisim -fsynopsys -fexplicit --vendor-library=unisim *.vhd -e
ghdl_mcode:note: top entity is "hdmi_design"
hdmi_design.vhd:64:9:warning: no assignment for port "hdmi_rx_cec"
hdmi_design.vhd:67:9:warning: no assignment for port "hdmi_rx_sda"
hdmi_design.vhd:75:9:warning: no assignment for port "hdmi_tx_cec"
hdmi_design.vhd:79:9:warning: no assignment for port "hdmi_tx_rscl"
hdmi_design.vhd:80:9:warning: no assignment for port "hdmi_tx_rsda"
hdmi_io.vhd:68:9:warning: no assignment for port "hdmi_rx_sda"
symbol_dump.vhd:72:12:warning: signal "hex" is never assigned
symbol_dump.vhd:100:12:warning: signal "baud_counter_max" is never assigned
symbol_dump.vhd:79:12:note: found RAM "memory", width: 30 bits, depth: 1024
symbol_dump.vhd:72:12:note: found ROM "hex", width: 10 bits, depth: 16
edid_rom.vhd:65:11:warning: signal "edid_rom" is never assigned
edid_rom.vhd:65:11:note: found ROM "edid_rom", width: 8 bits, depth: 256
tmds_encoder.vhd:97:25:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_EQ_SLV_INT
tmds_encoder.vhd:124:58:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_EQ_SLV_INT
tmds_encoder.vhd:64:12:warning: signal "data_word" is never assigned and has no default value
line_delay.vhd:84:12:note: found RAM "mem_4", width: 27 bits, depth: 512
line_delay.vhd:83:12:note: found RAM "mem_3", width: 27 bits, depth: 512
line_delay.vhd:82:12:note: found RAM "mem_2", width: 27 bits, depth: 512
line_delay.vhd:81:12:note: found RAM "mem_1", width: 27 bits, depth: 512
line_delay.vhd:80:12:note: found RAM "mem_0", width: 27 bits, depth: 512
```

There is one operation not yet supported too.

wtatarski, 2020-04-06:
@tgingold  Could you tell me how did you produce `../../unisim` directory? Also what is the content of `$GHDL` variable?

tgingold, 2020-04-06:
The `$GHDL` variable is the path of my ghdl executable.

For the unisim directory, create the directory and:
```
$GHDL -a --work=unisim unisim_VCOMP.vhd 
```

tgingold, 2020-04-24:
I close this issue as it has been answered.  Do not hesitate to re-open it if your issue is not solved.

closed 2020-04-24 by tgingold