ghdl/ghdl #1190: Accept file inclusion relative to the HDL file location

rodrigomelo9, 2020-04-03:
# Useful preface:

This request appears because I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I am now testing with ghdl-yosys-plugin (in the branch `ghdlsynth` of my repo). I solved the problem in Yosys (https://github.com/YosysHQ/yosys/pull/1670).

# The problem

When a file is included, its path is considered taken into account relative to the where GHDL is executed, but not relative to the HDL file location.

# Example

[issue.zip](https://github.com/ghdl/ghdl/files/4430354/issue.zip)

It has the following files:
* hdl/ram.vhdl: a memory which includes `data/memory.dat`.
* hdl/data/memory.dat: the initial content of the memory.
* run1.sh: to be executed where the `hdl` directory is located.
* hdl/run2.sh: to be executed into the `hdl` directory.

> NOTE: I am using the docker from ghdl-yosys-plugin.

If I execute `bash run2.sh` inside the `hdl` directory, it works. if I executed `bash run`.sh`it fails, because `data/memory.dat` is not found:
```
$ bash run.sh 
* Analysis with 'ghdl -a'
hdl/ram.vhdl:33:12:warning: declaration of "ram" hides entity "ram" [-Whide]
* Elaboration with 'ghdl -e'
* Running with 'ghdl -r'
ghdl:error: cannot open file "data/memory.dat"
  from: work.ram(rtl).init at ram.vhdl:23
ghdl:error: error during elaboration
* Synthesis with 'ghdl --synt'
hdl/ram.vhdl:22:18:error: cannot open file: data/memory.dat
...
```

# Suggested functionality

At least ISE, Vivado, Quartus, Libero-SoC, and now Yosys, check if the file exists in a location relative to where the HDL file was included. At least Yosys (I didn't check in other cases) first check relative to the execution directory and, if not found, check relative to the HDL file location.

I can use a workaround in my case, but the idea is to improve GHDL, being compatible with what other tools do (at least for Synthesis, which is what I am checking). What do you think about that? I know that is something with a very very very low priority but seems ok to be supported? (for the hypothetical case that I could try to implement it).

Don't hesitate to ask if I was no clear with my explanation. Regards

tgingold, 2020-04-04:
Humm, it makes sense.
I fear the feature you'd like to have is not very easy to describe, but in VHDL there is no way to include a file, just a way to open it.

rodrigomelo9, 2020-04-04:
Yes, you right, I was talking about open a file.

tgingold, 2020-04-04:
I think I have an idea on how to handle that, but only for synthesis.  Not sure about simulation.

rodrigomelo9, 2020-04-04:
I am interested in synthesis right now. In Yosys, the file open process is replaced with the file content formatted as a memory. I think that it is only useful for that in synthesis, provide the initial content of a memory.

rodrigomelo9, 2020-04-04:
More precisely, in Yosys `readmemh` and `readmemb` are supporting as described.

commit 0aac4efd (2020-04-13): synth: also try to open files (during synthesis) relative to current unit.

Fix #1190
closed 2020-04-13 by tgingold