ghdl/ghdl #1191: Crash in latest ghdl-gcc

csantosb, 2020-04-04:
**Description**
GHDL crashes when goind from first commit to second commit in git repo included in attached file.

**Expected behaviour**
Run correctly as in first commit, all ok.

**How to reproduce?**
Run runme.sh in attached file, including a git repo. First commit, all ok, runs with no issue. Second commit, bug, crashes.

[bug.tar.gz](https://github.com/ghdl/ghdl/files/4431991/bug.tar.gz)

**Context**
Please, provide the following information:

- OS: ArchLinux, gcc/ghdl-gcc-git 
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: c3b78f3370c1def22ff7240045d174fb120714e9

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-217-gc3b78f33) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/tmp/bug/
Command line:
/usr/bin/ghdl -m -o mux_fifo_tb -g -O2 --work=work --std=08 --ieee=synopsys -P/opt/CompiledLibraries/Vivado --mb-comments -fpsl -C -fexplicit -frelaxed-rules --no-vital-checks --warn-unused --warn-reserved --warn-parenthesis --warn-delayed-checks --warn-others --warn-pure --warn-static --work=work mux_fifo_tb
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Element_Subtype_Indication
Call stack traceback locations:
0x580bf2 0x46eef2 0x4f859a 0x4fb6b9 0x4fb96c 0x50d218 0x50f6e2 0x4ee23f 0x51a797 0x4e7d45 0x4e86a5 0x5476fc 0x54f1b9 0x55d612 0x4e3119 0x406ee1 0x7f08b6435021 0x406f5c 0xfffffffffffffffe
******************************************************************
```

**Additional context**

tgingold, 2020-04-07:
I haven't tested the whole design (I don't have the whole unisim), but at least the crash is gone.

Thank you for the issue.

closed 2020-04-07 by tgingold

csantosb, 2020-04-07:
Great, thanks !