ghdl/ghdl #1198: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT

rodrigomelo9, 2020-04-05:
I was [verifying GHDL `--synth` and the `ghdl-yosys-plugin` with examples from Xilinx](https://github.com/rodrigomelo9/verifying-foss-hdl-synthesizers). I found a problem with both of them, in an example (which works with ISE and Vivado), called  **for_loop.vhd**:
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.std_logic_unsigned.all;

entity for_loop is
 port(
  a     : in  std_logic_vector(7 downto 0);
  Count : out std_logic_vector(2 downto 0)
 );
end for_loop;

architecture behavior of for_loop is
begin
 process(a)
  variable Count_Aux : std_logic_vector(2 downto 0);
 begin
  Count_Aux := "000";
  for i in a'range loop
   if (a(i) = '0') then
    Count_Aux := Count_Aux + 1;
   end if;
  end loop;
  Count <= Count_Aux;
 end process;
end behavior;
```

```
$ docker run --rm -t -v $(pwd):/src -w /src ghdl/synth:beta ghdl --synth -fsynopsys examples/vivado/for_loop.vhd -e for_loop
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
examples/vivado/for_loop.vhd:25:28:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_INT
```

eine, 2020-04-05:
Might this be considered a up of #1193? One is about signed and the other one about unsigned.

rodrigomelo9, 2020-04-05:
Could be. I don't know if they will be solved together or separated. I can move the report there as a comment if you prefer and close it.

eine, 2020-04-05:
I think it's ok like this, because they don't need to be solved at the same time. I added the comment just for Tristan to be aware, in case he missed it.

tgingold, 2020-04-06:
I will soon push a fix for this issue.  The support of synopsys packages (std_logic_arith, std_logic_signed or std_logic_unsigned) is not as advanced as the standard numeric_std package.

Thank you for the test, keep going!

commit fa473ef1 (2020-04-06): synth-static_oper: add support for unsgined_add_slv_int.  Fix #1198
closed 2020-04-06 by tgingold