ghdl/ghdl #1201: Crash in latest ghdl-gcc/llvm with two successive non constrained arrays

csantosb, 2020-04-08:
**Description**
Ghdl crash with the provided code when `mysig_wrong` is comment out.

**Expected behaviour**
Should work with no issue, instead of crashing.

**How to reproduce?**
Run following code

```vhd :file: bug.vhd
library ieee;
use ieee.std_logic_1164.all;

entity bug is
  port (clk : in std_logic;
        rst : in std_logic);
end entity bug;

architecture Behavioral of bug is

  type rc_params is array (natural range <>) of std_logic_vector(7 downto 0);
  type rc_params_array is array (natural range <>) of rc_params;

  type rc_peripheral_1 is record
    clk    : std_logic;
    params : rc_params;
    enable : std_logic;
    rd     : std_logic;
    data   : std_logic_vector(7 downto 0);
    empty  : std_logic;
  end record rc_peripheral_1;

  type rc_peripheral_2 is record
    clk    : std_logic;
    params : rc_params_array;
    enable : std_logic;
    rd     : std_logic;
    data   : std_logic_vector(7 downto 0);
    empty  : std_logic;
  end record rc_peripheral_2;

  signal mysig_ok : rc_params_array(0 to 4)(0 to 4);  -- This is accepted by ghdl

  signal mysig_ok_2 : rc_peripheral_1(params(0 to 4));  -- This is accepted by ghdl

  -- signal mysig_wrong : rc_peripheral_2(params(0 to 4)(0 to 4));  -- This is not accepted

begin

end architecture Behavioral;
```

```sh :image: ghdl/ghdl:buster-mcode
mkdir sim_build
/usr/bin/ghdl -i --std=08 --workdir=sim_build bug.vhd
/usr/bin/ghdl -m -o sim_build/bug --std=08 --workdir=sim_build bug
```

**Context**
Please, provide the following information:

- OS: ArchLinux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

Using llvm:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-244-g773ce85c) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/tmp/bug/
Command line:
/usr/bin/ghdl1-llvm --std=08 --workdir=sim_build -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -c -fpic -o sim_build/bug.o bug.vhd
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f3b391cad6e 0x7f3b3a31d3a2 0x7f3b3a2310a5 0x55dcd5261c11 0x55dcd5490e46 0x55dcd542a30b 0x55dcd54222ee 0x55dcd54229b2 0x55dcd542b97f 0x55dcd542b9f5 0x55dcd5432594 0x55dcd541b54f 0x55dcd54a3219 0x55dcd53dd186 0x55dcd54a98ae 0x55dcd54ab275 0x55dcd54acc01 0x55dcd5252366 0x7f3b391b6021 0x55dcd52514cc 0xfffffffffffffffe
******************************************************************
```

Using gcc bakend:

```
analyze bug.vhd
/tmp/bug/bug.vhd:4: erreur interne du compilateur: dans new_offsetof, à vhdl/ortho-lang.c:1441
    4 | entity bug is
      | 
Veuillez soumettre un rapport d’anomalies complet,
avec la sortie du préprocesseur si nécessaire.
Voir <https://gcc.gnu.org/bugs/> pour les instructions.
/usr/bin/ghdl:error: exec error
```

**Additional context**

Checked using `ghdl-gcc-git-1.0dev.r4823.g773ce85c-1` and `ghdl-llvm-git-1.0dev.r4823.g773ce85c-1` packages, build from `aur/ghdl-gcc-git` and `aur/ghdl-llvm-git`.

Xiretza, 2020-04-08:
Possibly related to #756, #780.

tgingold, 2020-04-10:
sorry, but I cannot reproduce the crash.  Maybe the issue was already fixed.

csantosb, 2020-04-10:
Sorry, I meant remove comments in signal

```
  signal mysig_wrong : rc_peripheral_2(params(0 to 4)(0 to 4));
```

Just checked with latest commit, the bug is still there.

tgingold, 2020-04-10:
Indeed, I confirm the crash.

eine, 2020-08-05:
@csantosb, I think that this issue might be fixed in latest master, as I cannot reproduce. Do you mind testing?

csantosb, 2020-08-05:
@eine, Sure, no problem. Just checked ghdl with llvm/gcc backends and the issue is fixed. Thanks !

closed 2020-08-05 by csantosb