ghdl/ghdl #1203: GHDL under Windows 10 - path problem

hosszu, 2020-04-09:
I installed ghdl-0.31-mcode-win32 onto a Windows 10, my hardware is 64-bit into the c:\GHDL directory. I run first the set_ghdl_path.bat, then the reanalyze_libs.bat. I tried to analyse "hello.vhdl" file. However, I had to copy this file into the c:\GHDL\bin directory, and I had to call the ghdl.exe from this directory. Is there any way to call the ghdl.exe from a totally different directory?

Best regards,
Gábor

eine, 2020-04-09:
Hi @hosszu! v0.31 is really outdated. Is there any reason to use that instead of a more recent release (https://github.com/ghdl/ghdl/releases/tag/v0.37)?

hosszu, 2020-04-09:
Hi Eine! No, I just found a link to the version v0.31. I do not know the difference of "llvm" and "mcode" (cf. hdl-0.37-mingw32-mcode.zip). Is the ghdl-0.37-mingw64-llvm.zip the suitable for my Windows 10 ?

eine, 2020-04-09:
mcode is a built-in in-memory backend for x86, which is currently only supported for 32 bits on Windows. LLVM is a different backend which allows to target either 32 or 64 bits, apart from other architectures. You will get the same result, but different performance with either backend. See the table in https://ghdl.readthedocs.io/en/latest/getting/index.html for further details.

For your use case, I'd suggest to use `ghdl-0.37-mingw64-llvm.zip`, which is suitable for your Windows 10.

Which terminal are you goind to use? cmd? powershell? MSYS2/MINGW? cygwin?

hosszu, 2020-04-17:
I use cmd. Is it enough?

hosszu, 2020-04-17:
I unzipped ghdl-0.37-mingw64-llvm.zip; however, I did not find any installation information. Could you help me?

eine, 2020-04-17:
There is no installer. You just unzip it (as you did) and you can directly execute the binary in subdir `GHDL\0.37-mingw64-llvm\bin\`. For example:

```
>GHDL\0.37-mingw64-llvm\bin\ghdl.exe --version
GHDL 0.37 (v0.37) [Dunoon edition]
 Compiled with GNAT Version: 9.1.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Regarding you original question, note that I executed GHDL from my "Downloads" directory, even though the binary is in a different location (several directories below).

For easiness, you can set an envvar. Nonetheless, it is to be used as a "portable" installation.

Last, but not least, I suggest you to do the three exercises in https://ghdl.readthedocs.io/en/latest/examples/quick_start/README.html, so that you get a feel of the CLI usage of GHDL.

tgingold, 2020-04-24:
I close this issue as it has been answered.  Do not hesitate to re-open it if your issue is not solved.

closed 2020-04-24 by tgingold