ghdl/ghdl #1204: Revisit CLI syntax

eine, 2020-04-09:
Currently, the architecture of the CLI is based on (sub)commands and options. Commands are expected to be provided as the first argument, and do effectively decide which feature is to be used. OTOH, options provide parameters to the commands. However, there is no syntactical difference, as both commands and options start with `--` or `-i`. As a result, we rely on properly formating `--help` and on the documentation, to ensure that users understand the usage.

I think it might be easier to explain and to understand if (sub)commands were supported without `--` or `-`. In order to avoid a breaking change, I think that this should be added while the current behaviour is kept. Optionally, single letters can be capitalised.

At the same time, executing `ghdl` only produces `ghdl:error: missing command, try ghdl --help`. It might be useful if it called a subcommand by defaut instead (either `--help` or `-a`).

tgingold, 2020-04-09:
I am OK with allowing commands without - or -- like many tools.

However, if there is no command, the default can only be --help.

I don't think we should allow one letter commands.  So `-a` could be an alias of `analyze`.

tgingold, 2020-04-12:
So how to make progress on this issue ?  Do you want me to add all the commands ?

eine, 2020-04-12:
If it is not much effort, please do. I started having a look at it, but I'm finding it difficult to understand where are commands defined. For example https://github.com/ghdl/ghdl/blob/master/src/ghdldrv/ghdlcomp.adb#L127 and https://github.com/ghdl/ghdl/blob/master/src/ghdldrv/ghdldrv.adb#L1244 feel like a duplication, but `ghdl --help` shows the second one only.

tgingold, 2020-04-12:
ghdlcomp is for mcode while ghdldrv is for gcc/llvm; hence the overlap.

eine, 2020-06-09:
Coming from [#1362](https://github.com/ghdl/ghdl/issues/1362#issuecomment-641423542), perhaps we should consider adding `--sim` or `--simulate` as the "long form" of `-c`. That would make it more explicit that both are the "higher-level" (single command) approaches, while other commands are for more granular control. We might update or extend the Quick Start guide accordingly.

/cc @ozbenh

tgingold, 2020-06-10:
I am not sure the long form of "-c" is "--simulate".  Because there are two forms of "-c":
"-c .. -e" compile and elaborate
"-c .. -r" compilte, elaborate and runs.

Apparently the second one is not available with gcc/llvm backends.

eine, 2020-06-10:
Is that a technical limitation or a not implemented feature? My idea would be to support `-c [options] file... -<e|r|-elab-run> primary_unit [secondary_unit] [simulation_options...]` meaning that `-c` is "equivalent to appending `-a` before `-e|-r|--elab-run`".

Then,

`--sim [options] file... -e primary_unit [secondary_unit] [run_options...]` 

would be an alias of

`-c [options] file... --elab-run primary_unit [secondary_unit] [run_options...]`

, which would be consistent across backends.

Overall, the Quick Start Guide would be simplified to:

```sh
# For simulation
--sim [common_opts] [sim_opts] \
  path/to/src/files/*.vhd path/to/sim/files/*.vhd \
  -e primary_unit [secondary_unit] [run_options...]

# For synthesis
--synth [common_opts] [synth_opts] \
  path/to/src/files/*.vhd path/to/synth/files/*.vhd \
  -e primary_unit [secondary_unit]
```

tgingold, 2020-06-11:
> Is that a technical limitation or a not implemented feature? My idea would be to support -c [options] file... -<e|r|-elab-run> primary_unit [secondary_unit] [simulation_options...] meaning that -c is "equivalent to appending -a before -e|-r|--elab-run".

No limitation, that could be added.

umarcor, 2020-09-21:
- [x] #1467
- [ ] `--libghdl-*` and `--vpi-*` commands, defined/registered in `ghdlvpi.adb` and `ghdlsynth.adb`, respectively.
- [ ]  `ghdl disp` sub command.
- [ ] Make "no command" default to `ghdl help`.
- [ ] Provide `ghdl sim|synth FILEs -e UNIT ... `.
- [ ] Update the docs.