ghdl/ghdl #1206: VHPIDIRECT: NULL argv produces 'Aborted (core dumped)'

umarcor, 2020-04-11:
While preparing some examples about VHPIDIRECT, I found that the following example (https://github.com/umarcor/ghdl-cosim/tree/todo/vhpidirect/wrapping/basic) is successfully built and executed on my host, inside a docker container. However, the exact same example, in the same container, fails when executed on GitHub Actions.

```c
#include <stdio.h>

extern int ghdl_main(int argc, void** argv);

int main(int argc, void** argv) {
  printf("ghdl_main: %d\n", ghdl_main(0, NULL));
  return 0;
}
```

The solution is as simple as NOT passing NULL as argv, but to allocate at least a position (e.g. `char** myargv`). However, I am genuinely curious:

- Why is it different on my host and on GHA, if the container (environment) is the same?
- Shouldn't argv be ignored in ghdl_main, given that argc is 0?
- Has this always been a constraint? Is it undefined behaviour to pass NULL to ghdl_main? Is there anything that GHDL can do to avoid the crash?

Depending on the response to the second question, I might need to update the HINT in https://ghdl.readthedocs.io/en/latest/using/Foreign.html#wrapping-and-starting-a-ghdl-simulation-from-a-foreign-program, since I was convinced that it was safe to use NULL.

tgingold, 2020-04-11:
I am fixing the crash.  Don't know for GHA.

closed 2020-04-11 by tgingold

umarcor, 2020-04-11:
> I am fixing the crash.

Thanks!

> Don't know for GHA.

It was actually my fault... I was using the same container, but not exactly the same version. GHA was pulling the latest, while I had an older one locally. Hence, this was indeed a regression at some point. Hopefully it is fixed now!

commit 7a211109 (2020-04-11): grt: handle argc=0 in ghdl_main.  Fix #1206