ghdl/ghdl #1209: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_SLV

rodrigomelo9, 2020-04-11:
@tgingold don't hate me, I think that this is the last problem related with an unhandled function :P

I simplified the example because were also file open problems that I will post in another issue (I know, file open is not synthesizable, but It aborts with a *Bug occurred* message).

```
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.STD_LOGIC_UNSIGNED.ALL;
use IEEE.STD_LOGIC_arith.ALL;

entity issue is
 generic(
  data_width : integer := 4
 );
 port(
  clk : in  std_logic;
  di  : in  std_logic_vector(data_width - 1 downto 0);
  do  : out std_logic_vector(data_width - 1 downto 0)
 );
end issue;

architecture behavioral of issue is
    constant base_const : std_logic_vector(data_width - 1 downto 0) := conv_std_logic_vector(3, data_width);
    constant new_const  : std_logic_vector(data_width - 1 downto 0) := base_const + "0100";
begin
    do <= new_const;
end behavioral;
```

```
$ docker run --rm -t -v $(pwd):/src -w /src ghdl/synth:beta ghdl --synth -fsynopsys issue.vhd -e
ghdl:note: top entity is "issue"
issue.vhd:19:83:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_ADD_SLV_SLV
issue.vhd:13:3:warning: no assignment for port "do"
```

tgingold, 2020-04-11:
Don't be sorry.  That's rather easy to handle.

closed 2020-04-11 by tgingold

rodrigomelo9, 2020-04-11:
Ok @tgingold . I saw that it was a piece of cake (for you). I checked your solutions, and I can try to make something similar, but I will be scared about something missing :P

It is "easy" because it is already solved for simulation, right? The only needed thing is "to specify" that it is also supported for synthesis?

Thanks for your awesome work and quick response.

tgingold, 2020-04-11:
This operation was easy because it behaves exactly like numeric_std."+" for [unsigned, unsigned], which was already handled.

So, nothing really new.

commit 4a623031 (2020-04-11): synth-static_oper: handle more additions.  Fix #1209