ghdl/ghdl #1213: Latest release (`v0.37`) does not build with LLVM 10

FFY00, 2020-04-11:
**Description**
The latest release (`v0.37`) does not build with LLVM 10.

If this is supported in master but not in the latest release, would you please consider making a new release?

**How to reproduce?**
```
$ ./configure \
      --prefix=/usr \
      --with-llvm-config \
      --disable-werror \
      --enable-checks \
      --enable-libghdl \
      --enable-openieee \
      --enable-synth
$ make
...
Unhandled version llvm 10.0.0
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [X] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

eine, 2020-04-11:
Yes, it is supported in `master` but not in `v0.37`. However, GHDL is released once every year (around february). Not releases are published because of bugfixes or enhancements. See #1108.

FFY00, 2020-04-11:
In this case, could you provide a patch to make the latest release build with LLVM 10?

eine, 2020-04-11:
A patch would be equivalent to applying a bugfix. You can cherry-pick https://github.com/ghdl/ghdl/pull/1192/files.

However, I suggest to clone and build a specific SHA, instead of applying fixes on `v0.37`.

FFY00, 2020-04-11:
I am packaging this for Arch Linux so that is not an option. That's why I asked for a new release, if you feel a specific commit is stable enough to be distributed please make a release.

eine, 2020-04-11:
> I am packaging this for Arch Linux so building a random commit is not an option.

Please, see #1118 to understand why "building a random commit" is the *de facto* procedure to use/distribute GHDL.

> That's why I asked for a new release, if you feel a specific commit is stable enough to be distributed please make a release.

Packages for Arch Linux do exist already. Some use fixed tags, others use specific commits, others use `master`. Depending on the procedure chosen by the packager, specific fixes (such as the one I suggested to cherry-pick) are to be applied by the maintainer of the package. This is a general procedure that applies to packaging for Arch, Fedora, Debian, etc.

Don't take me wrong. I'd love if GHDL had a more frequent release schedule and if certain (build) bugfixes were backported to the latest tagged release, at least. Unfortunately, it has been discussed several times before, and it does not seem it is going to change in the near future. For context, see #1108 and #514.

Regarding packaging for Arch Linux see #901, #507 and https://github.com/eine/ghdl-packaging. I believe that the CI failures are related to the setup of the CI, and not because the packages are broken. Although focused on Fedora, #888 can also be interesting because of how patches are managed. Note that packages for MSYS2 are based on pacman/PKGBUILD too: https://github.com/ghdl/ghdl/tree/master/dist/msys2-mingw. Since these are used in CI, they are up to date for sure.

FFY00, 2020-04-11:
> Please, see #1118 to understand why "building a random commit" is the _de facto_ procedure to use/distribute GHDL.

That is simply awful.

I don't understand why you publish releases then.

> Packages for Arch Linux do exist already. Some use fixed tags, others use specific commits, others use `master`. Depending on the procedure chosen by the packager, specific fixes (such as the one I [...]

There were no official packages for ghdl on Arch Linux, I pushed them today.

Packagers don't do what they want, there are specific guidelines for each distribution. In Arch Linux we don't package unreleased or beta/unstable software. There can be exceptions, but those cases are carefully thought and the decision to package them does not come lightly.

Backporting fixes is fine and common practice.

> Don't take me wrong. I'd love if GHDL had a more frequent release schedule and if certain (build) bugfixes were backported to the latest tagged release, at least. Unfortunately, it has been [...]

That is fine, I can backport patches myself if they're not too complex.

> Regarding packaging for Arch Linux see #901, #507 and [eine/ghdl-packaging](https://github.com/eine/ghdl-packaging). I believe that the CI failures are related to the setup of the CI, and not because the packages are broken. Although focused on Fedora, #888 can also be interesting because of how patches are managed. Note that packages for MSYS2 are based on pacman/PKGBUILD too: [/dist/msys2-mingw@`master`](https://github.com/ghdl/ghdl/tree/master/dist/msys2-mingw) [](/ghdl/ghdl/tree/HEAD@{2020-04-11T19:43:14Z}/dist/msys2-mingw). Since these are used in CI, they are up to date for sure.

No, the CI failures are related to bad packaging (ghdl-gcc is conflicting with gcc). The packages also definitely do not follow the guidelines.

eine, 2020-04-12:
> > Please, see #1118 to understand why "building a random commit" is the _de facto_ procedure to use/distribute GHDL.
> 
> That is simply awful.
> 
> I don't understand why you publish releases then.

This is starting to get unpolite... I'd like your wording to be a little more respectful with the work that has been done and given for free during almost 20 years. 

> There were no official packages for ghdl on Arch Linux, I pushed them today.

I didn't say they were official. But at least 4 actively maintained packages exist on AUR. If someone was to provide an official one, I'd expect one of those to be enhanced and promoted, instead of starting a new one from scratch. At least, I'd expect you to provide a reference to existing solutions and to discuss what and why you did on top of them, without me having to bring those references into de conversation. At least, I'd expect you to say thank you, for providing you at least 8 relevant references which you apparently were not aware about. Coyping your wording, this is simply awful development and communication practice.

> In Arch Linux we don't package unreleased or beta/unstable software. There can be exceptions, but those cases are carefully thought and the decision to package them does not come lightly.

Then, you should maybe think that this is the reason for an official package of GHDL not to exist yet, since GHDL is beta/unstable software. Releasing more frequent beta/unstable release won't make it stable.

> That is fine, I can backport patches myself if they're not too complex.

The point is that, potentially, all new commits in `master` are subject to be considered backport patches (#1118). This is because, as said, GHDL is a beta, and many nuclear features are incomplete. The vast majority of new commits (specially these last two years) do fix bugs. That's the reason for existing packages for Fedora to use "random commits".

> No, the CI failures are related to bad packaging (ghdl-gcc is conflicting with gcc). The packages also definitely do not follow the guidelines.

We'd be so glad to receive any help in this regard. It worked some months ago, but using a docker container to build Arch Linux packages is quite more complex than building for other environments. The main motivation to include testing for Arch Linux in CI, is precisely to provide a reference to packagers about which commits in master are "safe" to be picked as a release candidate.

tgingold, 2020-04-12:
> On 12 Apr 2020, at 01:06, Filipe Laíns <notifications@github.com> wrote:
> 
> 
> Please, see #1118 to understand why "building a random commit" is the de facto procedure to use/distribute GHDL.
> 
> That is simply awful.
> 
> I don't understand why you publish releases then.

First, thank you for your effort to create an official Arch linux package.

The strategy for releases is very simple: spend as little time as possible on it (and spend time on development).

So the release process should be as simple as possible: update the version, tag it, push.
From time to time, this workflow doesn't work because a script has to be adjusted.

This simplicity is possible because of the large testsuite, which ensure there is no regression.  And this no-regressions policy allows anyone to use the latest commit.

We still create releases because it is reassuring for users, it allows announcements and also allows to create disrupting points.  For example, after the latest release, the mentor favour of the ieee library has been removed.

In your case, you only need one simple patch for llvm-10.  The simplest solution is to apply this patch as part as your package process.

FFY00, 2020-04-14:
> This is starting to get unpolite... I'd like your wording to be a little more respectful with the work that has been done and given for free during almost 20 years.

I was a bit blunt here, sorry. It was not meant as any sort of personal attack, I was targeting only the suggestion to package the last commit. It is very different to 

> I didn't say they were official. But at least 4 actively maintained packages exist on AUR. If someone was to provide an official one, I'd expect one of those to be enhanced and promoted, instead of starting a new one from scratch. At least, I'd expect you to provide a reference to existing solutions and to discuss what and why you did on top of them, without me having to bring those references into the conversation. At least, I'd expect you to say thank you, for providing you at least 8 relevant references which you apparently were not aware about. Coyping your wording, this is simply awful development and communication practice.

I was aware of the packages in the AUR. From a quick look right away I can see they do not follow the distribution guidelines, in these cases I usually just end up writing it from scratch to make sure everything is done properly. The existent PKGBUILDs also only build one backend, and I wanted to build all backends, and have a more complicated gcc build than needed. If I had used it as a starting point I would have end up rewriting it entirely.

I don't see why I would have to reference existing PKGBUILDs as they are unrelated to this issue. I provided the steps to reproduce, if you run them on any system with LLVM 10 you should get the same result, no matter the environment or distribution. This issue is not bind to my PKGBUILD, so I fail to see how any of this is relevant.

> Then, you should maybe think that this is the reason for an official package of GHDL not to exist yet, since GHDL is beta/unstable software. Releasing more frequent beta/unstable release won't make it stable.

There was a package at some point, it got dropped because there was no maintainer.

When I talk about beta/unstable software here I don't mean API stability, missing features, etc. What I mean is if GHDL itself works properly.

> The point is that, potentially, all new commits in `master` are subject to be considered backport patches (#1118). This is because, as said, GHDL is a beta, and many nuclear features are incomplete. [...]

As I said above, for packaging we don't care too much about incomplete features. If packaging random commits is compatible with their guidelines, great.

> We'd be so glad to receive any help in this regard. It worked some months ago, but using a docker container to build Arch Linux packages is quite more complex than building for other environments. [...]

My PKGBUILD[1] builds correct packages which are according to the guidelines, you can use it.

However there are still a few things that should be improved, eg. fix the missing pie and full relro, and running the test suite. It looks like `LDFLAGS` is not being respected, I will investigate this and open an issue if needed.

[1] https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/ghdl

> First, thank you for your effort to create an official Arch linux package. The strategy for releases is very simple: spend as little time as possible on it (and spend time on development). So the [...]

The thing is that regression tests don't ensure that things are not broken. Having regression tests is great, and makes a huge difference, but at the end of the day the tests are still software and as so, error prone. It's really hard to write tests that catch all the cases, especially for a project as big as this one.

Let's say someone submits a pull request touching a file that is not covered by tests or has a bad test, and gets merged. Now I, following your recommendation package the last commit. It turns out the last commit is broken. The software is already in repos and there are a lot of people with broken setups now.
I am packaging for Arch Linux, which is a rolling distribution, so I can just go ahead and push a fix and ask people to update. But there happens to exist a lot of downstreams which pull packages from the Arch Linux repos. Poeple on those distributions will have to wait for their repos to sync, and this might take a while. Some of them do not have rolling releases, their users will have to wait 2 weeks, 1 month, 2 months, 3 months, etc. for them to sync. Let's say I am packaging for a distribution like Ubuntu or Debian with major distro releases, if I push a bad release during the last users are stuck with it until they upgrade to a newer major distro version.

tgingold, 2020-04-14:
Just answering to a particular point:

> On 14 Apr 2020, at 22:06, Filipe Laíns <notifications@github.com> wrote:

[...]

> The thing is that regression tests don't ensure that things are not broken. Having regression tests is great, and makes a huge difference, but at the end of the day the tests are still software and [...]

We all agree about that, and that's why there are also releases.  If you need the latest feature use the master branch.  If you just want to do vhdl simulation with a ready to use tool, take the package of the distribution or the release.

But it is difficult to have both.  We could of course do more releases but I prefer not to.

We could backport the patch to support llvm-10.  But first I don't know how well the llvm-10 backend works as I haven't tested it and it is not part of any CI.  And second, if we start to backport a patch I fear we will have more requests.  So I prefer to let that door closed.  We could revisit this option if someone is interested in maintaining the release.  So far I am not sure it is worth.

Matter of taste and colors but also a matter of priorities!

closed 2020-04-24 by FFY00