ghdl/ghdl #1214: Building ghdllib does not work with the GCC backend

FFY00, 2020-04-11:
**How to reproduce?**
```sh
$ cd path/to/ghdl
$ ./configure \
      --prefix=/usr \
      --with-gcc=path/to/gcc \
      --disable-werror \
      --enable-checks \
      --enable-libghdl \
      --enable-openieee \
      --enable-synth
$ make copy-sources
$ cd path/to/gcc
$ path/to/gcc/configure \
      --prefix=/usr \
      --libdir=/usr/lib \
      --libexecdir=/usr/lib \
      --enable-languages=vhdl \
      --enable-default-pie \
      --with-system-zlib \
      --with-isl \
      --disable-bootstrap \
      --disable-lto \
      --disable-multilib \
      --disable-libssp \
      --disable-libgomp \
      --disable-libquadmath
$ make
$ cd path/to/ghdl
$ make ghdllib
...
make[1]: *** No rule to make target '/usr/bin/ghdl', needed by 'lib/ghdl/std/v87/std-obj87.cf'.  Stop.
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [X] Released binaries: `https://github.com/ghdl/ghdl/archive/v0.37.tar.gz`
  - [ ] Built from sources: `commit SHA`

eine, 2020-04-11:
Please, do provide information about the context, so that we can reproduce the issue: OS, version of GCC, version of GHDL, specific set of commands (where did you clone ghdl and gcc), etc.

Building ghdllib with GCC backend does work for sure, since it is tested in CI (https://github.com/ghdl/ghdl/runs/579067868?check_suite_focus=true) and it is the main backend used by all the users that want code coverage features.

Instead of trying to come up with your own build script, you might want to use https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh as a reference. That is the script that we use in CI and to generate all the `ghdl/ghdl:*gcc*` docker images. Moreover, for Arch Linux, see https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ghdl (stable) and https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ghdl-gcc-git (`master`).

FFY00, 2020-04-11:
You can find my build instructions here: https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/ghdl

To build simply invoke `makepkg` in an Arch Linux machine (it is actually advised to build in a clean chroot so you should run `extra-x86_64-build` instead)

Either way, I think I figured out the issue. You rely on installing gcc before building ghdllib. AFAIK this is not supported by several distribution build environments, including arch. We have separate build and install steps. Setting `GHDL_GCC_BIN` and `GHDL1_GCC_BIN` would be a hack around this but it does seem to be working for me.

```
$ make \
		GHDL_GCC_BIN="$srcdir"/gcc-build/gcc/ghdl \
		GHDL1_GCC_BIN="--GHDL1=$srcdir/gcc-build/gcc/ghdl1" \
		ghdllib
...
/build/ghdl/pkg/ghdl-gcc/usr/bin/ghdl --disp-standard --std=87 > /build/ghdl/pkg/ghdl-gcc/usr/lib/ghdl/src/std/v87/standard.vhdl
/bin/sh: /build/ghdl/pkg/ghdl-gcc/usr/bin/ghdl: No such file or directory
```

I am not sure what I am doing wrong here.

> Instead of trying to come up with your own build script, you might want to use [/dist/ci-run.sh@`master`](https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh) [](/ghdl/ghdl/blob/HEAD@{2020-04-11T19:58:27Z}/dist/ci-run.sh)

Your script seems overcomplicated, at least for what we are trying to do here, and it's definitely not compatible with packaging guidelines.

eine, 2020-04-12:
> You can find my build instructions here: [git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/ghdl](https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/ghdl)

I see no execution of the testsuite. How do you handle that?

> To build simply invoke `makepkg` in an Arch Linux machine (it is actually advised to build in a clean chroot so you should run `extra-x86_64-build` instead)

Unfortunately, I don't have an Arch Linux machine. As said in #1213, I'd be glad to receive any help to have it done in a Docker container. Should that work, including running the testsuite, I'd be glad to have your package added to CI (either in this repo or in ghdl/docker).

> Either way, I think I figured out the issue. You rely on installing gcc before building ghdllib. AFAIK this is not supported by several distribution build environments, including arch. We have [...]

Both packages referred above do use the envvars, apparently with successful results. Can you please tell us exactly where are you using them? Specific reproduction steps would be welcome.

Regarding the requirement of installing GCC before building ghdllib, I feel that it might be a technical limitation. This is because you are building a GCC and patching it to add a front-end. @tgingold, can you please clarify?

> Your script seems overcomplicated, at least for what we are trying to do here, and it's definitely not compatible with packaging guidelines.

Yes. The purpose of the script is to be used on CI for multiple distributions and different configuration setups. It is used to generate tarballs, to generate docker images and to run the testsuites in them. Because it is meant for multiple distributions, we cannot rely on any specific package manager, and we need to replicate the functionality ourselves. However, the build and test functions are essentially what you need to add to any package. You just need to ignore the snippets that don't correspond to the specific backend you are working with.

In the mid term, I'd like to replace the script with specific packaging sources for each distribution. For Windows, we are already using pacman's PKGBUILD files, instead of custom scripts. This is interesting, because we can provide nightly builds for each commit as in https://github.com/eine/ghdl-packaging/actions/runs/72271631. This might not make much sense on Arch Linux, because users are expected to build the packages locally. Nonetheless, I expect you get the idea.

hackfin, 2020-04-12:
> Either way, I think I figured out the issue. You rely on installing gcc before building ghdllib. AFAIK this is not supported by several distribution build environments, including arch. We have [...]

It's actually not a hack, but the only feasible way to do it when you need to do a two-stage (bootstrap) build. It's kinda 'standard' when building cross compilers and libraries, so you'll find these kind of constructs in a few gcc build scripts/makefiles.

If you prefer the makefile approach rather than the linear scripts, you might grab some templates from https://github.com/hackfin/ghdl-cross.mk, I'm cross-building GHDL targets/packages for several platforms that way, native builds are a bit simpler (and you might skip the sandbox build). Unfortunately, the full auto-build won't currently work on Docker hub, you'd have to build in a local container for your target platform.

umarcor, 2020-04-12:
> Unfortunately, the full auto-build won't currently work on Docker hub, you'd have to build in a local container for your target platform.

Actually, this is straightforward in other CI services (say Travis or GitHub Actions) as long as the tools you want to use are supported by qemu-user (the DBM, not the KVM). Project [dbhi/qus](https://github.com/dbhi/qus) provides helpers and usage examples. It is used in [dbhi/docker](https://github.com/dbhi/docker) to build multiarch images including GHDL, GtkWave, Octave, etc. (you can check CI logs).

Unfortunately, it seems that arm64v8/archlinux and others do not exit (https://github.com/docker-library/official-images#architectures-other-than-amd64). For Debian, Ubuntu, Alpine, etc. images exist and this allows to avoid cross-building. Just use the same commands that you would use on the target (either natively or inside a container).

Note that dbhi/qus is not required on amd64 Windows hosts, because docker-for-win includes qemu-user as a built-in feature. See docker/binfmt#17.

hackfin, 2020-04-15:
Just on a sidenote: It would work on Docker hub in theory, but the full build dance just takes too long or uses too much disk space, therefore gets cancelled. It could be migrated to gitlab, probably.
And I should have written: building a cross-GHDL(analyze on arch X, emit for Y), not particularely cross-building GHDL.

FFY00, 2020-04-16:
> I see no execution of the testsuite. How do you handle that?

I don't yet, that's something to improve.
 
> Unfortunately, I don't have an Arch Linux machine. As said in #1213, I'd be glad to receive any help to have it done in a Docker container. Should that work, including running the testsuite, I'd be glad to have your package added to CI (either in this repo or in ghdl/docker).

Sure, I did this recently for prjtrellis. See https://github.com/SymbiFlow/prjtrellis/pull/134.

Unfortunately Github Actions is not compatible with this (I have opened a request `https://github.community/t5/GitHub-Actions/Access-to-cgroups/m-p/54299`) so it would have to be done with travis. Well, I guess I could make it work on Github Actions by not building inside a clean chroot. Let me know what you want to do here.
 
> Both packages referred above do use the envvars, apparently with successful results. Can you please tell us exactly where are you using them? Specific reproduction steps would be welcome.

That was just me being dumb, ghdl needs to be built before ghdllib as we need it to build ghdllib.

> Regarding the requirement of installing GCC before building ghdllib, I feel that it might be a technical limitation. This is because you are building a GCC and patching it to add a front-end. @tgingold, can you please clarify?

Well, I could provide my gcc build directory to the build system and then it could find the required binaries automatically.

> In the mid term, I'd like to replace the script with specific packaging sources for each distribution. For Windows, we are already using pacman's PKGBUILD files, instead of custom scripts. This is interesting, because we can provide nightly builds for each commit as in [eine/ghdl-packaging/actions/runs/72271631](https://github.com/eine/ghdl-packaging/actions/runs/72271631). This might not make much sense on Arch Linux, because users are expected to build the packages locally. Nonetheless, I expect you get the idea.

Arch Linux users aren't expected to build packages locally, that is the model from the AUR but it has different reasons. You can still provide an Arch Linux repository with -git packages.

closed 2020-04-21 by FFY00

eine, 2020-04-24:
> Sure, I did this recently for prjtrellis. See [SymbiFlow/prjtrellis#134](https://github.com/SymbiFlow/prjtrellis/pull/134).
> 
> Unfortunately Github Actions is not compatible with this (I have opened a request [github.community/t5/GitHub-Actions/Access-to-cgroups/m-p/54299](https://github.community/t5/GitHub-Actions/Access-to-cgroups/m-p/54299)) so it would have to be done with travis. Well, I guess I could make it work on Github Actions by not building inside a clean chroot. Let me know what you want to do here.

IMHO, it is better not to use a "docker flavoured GitHub Action". Instead, I'd use any other script/makefile and call it as a step in a/any (CI) workflow. Apart from that, thanks a lot for the reference. I will check it carefully to rework eine/ghdl-packaging.
 
> Arch Linux users aren't expected to build packages locally, that is the model from the AUR but it has different reasons. You can still provide an Arch Linux repository with -git packages.

What are your packaging plans?

- Will you provide a single PKGBUILD file that generates three packages (one for each backend), and users need to install one of them only?
- Will you provide two or more packages so that "stable" and `master`/`git` versions are available?
  - Then, users would install up to 6 different versions (all of the incompatible with each other)? Well, stricly, they might be partially compatible, by pairs.
- Are you going to provide some helper script to allow users to install full IEEE libs, even though you are not allowed to distribute them?

Note that these are general questions that arise due to the variablity when configuring GHDL.

FFY00, 2020-04-24:
> IMHO, it is better not to use a "docker flavoured GitHub Action". Instead, I'd use any other script/makefile and call it as a step in a/any (CI) workflow. Apart from that, thanks a lot for the reference. I will check it carefully to rework eine/ghdl-packaging.

I managed to get it to work. See https://github.com/FFY00/build-arch-package.

> What are your packaging plans?
> 
>     * Will you provide a single PKGBUILD file that generates three packages (one for each backend), and users need to install one of them only?

Yes, one PKGBUILD builds all packages. This makes it easier to update, only one PKGBUILD needs to be modified. Users can only install one of them, all provide `ghdl` so any user that tries to install it will be prompt to choose between one of them.

>     * Will you provide two or more packages so that "stable" and `master`/`git` versions are available?

In the official repos, only stable (last tag). If there's any need I could provide -git packages in a custom repo.

>       * Then, users would install up to 6 different versions (all of the incompatible with each other)? Well, stricly, they might be partially compatible, by pairs.

No, because all packages provide and conflict `ghdl` they can only have one installed.

>     * Are you going to provide some helper script to allow users to install full IEEE libs, even though you are not allowed to distribute them?

No, I can however put a modified PKGBUILD in AUR, but this requires the users to build the package themselves.

Do the openieee and full IEEE libs conflit? I guess I could package the IEEE libs separate from GHDL, this way we could have an AUR package that just builds the full IEEE libs. It should be much easier for the users as they wouldn't need to build GHDL, which can be resource intensive.

> Note that these are general questions that arise due to the variablity when configuring GHDL.

I will do my best to get the official package to work fine for most people. Full IEEE is something we cannot have, so I will have to compromise there.

marzoul, 2020-04-25:
Hi,

> If there's any need I could provide -git packages in a custom repo.

Activity around GHDL in Archlinux has got much more serious recently.
Maybe my relatively casual maintenance of AUR packages is not sufficient.
If that is the case, I can hand these over to someone else.
https://aur.archlinux.org/packages/ghdl-gcc-git/
https://aur.archlinux.org/packages/ghdl-llvm-git/
https://aur.archlinux.org/packages/ghdl-mcode-git/

Regards,
Adrien

eine, 2020-07-23:
I updated all the PKGBUILD files in [eine/ghdl-packaging](https://github.com/eine/ghdl-packaging). @FFY00's PR/packages were cherry-picked, and latest @marzoul's fixes were applied. Unfortunately, there are still some issues which are preventing some CI jobs from passing:

---

community-git (@FFY00): https://github.com/eine/ghdl-packaging/runs/901175445?check_suite_focus=true#step:3:9801
```
...
  ==> Creating package "ghdl-mcode-git"...
    -> Generating .PKGINFO file...
    -> Generating .BUILDINFO file...
    -> Generating .MTREE file...
    -> Compressing package...
  ==> Starting package_ghdl-llvm-git()...
  /wrk/PKGBUILD: line 139: cd: ghdl-llvm-git: No such file or directory
  ==> ERROR: A failure occurred in package_ghdl-llvm-git().
      Aborting...
```

I think this might be because of https://github.com/eine/ghdl-packaging/blob/master/archlinux/community-git/PKGBUILD#L129 and https://github.com/eine/ghdl-packaging/blob/master/archlinux/community-git/PKGBUILD#L139. @FFY00 should I rewrite the latter as the former?

---

community (@FFY00): https://github.com/eine/ghdl-packaging/runs/901175480?check_suite_focus=true#step:3:11329
```
---
  gna bug0108: ok
  gna bug0109: failed
  analyze repro.vhdl
  elaborate and simulate repro
  repro.vhdl:33:9:@0ms:(report note): z = image-ent
  analyze err.vhdl
  err.vhdl:3:10: unit "fixed_pkg" not found in library "ieee"
  err.vhdl:9:28: entity 'phz_calc' was not analysed
  ghdl: compilation error
```

This is because openieee is used instead of full IEEE libs. See #1255.

---

gcc-git (@marzoul): https://github.com/eine/ghdl-packaging/runs/901175463?check_suite_focus=true#step:3:18871
```
...
  checking package integrity...
  loading package files...
  checking for file conflicts...
  error: failed to commit transaction (conflicting files)
  /usr/lib/libiberty.a exists in both 'gcc' and 'ghdl-gcc-git'
  Errors occurred, no packages were upgraded.
```

I'm not sure about this. I cherry-picked @marzoul's change that removes `usr/include/libiberty`, to avoid conflicts. However, I don't see `libiberty.a` being explicitly excluded/removed in https://aur.archlinux.org/packages/ghdl-gcc-git/. Any guess?

---

gcc: https://github.com/eine/ghdl-packaging/runs/901175500?check_suite_focus=true#step:3:25837
```
...
  checking package integrity...
  error: failed to commit transaction (conflicting files)
  loading package files...
  checking for file conflicts...
  /usr/lib/libiberty.a exists in both 'gcc' and 'ghdl-gcc'
  Errors occurred, no packages were upgraded.
```

This is the same error/case as gcc-git.

eine, 2020-07-23:
@FFY00:

> Yes, one PKGBUILD builds all packages. This makes it easier to update, only one PKGBUILD needs to be modified. Users can only install one of them, all provide ghdl, so any user that tries to install it will be prompt to choose between one of them.

> No, because all packages provide and conflict ghdl they can only have one installed.

There are use cases where having more than one backend available is desirable. Most packagers do as you, and generate incompatible packages, so users need to choose one of them. However, Debian packagers allow multiple backends to be installed at the same time: https://salsa.debian.org/electronics-team/ghdl/ghdl/-/tree/master/debian.

The complexity related to having multiple backends installed at the same time is that all VHDL sources can be shared, but LLVM's and GCC's `*.o` objects are NOT compatible. In Debian, they create separate subdirs (`/usr/lib/ghdl/mcode`, `/usr/lib/ghdl/llvm` and `/usr/lib/ghdl/gcc`) and all VHDL sources are duplicated too.

Ideally, I think that all packagers (Debian, Fedora, Arch, MSYS2...) should agree on a similar structure, so we can advance towards providing a similar CLI experience to users on any platform.

See also msys2/MINGW-packages#6688.

> In the official repos, only stable (last tag). If there's any need I could provide -git packages in a custom repo.

In the end, you provided them. Thank you! As commented above, there is some issue, but overall it looks good.

`git` packages/workflows in [eine/ghdl-packaging](https://github.com/eine/ghdl-packaging) are to be used as canaries for packagers. I will make this repo dispatch events there, so we see packages breaking as soom as incompatible changes are pushed to master here.

---

> Activity around GHDL in Archlinux has got much more serious recently.
Maybe my relatively casual maintenance of AUR packages is not sufficient.
If that is the case, I can hand these over to someone else.

@marzoul, we can co-maintain those three packages betweem AUR and eine/ghdl-packaging. If you compare https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ghdl-gcc-git and https://github.com/eine/ghdl-packaging/blob/master/archlinux/gcc-git, you will find that the differences are only pkgver and how the testsuite is handled. I will take care of CI and issues specific to GHDL, and you can take care of issues related to Arch, AUR and pacman. What do you think?

FFY00, 2020-07-23:
> The complexity related to having multiple backends installed at the same time is that all VHDL sources can be shared, but LLVM's and GCC's `*.o` objects are NOT compatible. In Debian, they create [...]

I am totally up for it, but IMO it is something that would preferably be provided by the upstream Makefile. Would it be possible to add this scheme? Or perhaps entirely switch to this scheme.