ghdl/ghdl #1217: synth: Incorrect bit reversal of record

daveshah1, 2020-04-12:
In synthesis assigning a record; with another record signal in the assignment; it seems like the record becomes bit-reversed.

Consider:
```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
port (p, q : out std_logic);
end entity;

architecture arch of top is
	type subrecord_r is record
		c : std_logic;
		d : std_logic;
	end record;
	type record_r is record
		s : subrecord_r;
		a : std_logic;
		b : std_logic;
	end record;
	signal s : subrecord_r;
	signal r : record_r;
begin
	s <= ('0', '0');
	r <= (s, '0', '1');
	p <= r.a;
	q <= r.b;
end architecture;
```

I am running `yosys -m ~/ghdlsynth-beta/ghdl.so -p "ghdl record.vhd -e top; prep -top top; dump"`.

I would expect `p` to be 0 and `q` to be 1. But both values are 0 and it seems like the whole record is becoming bit-reversed.

BTW, I may well be filing a few more issues over the next few weeks, as my ultimate aim is to get the MiST SNES core working as a nice complex ECP5 open source test case - this issue was reduced from the 65C816 CPU core. I haven't done VHDL for a couple of years so hopefully I'm not mistaken here.

eine, 2020-04-12:
Hi @daveshah1! Nice to see you around! Just a couple of hints:

- Dependending on the type of bugs that you get, you might want to open issues in https://github.com/ghdl/ghdl-yosys-plugin instead of or apart from here. As a rule of thumb, you should be able to execute `ghdl --synth record.vhd -e top` in the same environment where you are using yosys and the plugin. However, using `--synth` takes yosys out of the flow. If synth fails, the issue belongs to the codebase in this repo. If synth is correct but yosys fails, it's likely an issue with the plugin. Docs were updated in this regard just a couple of days ago: https://ghdl.readthedocs.io/en/latest/using/Synthesis.html
- There is #974 to gather projects which we would like to have tested. It seems that the SNES core that you are testing is part of [MiSTer-devel](https://github.com/MiSTer-devel), isn't it? This is just for me to know whether I should add a new item to the list.

Are you building and installing GHDL, ghdl-yosys-plugin and yosys manually?

daveshah1, 2020-04-12:
1. Thanks, I didn't know about --synth.
2. Yes, https://github.com/MiSTer-devel/SNES_MiSTer. There are a few issues that I've hacked around and need to go back and create proper test cases for. It also relies on Altera memory primitives which I'm in the process of replacing.
3. Yes, all built from scratch

eine, 2020-04-12:
Thanks! In fact, building them from scratch is the suggested solution for your use case. Docker images exist (https://github.com/ghdl/docker#-cache-5-jobs--max-4--11-images-weekly, YosysHQ/nextpnr#383), but yosys, nextpnr, etc. are updated weekly.

Xiretza, 2020-04-12:
For the record, this is an issue with `--synth`. Normally, the bits are reversed when assigned to the record to match the `downto` direction of the vector:

```vhdl
type record_r is record
	a : std_logic;
	b : std_logic;
end record;
[...]
r <= ('0', '1');
```

becomes

```vhdl
r <= "10";
[...]
-- p
n6_o <= r (0);
-- q
n7_o <= r (1);
```

when synthesized (which is correct), while the original code with a subrecord

```vhdl
type record_r is record
	s : subrecord_r;
	a : std_logic;
	b : std_logic;
end record;
[...]
r <= (s, '0', '1');
```

becomes

```vhdl
n7_o <= s & '0' & '1';
-- p
n8_o <= r (2);
-- q
n9_o <= r (3);
```

so both outputs wrongly point into `s`.

closed 2020-04-12 by tgingold

tgingold, 2020-04-12:
Thanks for the issue and ready for the following ones!

commit 3cfe2f9a (2020-04-12): synth-aggr: fix record aggregate element order.  Fix #1217