ghdl/ghdl #1218: synth: Is_Positive is overly pessimistic

daveshah1, 2020-04-12:
**Description**
Shifting right by a positive integer plus 1 gives a GHDL bug.

**Expected behaviour**
Builds OK

**How to reproduce?**

```vhd :file: srl_int.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity top is
port (
	ch : in integer range 0 to 7;
	din : in unsigned(7 downto 0);
	dout : out unsigned(7 downto 0)
);
end entity;

architecture arch of top is
begin
	dout <= din srl (ch + 1);
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth srl_int.vhd -e top
```

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-293-g81ad114f) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/home/david/test/
Command line:
ghdl --synth srl_int.vhd -e top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-oper.adb:593
Call stack traceback locations:
0x5576e6d342a7 0x5576e6d33146 0x5576e6d48368 0x5576e6d1dc54 0x5576e6d20a8f 0x5576e6d2dc56 0x5576e6d2decf 0x5576e6d1d20b 0x5576e6d1d37f 0x5576e6d15d38 0x5576e6d8123a 0x5576e6ca7dc3 0x5576e6e344f1 0x5576e6a67dfc 0x7ff3a8854021 0x5576e6a6659c 0xfffffffffffffffe
******************************************************************
```

Xiretza, 2020-04-12:
Not directly related to your problem, but FYI, the shift operators have somewhat nonsensical behaviour, the `shift_[left/right]` functions from `numeric_std` are a lot more reliable: http://jdebp.eu./FGA/bit-shifts-in-vhdl.html

daveshah1, 2020-04-12:
Yes, I even have bad memories of being bitten by them myself when I was learning FPGAs about 4 years ago (I started out with VHDL).

tgingold, 2020-04-13:
So the negative values will also be handled, and yosys should be able to optimize the logic.

Thank you for the issue.

commit 5ed51a3a (2020-04-13): synth-oper: also handle negative values for shift.  Fix #1218
closed 2020-04-13 by tgingold