ghdl/ghdl #1220: synth: Support for multi-dimensional ROMs

daveshah1, 2020-04-12:
**Description**
Currently, using a 2D ROM causes an internal error. It would be nice if either they were supported, or caused a clear error if supporting them is difficult (I can work around them easily enough in this case so it's not blocking).

**Expected behaviour**
Works, or produces clearer error.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: const2d.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity top is
	port(
		clock : in std_logic;
		x : in std_logic_vector(1 downto 0);
		y : in std_logic_vector(1 downto 0);
		data : out std_logic_vector(3 downto 0)
	);
end entity;

architecture arch of top is

	type rom_t is array(0 to 3, 0 to 3) of std_logic_vector(3 downto 0);
	constant rom : rom_t := (
		("0001", "0010", "0100", "1000"),
		("0011", "0110", "1100", "1001"),
		("1110", "1101", "1011", "0111"),
		("0000", "1111", "1111", "0000")
	);
begin
	process (clock)
	begin
		if rising_edge(clock) then
			data <= rom(to_integer(unsigned(x)), to_integer(unsigned(y)));
		end if;
	end process;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth const2d.vhd -e top
```

**Context**

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-293-g81ad114f) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/home/david/test/
Command line:
ghdl --synth -v const2d.vhd -e top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:313
Call stack traceback locations:
0x55b27ee158c0 0x55b27ee15c7b 0x55b27ee16415 0x55b27ee1650b 0x55b27ee1ac89 0x55b27ef842db 0x55b27ef8437f 0x55b27ef7cd38 0x55b27efe823a 0x55b27ef0edc3 0x55b27f09b4f1 0x55b27eccedfc 0x7f0348c58021 0x55b27eccd59c 0xfffffffffffffffe
******************************************************************
```

eine, 2020-04-12:
Ref #1069 (not dup).

tgingold, 2020-04-13:
Now fixed.  I think the is the first real case of a multi-dim ROM.  I certainly didn't look for that enough.

Should I also expect a multi-dim RAM ?

Thank you for the issue.

closed 2020-04-13 by tgingold

daveshah1, 2020-04-13:
I haven't seen any multi-dim RAMs, although all the RAMs in the SNES project seem to use Altera blackboxes that I am currently replacing with Verilog modules (partly because that's what I'm more comfortable with, partly as an excuse to test mixed-language builds).

tgingold, 2020-04-13:
Currently the plugin does not instantiate abstract verilog modules.  There is a pending issue for that.
So maybe this is the most urgent item...

daveshah1, 2020-04-13:
It seems like it works if you have a VHDL `component` (but not using `entity work.x` syntax)

donnie-j, 2020-04-13:
> I certainly didn't look for that enough.  Should I also expect a multi-dim RAM ?

FWIW we had 2 places where we used 2D RAMs in a real design (high speed data link).  Xilinx ISE handled it fine, but Cadence DC could not when the design went to ASIC, so they were flattened... not sure if both versions are still maintained in the repo, I'll try and have a look.

So I'd say nice to have, but some commercial tools don't, so likely rare in the wild.

commit 600f62bf (2020-04-13): synth: handle multi-dim ROMS.  Fix #1220