ghdl/ghdl #1222: ghdl has recently stopped workiing with cocotb

benreynwar, 2020-04-13:
**Description**
As of bb75c67563d6dc8920507909779f394d6760dd81 (April 10) ghdl has stopped working with cocotb.  The error seen is:

```
loading VPI module '/home/ben/Code/cocotb/tests/test_cases/test_vhdl_access/build/libs/x86_64/libvpi.so'
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:102  in set_program_name_in_venv        Using Python virtual
 environment interpreter at /home/ben/.virtualenvs/slvcodec/bin/python
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
/usr/local/bin/ghdl: symbol lookup error: /home/ben/Code/cocotb/tests/test_cases/test_vhdl_access/build/libs/x86_64/libvpi.so: undefined 
symbol: vpi_register_systf
```

**How to reproduce?**
It can be reproduced by checking out the cocotb repository and running the tests that come with cocotb.
```
export SIM=ghdl
make test
```

tgingold, 2020-04-13:
Are you using linux ?

tgingold, 2020-04-13:
I cannot reproduce the issue.  Are you sure you have correctly installed ghdl ?

Also, try 'nm -D' on the executable produced by 'ghdl -e' and check if vpi_register_systf is present.

eine, 2020-04-13:
I don't know if it is relevant, but cocotb seems not to use `ghdl -e`: cocotb/cocotb#1054

tgingold, 2020-04-13:
I tried to reproduce the issue using `make` in a cocotb example.

benreynwar, 2020-04-13:
I created a dummy github repo to get travis to run cocotb/tests/test_cases/test_configuration
with ghdl.  It reproduces the issue that I observed.
If nothing else it will make clear what I'm doing wrong.

https://travis-ci.org/github/benreynwar/check_cocotb_ghdl

tgingold, 2020-04-13:
I think that's the best way to create a reproducer for a complex case.

I missed you were using the mcode backend.  I think I now understand the reason for the failure.

tgingold, 2020-04-13:
I think cocotb makefile need to be adjusted so that the .so depends on libghdlvpi.

closed 2020-04-13 by tgingold

benreynwar, 2020-04-13:
Thank you!  My understanding is that cocotb creates the .so during "pip install" process.  Normally when I update ghdl I just do a "pip uninstall cocotb" followed by a reinstall.  The makefiles are just used for generating the simulation and won't regenerate the libraries used.  But I'm not confident about this so I'll ping a cocotb maintainer to make sure I'm correct.

commit 430ca119 (2020-04-13): Makefile.in: link ghdl_mcode with GRT_EXEC_OPTS.  Fix #1222