ghdl/ghdl #1224: VHDL2008: unconstrained array of unconstrained std_logic_vector

RocketRoss, 2020-04-13:
**Description**
Unconstrained arrays of std_(u)logic_vectors are spec'ed in VHDL 2008. An implementation of it causes a bug print out.

**How to reproduce?**
```
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end;

architecture arch of tb is

  type ulogic_mat_t is array(natural range <>) of std_logic_vector; --this requires vhdl08

begin
  process
    constant logic2D: ulogic_mat_t(0 to 1) := (('U', 'X', '0', '1', 'Z', 'W', 'L', 'H', '-'), ('U', 'X', '0', '1', 'Z', 'W', 'L', 'H', '-'));
  begin

    for i in logic2D'range loop
      for j in logic2D'range(1) loop
        report "2D [" & integer'image(i) & "," & integer'image(j) & "]: " & std_logic'image(logic2D(i)(j)) severity note;
      end loop ;
    end loop ;

    wait;
  end process;
end;
```

Build.sh:
```
ghdl -a --std=08 tb.vhd
ghdl -e --std=08 tb &&
./tb
```

- OS:
- Origin:
  - WSL Ubuntu.
  - [x] Built from sources: `547d31b8d955325cbad13e47211d4e4049bf03d0`
  - [x] LLVM backend

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-274-g93d01c6d) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/d/Development/OpenSource/ghdl-cosim/vhpidirect/cinterface/test/
Command line:
/etc/ghdl-llvm-PIC/bin/ghdl1-llvm --std=08 -P/etc/ghdl-llvm-PIC/lib/ghdl/ieee/v08/ -P/etc/ghdl-llvm-PIC/lib/ghdl/ -c -fpic -o tb.o tb.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-utils.adb:1149
Call stack traceback locations:
0x7f08d88147b1 0x4c925b 0x4c9287 0x629b1b 0x636d50 0x5d1cd4 0x5d226d 0x5d236c 0x5dc07f 0x5c0096 0x5c67f1 0x647630 0x58ab55 0x64d82c 0x64f108 0x650a01 0x4086c9 0x7f08d78a1b95 0x4079b8 0xfffffffffffffffe
```

RocketRoss, 2020-04-13:
Same thing on `f9878709`:

eine, 2020-04-13:
Is this a dup of #756? See also #899, #961 and #1067. Overall, unconstrained arrays of uncontrained types is a VHDL 2008 feature that is not fully supported yet.

RocketRoss, 2020-04-13:
Pretty much! 
Sorry, I should have know people would have bumped in to this.

tgingold, 2020-04-13:
Let's keep this issue open.  The testcase is simple and should work.

RocketRoss, 2020-04-13:
For interest's sake the following does work (2D unconstrained array of bits):

```
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end;

architecture arch of tb is

  type ulogic_mat_t is array(natural range <>, natural range <>) of std_logic; --this requires vhdl08

begin
  process
    constant logic2D: ulogic_mat_t(0 to 1, 0 to 8) := (('U', 'X', '0', '1', 'Z', 'W', 'L', 'H', '-'), ('U', 'X', '0', '1', 'Z', 'W', 'L', 'H', '-'));
  begin

    for i in logic2D'range loop
      for j in logic2D'range(2) loop
        report "2D [" & integer'image(i) & "," & integer'image(j) & "]: " & std_logic'image(logic2D(i,j)) severity note;
      end loop ;
    end loop ;

    wait;
  end process;
end;
```

eine, 2020-04-13:
> Pretty much!
> Sorry, I should have know people would have bumped in to this.

It's ok! We are not very strict with duplicates, because there are tiny differences that can trigger different bugs. This is not an exact dup of #756, because generics are used there. That's why I left the issue open, even though I marked it as a bug. For future issues, it would be helpful if you checked the "suggested similar issues" and added a few to your first message if you feel they are related.

umarcor, 2020-04-13:
Indeed... #1055

RocketRoss, 2020-04-18:
I have started trying to investigate this. The error has moved from vhdl-utils.adb:1149->1177 on 0ee5e564.

RocketRoss, 2020-04-20:
And now it has jumped file to `trans-chap3.adb:48` on `de0014dd`...

RocketRoss, 2020-04-28:
~An alternative for the interim is to use package declared type, of unconstrained slv_arr, and then constrain the bounds in both the entity's ports and the signals.....~

commit d41483dd (2021-08-04): testsuite/gna: add a test and close #1224
closed 2021-08-04 by tgingold