ghdl/ghdl #1227: synth: Clock inverter added for falling_edge clocks results in dubious timing

daveshah1, 2020-04-14:
**Description**
Currently, synthesis converts falling edge clocks to an inverter and a rising edge clock. While functionally correct, this creates a new clock domain with some skew caused by the inverter LUT if the FPGA toolchain doesn't fold the inverter back into a falling edge DFF (which Yosys doesn't in general).

**Expected behaviour**
Falling edge clocks result in a falling edge DFF post-synthesis

**How to reproduce?**

```vhd :file: falling_edge.vhd
library ieee;
use ieee.std_logic_1164.all;

entity top is
	port(
		clk : in std_logic;
		d : in std_logic;
		q : out std_logic
	);
end entity;

architecture arch of top is
begin
	process (clk)
	begin
		if falling_edge(clk) then
			q <= d;
		end if;
	end process;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth falling_edge.vhd -e top
```

tgingold, 2020-04-14:
I suppose that removing any inverter before the clock input of a posedge DFF  and converting the DFF to a negedge one is not what the user prefer.  Is it correct ?

daveshah1, 2020-04-14:
There might be a case where someone wants a whole clock domain inverted, with posedge FFs (perhaps because some other blocks in that domain don't have configurable polarity), then keeping the inverter and posedge DFFs would be necessary imo.

tgingold, 2020-04-14:
Yes.  I will adjust the code.

tgingold, 2020-04-15:
Now fixed.  You should update both ghdl and the yosys plugin.

Thank you for the issue.

commit 15539c0a (2020-04-15): synth: rework edge handling to properly support falling edge.  Fix #1227
closed 2020-04-15 by tgingold