ghdl/ghdl #1229: VHDL2008: Generic Package with VHPIDIRECT

RocketRoss, 2020-04-15:
**Description**
MWE is a little complicated, please bear with me.
- A: VHDL 2008 Generic Packages work.
- B: VHDL 2008 Packaged VHPIDIRECT functions work
- The two together creates a bug dump.

**Expected behaviour**
I expect to be able to pass a generic to a generic package and use a VHPIDIRECT function declared in the same package.

**How to reproduce?**
I have split A and B required code with a long ----- comment line. Currently A is commented out and B is in effect. The bug occurs when A and B are in effect (A is uncommented (Ctrl+/ in vscode) ). The MWE enables (relatively) easy and concise reproduction (I hope).

**tb.vhd**
```vhd :file: tb.vhd
entity tb is
end entity tb;
architecture RTL of tb is
	-- -- A
	-- package tbpkg is new work.pkg generic map (N => 3);
	---------------------------------------------
	-- -- B
	use work.pkg.all;
	-- -- END
begin
	process
		-- -- A
		-- use tbpkg.all;
		-----------------------------------------
		-- -- B
		variable int : integer := c_int;
		-- -- END
	begin	
		-- -- A	
		-- showPackageN;
		----------------------------------------------
		-- -- B
		report "C Int: " & integer'image(int);
		-- -- END
		wait;
	end process;
end architecture RTL;
```
**pkg.vhd**
```vhd :file: pkg.vhd
package pkg is
	-- -- A
	-- generic (N : integer := 2);
	-- procedure showPackageN;
	------------------------------------------------------------------------
	-- -- B
	impure function c_int return integer; attribute foreign of c_int : function is "VHPIDIRECT getInt";
	-- -- END
end package pkg;
package body pkg is
	-- -- A
	-- procedure showPackageN is begin report integer'image(N); end;
	---------------------------------------------------------
	-- -- B
	impure function c_int return integer is begin assert false report "c_int VHPI" severity failure; end c_int;
	-- -- END
end package body pkg;
```
**aux.c**
```c :file: aux.c
int getInt(){return 141;}
```
**run.sh**
```sh :image: ghdl/ghdl:buster-llvm
#!/usr/bin/env sh
ghdl -a --std=08 pkg.vhd tb.vhd
ghdl -e -Wl,aux.c --std=08 tb
./tb
```

**Context**
Please, provide the following information:

- OS: WSL: Ubuntu 18.04
- Origin:
  - [x] Built from sources: `f9878709`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-274-g93d01c6d) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/d/Development/Code/HDL/testLogVecArr/
Command line:
/etc/ghdl-llvm-PIC/bin/ghdl1-llvm --std=08 -P/etc/ghdl-llvm-PIC/lib/ghdl/ieee/v08/ -P/etc/ghdl-llvm-PIC/lib/ghdl/ -c -fpic -o tb.o tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-sem_inst.adb:271
Call stack traceback locations:
0x4fe505 0x4ff499 0x4fe480 0x4ff499 0x4fe37a 0x500554 0x579d59 0x56b200 0x56b389 0x53c62f 0x573f4f 0x57ab43 0x532ea4 0x64d4d5 0x64f108 0x650a01 0x4086c9 0x7f45d22a1b95 0x4079b8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
I don't think I can strip too much more away and still have the issue.

umarcor, 2020-04-15:
Does the bug persist if you define `package tbpkg is new work.pkg generic map (N => 3);` OUTSIDE of the architecture (make a global instantiation)? I believe that this bug is related to GHDL's partial support of some VHDL 2008 features, such as being able to use package instantiations with limited scope. The same applies to trying to pass type generics to entities, JFYTK.

RocketRoss, 2020-04-16:
Identical bug print out.

Changed `tb.vhd` to:

```
package tbpkg is new work.pkg generic map (N => 3);
use tbpkg.all;

entity tb is
end entity tb;

architecture RTL of tb is
	-- -- A
	---------------------------------------------
	-- -- B
	-- use work.pkg.all;
	-- -- END
begin
	process
		-- -- A
		-----------------------------------------
		-- -- B
		variable int : integer := c_int;
		-- -- END
	begin	
		-- -- A	
		showPackageN;
		----------------------------------------------
		-- -- B
		report "C Int: " & integer'image(int);
		-- -- END
		wait;
	end process;
end architecture RTL; 
```

closed 2020-04-16 by tgingold

RocketRoss, 2020-04-16:
@tgingold Beautiful, thanks!

commit 286abe8f (2020-04-16): vhdl: handling attribute specification in instantiations.  Fix #1229