ghdl/ghdl #1230: raised TYPES.INTERNAL_ERROR : synth-environment.adb:1373

rodrigomelo9, 2020-04-16:
Hi. It is a difficult bug to explain (at least for me). Here you have the files, ready to reproduce the problem: [issueOdd.zip](https://github.com/ghdl/ghdl/files/4487190/issueOdd.zip)

* Running `bash runOK.sh` it works.
* Running `bash runERROR.sh` it fails, with the following message:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-311-g430ca119) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/wrk/
Command line:
ghdl --synth OddSymTranspConvFIR
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-environment.adb:1373
Call stack traceback locations:
0x562aca71fd86 0x562aca7207db 0x562aca720a50 0x562aca720a95 0x562aca72dfd4 0x562aca72e975 0x562aca739033 0x562aca72f0b9 0x562aca739016 0x562aca72f0b9 0x562aca739016 0x562aca739934 0x562aca73b2ed 0x562aca73b4a7 0x562aca72b170 0x562aca72b2c2 0x562aca724394 0x562aca78abe7 0x562aca6ba71f 0x562aca8357a5 0x562aca48a9d1 0x7f09a8068099 0x562aca4891a8 0xfffffffffffffffe
******************************************************************
```

rodrigomelo9, 2020-04-16:
Let me say that I know how to avoid the problem (basically, providing the correct DelayLine component). The thing is, that I accidentally found this internal error and I am reporting it as requested by the message.

Xiretza, 2020-04-16:
The problematic line is `DelayLine.vhd:60`, which uses out-of-bounds indices for `Depth := 1`.

tgingold, 2020-04-16:
Now fixed.  This is a null assignment as hinted by @Xiretza.  It is now ignored.

commit a60a736b (2020-04-16): synth-stmts: handle null assignments.  Fix #1230
closed 2020-04-16 by tgingold