ghdl/ghdl #1232: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:860

vijaykumarvg1, 2020-04-16:
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-311-g430ca119) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/src/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:860
******************************************************************
ERROR: vhdl import failed.

Does anyone have an idea about this error?

eine, 2020-04-16:
It seems that all the relevant context has been removed. Please, provide the full bug report and, preferredly, a (minimal) reproducer.

tgingold, 2020-04-21:
I close this issue, assuming it has been fixed by another one.  If not, please reopen with an mwe.

closed 2020-04-21 by tgingold