ghdl/ghdl #1234: synth_expression_with_type: cannot handle IIR_KIND_PARENTHESIS_EXPRESSION

vijaykumarvg1, 2020-04-16:
**Description**
As GHDL doesnt support OTHERS => '0', instead we use 'range => '0'.
Here is my code: 
```vhdl
WINDOW<=((others=> (others=> (others=>'0'))));

    ADD_DEPTH_1<=((others=> (others=> (others=>'0'))));Enable_STAGE_1<='0';
    ADD_DEPTH_2<=((others=> (others=> (others=>'0'))));Enable_STAGE_2<='0';

    MULT_1<=((others=> (others=> (others=>'0'))));    DOUT_BUF_1_1<=(others => '0');BIAS_1<=(others => '0');ReLU_1<=(others => '0');
    MULT_2<=((others=> (others=> (others=>'0'))));    DOUT_BUF_2_1<=(others => '0');BIAS_2<=(others => '0');ReLU_2<=(others => '0');
    MULT_3<=((others=> (others=> (others=>'0'))));    DOUT_BUF_3_1<=(others => '0');BIAS_3<=(others => '0');ReLU_3<=(others => '0');
```

And I changed to:
```vhdl
   WINDOW <=((WINDOW 'range=> (WINDOW 'range=> (WINDOW 'range=>'0'))));
    

    ADD_DEPTH_1<=((ADD_DEPTH_1 'range=> (ADD_DEPTH_1 ' range=> (ADD_DEPTH_1 'range=>'0'))));Enable_STAGE_1<='0';
    ADD_DEPTH_2<=((ADD_DEPTH_2 'range=> (ADD_DEPTH_2 'range=> (ADD_DEPTH_2 'range=>'0'))));Enable_STAGE_2<='0';

    MULT_1<=((MULT_1'range=> (MULT_1 'range=> (MULT_1 'range=>'0'))));    DOUT_BUF_1_1<=(DOUT_BUF_1_1 'range => '0');BIAS_1<=(BIAS_1 'range => '0');ReLU_1<=(ReLU_1 'range => '0');
    MULT_2<=((MULT_2 'range=> (MULT_2 'range=> (MULT_2 'range=>'0'))));    DOUT_BUF_2_1<=(DOUT_BUF_2_1 'range => '0');BIAS_2<=(BIAS_2 'range => '0');ReLU_2<=(ReLU_2 'range => '0');
    MULT_3<=((MULT_3 'range=> (MULT_3 'range=> (MULT_3 'range=>'0'))));    DOUT_BUF_3_1<=(DOUT_BUF_3_1 'range => '0');BIAS_3<=(BIAS_3 'range => '0');ReLU_3<=(ReLU_3 'range => '0');
```
By doing so I get the following error:
synth_expression_with_type: cannot handle IIR_KIND_PARENTHESIS_EXPRESSION (LAYER_1.vhd:220:13)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-311-g430ca119) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/src/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
**************************

eine, 2020-04-16:
VHDL does support `(others=>'0')` (https://github.com/ghdl/ghdl/search?l=VHDL&q=others), however support of unconstrained arrays is not complete yet.

Please, provide a Minimal Working Example. The one provided is currently lacking the declaration of the signals that are being assigned, as well as the entity, architecture, testbench and command line arguments.

anhangaba, 2020-04-16:
Maybe related to #1182?

vijaykumarvg1, 2020-04-16:
```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.std_logic_arith.all;

entity CONV_LAYER_1 is

GENERIC
 	( 
	constant PERCISION      : positive := 5; 	
	constant DOUT_WIDTH     : positive := 5; 	
	constant BIAS_SIZE      : positive := 5;	
	constant MULT_SIZE      : positive := 13;	
	constant MULT_SUM_SIZE  : positive := 6;	
	constant DIN_WIDTH      : positive := 8;	
	constant BASE_DIN_WIDTH : positive := 8;	
	constant IMAGE_WIDTH    : positive := 13;	
	constant IMAGE_SIZE     : positive := 169;	
	constant F_SIZE         : positive := 2;	
	constant WEIGHT_SIZE    : positive := 5;	
	constant BIASES_SIZE	: positive := 2;
	constant PADDING        : positive := 1;	
	constant STRIDE         : positive := 1;	
	constant FEATURE_MAPS   : positive := 3;	
	constant VALID_CYCLES   : positive := 144;	
	constant STRIDE_CYCLES  : positive := 12;	
	constant VALID_LOCAL_PIX: positive := 12;	
	constant ADD_TREE_DEPTH : positive := 2;	
	constant INPUT_DEPTH    : positive := 1;
	constant FIFO_DEPTH     : positive := 12;	
	constant USED_FIFOS     : positive := 1;	
	constant ADD_1          : positive := 2;    	
	constant ADD_2          : positive := 1;    	
	constant LOCAL_OUTPUT   : positive := 5	
		); 

port(
	DIN                 :IN std_logic_vector(DIN_WIDTH-1 downto 0);
	CLK,RST             :IN std_logic;
	EN_STREAM           :IN std_logic; 					-- S_AXIS_TREADY  : Ready to accept data in 
	EN_STREAM_OUT_1     :OUT std_logic; 				-- M_AXIS_TREADY  : Connected slave device is ready to accept data out/ Internal Enable
	VALID_OUT_1         :OUT std_logic;  				-- M_AXIS_TVALID  : Data out is valid
	EN_LOC_STREAM_1     :IN std_logic;
	DOUT_1_1            :OUT std_logic_vector(DOUT_WIDTH-1 downto 0);
	DOUT_2_1            :OUT std_logic_vector(DOUT_WIDTH-1 downto 0);
	INTERNAL_RST        :OUT std_logic
	);	

end CONV_LAYER_1;

------------------------------ ARCHITECTURE DECLARATION - START---------------------------------------------

architecture Behavioral of CONV_LAYER_1 is

------------------------------ INTERNAL FIXED CONSTANT & SIGNALS DECLARATION - START---------------------------------------------
type       FILTER_TYPE             is array (0 to F_SIZE-1, 0 to F_SIZE-1) of signed(WEIGHT_SIZE- 1 downto 0);
type       FIFO_Memory             is array (0 to FIFO_DEPTH - 1)          of STD_LOGIC_VECTOR(DIN_WIDTH - 1 downto 0);
type       SLIDING_WINDOW          is array (0 to F_SIZE-1, 0 to F_SIZE-1) of STD_LOGIC_VECTOR(DIN_WIDTH- 1 downto 0);
signal     VALID_NXTLYR_PIX        :integer range 0 to STRIDE_CYCLES;
signal     PIXEL_COUNT             :integer range 0 to VALID_CYCLES;
signal     OUT_PIXEL_COUNT         :integer range 0 to VALID_CYCLES;
signal     EN_NXT_LYR_1            :std_logic;
signal     FRST_TIM_EN_1           :std_logic;
signal     Enable_MULT             :std_logic;
signal     Enable_ADDER            :std_logic;
signal     Enable_ReLU             :std_logic;
signal     Enable_BIAS             :std_logic;
signal     SIG_STRIDE              :integer range 0 to IMAGE_SIZE;
signal     PADDING_count           :integer range 0 to IMAGE_SIZE; -- TEMPORARY
signal     ROW_COUNT               :integer range 0 to IMAGE_SIZE; -- TEMPORARY
[... 264 lines cut ...]

		end if; -- VALIDPIXELS

		if VALID_NXTLYR_PIX=((IMAGE_WIDTH*STRIDE)-1) then VALID_NXTLYR_PIX<=0;SIG_STRIDE<=STRIDE;   -- reset sride and valid pixels
		else VALID_NXTLYR_PIX<=VALID_NXTLYR_PIX+1;end if; 

	end if;  --ReLU

elsif OUT_PIXEL_COUNT>=VALID_CYCLES  then INTERNAL_RST<='1';SIG_STRIDE<=STRIDE;EN_NXT_LYR_1<='1';  -- order is very important
else  EN_NXT_LYR_1<='0';-- In case stream stopped

end if; -- end enable 
end if; -- for RST	
end if; -- rising edge
end process LAYER_1;

EN_STREAM_OUT_1<= EN_STREAM_OUT_2;
VALID_OUT_1<= VALID_OUT_2;
DOUT_1_1<=DOUT_1_2;
DOUT_2_1<=DOUT_2_2;

end Behavioral;
------------------------------ ARCHITECTURE DECLARATION - END---------------------------------------------
```
CL: ghdl -fsynopsys CONV_LAYER_1.vhd

eine, 2020-04-16:
This *complete* and *reproducible* example works for me:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
  generic (
    constant PERCISION      : positive := 5; --sic
    constant FEATURE_MAPS   : positive := 3;
    constant ADD_1          : positive := 2
);
end entity;

architecture arch of ent is
  type    AADD_DEPTH_1	is array (0 to ADD_1-1, 0 to FEATURE_MAPS-1) of signed(PERCISION-1 downto 0);
  signal  ADD_DEPTH_1: AADD_DEPTH_1;
begin
  ADD_DEPTH_1 <= (
    others => (
      others => (
        others => '0'
      )
    )
  );
end;
```

```sh
ghdl -a -fsynopsys ent.vhd
ghdl -e ent
./ent
```

anhangaba, 2020-04-16:
The _reproducible_ example worked for me with --synth. But the _complete_ didn't. I'm using v0.37.0-337-g8346c712.

This seems to be the same issue as #1182. 

If you move the constant declarations from the generic interface list to the architecture declarative part the code synthesis.

anhangaba, 2020-04-16:
Minimal example of his issue:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity conv_layer_1 is
    end conv_layer_1;

architecture behavioral of conv_layer_1 is
    constant x : positive := 2; 
    constant y : positive := 8; 

    type sliding_window is array (0 to x-1, 0 to x-1)
                    of std_logic_vector(y- 1 downto 0);
    signal window : sliding_window; 

begin
    window <= ((others=> (others=> (others=>'0'))));
end architecture;
```
```
ghdl -a --std=08 conv_layer.vhdl
```
```
conv_layer.vhdl:16:26:error: 'others' choice not allowed for an aggregate in this context
```
GHDL release: 1.0-dev (v0.37.0-337-g8346c712) [Dunoon edition]

eine, 2020-04-16:
> There is one more parenthesis than necessary.

Yes. I should have specified that this was the main difference in my example [above](https://github.com/ghdl/ghdl/issues/1234#issuecomment-614926746).

anhangaba, 2020-04-16:
I managed to get the `IIR_KIND_PARENTHESIS_EXPRESSION`.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    port (foobar : out std_logic_vector (3 downto 0));
end issue;

architecture beh of issue is
begin
    foobar <= ((foobar'range=>'0'));
end architecture;
```
With the unnecessary parenthesis GHDL crashes (only with ``--synth``).
```
synth_expression_with_type: cannot handle IIR_KIND_PARENTHESIS_EXPRESSION (issue.vhdl:10:15)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-337-g8346c712) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl --synth issue.vhdl -e issue
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x556b6d4cd616 0x556b6d5f8706 0x556b6d5ccb92 0x556b6d5cf9fb 0x556b6d5dd006 0x556b6d5dd27f 0x556b6d5cc149 0x556b6d5cc2bd 0x556b6d5c4c6c 0x556b6d630b86 0x556b6d556761 0x556b6d6e3e3d 0x556b6d314dfc 0x7fd344b80151 0x556b6d31359c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-04-17:
So I have just fixed the issue related to the title.  There might be other issues, but they will be handled in a different github number.

Thank you for the issue.

commit 465b041e (2020-04-17): synth-expr: handle parenthesis_expression.  Fix #1234
closed 2020-04-17 by tgingold