ghdl/ghdl #1235: synth: crash on process with multiple clock edge conditions

Xiretza, 2020-04-16:
**Description**
The code below illustrates a simple valid/got interface for a component with a global clock enable. When the `valid` signal is asserted, the `got` signal is asserted on the next enabled clock cycle, then deasserted *immediately* on the next clock cycle, enabled or not. Using separate if statements avoids having to wrap all the other (clock-enabled) logic in another layer of indentation.

**Expected behaviour**
The design synthesizes and behaves the same as in simulation. An alternative way of writing it would be:

```vhdl
process(clk)
begin
    if rising_edge(clk) then
        got <= '0';

        if ce = '1' then
            if valid then
                got <= '1';
            end if;
        end if;
    end if;
end process;
```

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        clk, ce, valid : in std_logic;
        got : out std_logic
    );
end ent;

architecture a of ent is
begin
    process(clk)
    begin
        if rising_edge(clk) then
            got <= '0';
        end if;

        if rising_edge(clk) and ce = '1' then
            if valid then
                got <= '1';
            end if;
        end if;
    end process;
end a;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 90d7bfe9

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-320-g90d7bfe9) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
ghdl --synth --std=08 ent.vhd -e ent
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:529
Call stack traceback locations:
0x584b62 0x435582 0x492622 0x49287b 0x520518 0x52fa49 0x52f668 0x527e1b 0x52372a 0x56a143 0x4dce29 0x406e51 0x7fcf57380021 0x406ecc 0xfffffffffffffffe
******************************************************************
```

Xiretza, 2020-04-16:
Might also be relevant to #1069.

tmeissner, 2020-04-16:
For me, the reproducer is a coding style, we would never allow this code in our company ;)

Especially this:

```vhdl
if rising_edge(clk) and ce = '1' then
            if valid then
                got <= '1';
            end if;
        end if;
```

In synthesis code I would never use detection of clock edges and the enable in the same if construct. I think some years ago such code couldn't handled by synthesis tools either. Maybe this has changed, but another question with this is: Do you have to put ce in the sensitivity list?

Xiretza, 2020-04-16:
> Do you have to put ce in the sensitivity list?

Interesting question, and I don't have an answer to it (my guess is no, though, because it should be the same as two nested if statements due to short-circuiting). Maybe someone with a better undersanding of the LRM can chime in.

tgingold, 2020-04-17:
To be honest, I am not sure I want to support this style (two 'if' statements with the same edge).

I suppose this is a low priority issue.  I could however improve the error message!

Xiretza, 2020-04-17:
That's fine, I was just playing around with the design a bit and since it works completely fine in simulation I thought I'd open an issue. It's easy to work around, obviously.

dalex78, 2026-08-19:
# Issue #1235 — Crash: process with multiple `rising_edge(clk)` conditions

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

A process with two separate `if rising_edge(clk) then ... end if;` statements (a default-assignment block followed by a second, conditional one), instead of a single enclosing `if rising_edge(clk)` with nested logic, crashed `ghdl --synth` with an internal `ASSERT_FAILURE` at `netlists.adb:529`.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth --std=08 ent.vhd -e ent` now produces a correct netlist — the two clocked `if` statements are correctly merged into a single clocked process with a mux computing the combined default/conditional update — instead of crashing. This must have been fixed as part of general improvements to multi-clocked-statement merging in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1235/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold