ghdl/ghdl #1236: IIR_KIND_INTERFACE_TYPE_DEFINITION

anhangaba, 2020-04-17:
Trying to write a minimal example with comprehensive coverage for another issue, I found `--synth` doesn't support generic types.

```vhd :file: issue2.vhd
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    generic (type t_type);
end issue;

architecture beh of issue is
begin
end architecture beh;
```
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-337-g8346c712) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl --synth --std=08 issue2.vhdl -e issue
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Default_Value
Call stack traceback locations:
0x5562d29624e2 0x5562d27010d7 0x5562d286d591 0x5562d28941c6 0x5562d2894b14 0x5562d27ba761 0x5562d2947e3d 0x5562d2578dfc 0x7f1f7cf0f151 0x5562d257759c 0xfffffffffffffffe
******************************************************************
```
---
Another related issue:

```vhd :file: issue1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    generic (type t_type);
    port (foo : out t_type(3 downto 0));
end issue;

architecture beh of issue is
begin
    foo <= (others=>'0');
end architecture beh;
```
```
sem_subtype_constraint: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION (issue1.vhdl:6:19)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-337-g8346c712) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl --synth --std=08 issue1.vhdl -e issue
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x564e30a38616 0x564e30ae8650 0x564e30ae8764 0x564e30aff82b 0x564e30b00a30 0x564e30b0de4a 0x564e30b153a6 0x564e30aca111 0x564e30acb034 0x564e30b73ddb 0x564e30b9b175 0x564e30b9bb14 0x564e30ac1761 0x564e30c4ee3d 0x564e3087fdfc 0x7f97a7115151 0x564e3087e59c 0xfffffffffffffffe
******************************************************************
```

It also doesn't work when instantiated in another entity.

I reckon this is not a critical issue. I'm not sure if many people use this feature.

eine, 2020-04-17:
See #726.

tgingold, 2020-04-17:
That's an interesting case, because it is now easier to handle in synthesis than in simulation.

But lower priority...

eine, 2020-04-17:
> it is now easier to handle in synthesis than in simulation.

I'm curious about why is this...

tgingold, 2020-04-17:
In synthesis, a sub-module is synthesized for each set of generics.
In simulation, entities and architectures are translated only once.

That worked well (except for speed) until vhdl-2008 came.  Because it is inefficient (and more difficult) to generate code for any type.

tgingold, 2020-04-17:
Ok, I will cheat.  This design will be rejected because you cannot have such a type generic at the top-level.

Feel free to open a new issue for use of a type generic!

Thanks.

commit 0c48fe92 (2020-04-17): vhdl: reject top-level entities with a type generic.  Fix #1236
closed 2020-04-17 by tgingold