ghdl/ghdl #1237: if-generate blocks with false conditions break analysis

kammoh, 2020-04-17:
**Description**
I have blocks of code inside if generate statements, each block corresponding to a specific value of a constant parameter `P` (e.g. set in the package file or from a generic).

I'm expecting the content of the if-generate blocks with (constant) false condition not to be analyzed or elaborated at all. This seems to be the behaviour or other tools as well.

**Expected behaviour**
`if-generate` blocks with false conditions should not to be analyzed. Same code is synthesized and simulated correctly (no warnings/errors, expected behaviour) on Vivado and Quartus.

**How to reproduce?**

```vhd :file: mwe.vhd
library ieee;
use ieee.std_logic_1164.all;

entity MWE is
end MWE;

architecture test of MWE is
  constant P       : integer := 1;
  signal   my_sig  : std_logic_vector(P downto 0);
begin
  block2: if P = 2 generate
    my_sig(2) <= '1';
  end generate;

  block1: if P = 1 generate
    my_sig(1) <= '1';
  end generate;

  -- even this block alone breaks during analysis
  blockf: if false generate
    my_sig(2) <= '1';
  end generate;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a mwe.vhd
```
results in:
```
mwe.vhd:13:12:error: static expression violates bounds
mwe.vhd:21:12:error: static expression violates bounds
ghdl:error: compilation error
```

**Context**
Please, provide the following information:

- OS: macOS/Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `v0.37.0-332-gb4529588`

kammoh, 2020-04-17:
I believe this is as well relevant to VHDL-1993.

Questa 2020.1, Riviera Pro 2017.02, and VCS 2019.06 also have no problem with the code and elaborates conditionally as expected (in either '93 or 2008 mode):

https://www.edaplayground.com/x/2d_u

GHDL 0.35 and a couple of other tools fail though.

Section 11.8 of 2008 LRM says:
"A generate statement provides a mechanism for iterative or **conditional elaboration** of a portion of a description." (bold emphasis from me)

In section 14.5.3 (more specifically for the 2008 `elsif-generate`) it goes on to say:
"For an if generate statement, elaboration consists of the evaluation, in succession, of the condition specified after if and any conditions specified after elsif (treating a final else as elsif TRUE generate) until one evaluates to TRUE or all conditions are evaluated and yield FALSE. If one condition evaluates to TRUE, then **exactly one block statement is generated**; otherwise, *no block statement is generated*. (bold/emphasis from me)

diogratia, 2020-04-17:
Your 11.8 quoted text addresses elaboration while the error shown occurred during analysis.

`my_sig` has a locally static index constraint and the two errors are locally static indexed names (see 9.4.2 Locally static expressions paragraph 1 item n) which can be evaluated during analysis (see 9.4.1 paragraph 2 "Certain forms of expression can be evaluated during the analysis of the design unit in which they appear; such an expression is said to be *locally static*.") and in doing so generates the two errors.

(And I managed to restrain my self from emboldening **can be** in the quoted sentences from 9.4.1 above.)

See Annex D (Informative) Potentially nonportable constructs

>A description is considered portable if it  
>>a) Compiles, elaborates, initializes, and simulates to termination of the simulation cycle on all conformant implementations, and  
>>b) The time-variant state of all signals and variables in the description are the same at all times during the simulation,  
 >  
>under the condition that the same stimuli are applied at the same times to the description. The stimuli applied to a model include the values supplied to generics and ports at the root of the design hierarchy of the model, if any.

This perceived problem has arisen here in a previously reported issue. ghdl is a conformant implementation for purposes of Annex D a) above for this construct and is allowed to evaluate locally static expressions at analysis (compile) time. 

Here 8.4 Indexed names requires the errors when the indexed names are evaluated. ("For the evaluation of an indexed name, the prefix and the expressions are evaluated. It is an error if an index value does not belong to the range of the corresponding index range of the array.")

The construct is nonportable. A conforming implementation **can** evaluate locally static index names during analysis. The indexed names produced errors.

A portable construct would not rely on an error involving the evaluation of a locally static name being caught during simulation when the standard gives explicit permission to do so during analysis.

There **can be** execution time advantages to evaluating locally static indexed names during analysis instead of during simulation.

eine, 2020-04-17:
> I believe this is as well relevant to VHDL-1993.

You are correct. I was sleepy and my brain did a short circuit to #106.

diogratia, 2020-04-17:
J'accuse. (And thanks for pointing to issue #106.)

kammoh, 2020-04-17:
I don't think this should be *all* about whether this issue makes ghdl legally "compliant" and approaching the issue like a lawyer :) 
IMO, there are at least two things to consider:
1. Conditional generate is very common, this kind of conditional block generation can be necessary in many different situations. AFAIK SystemVerilog `generate-if` works fine in this particular situation as well. This is the behaviour that most HDL designer expect.
At the same time, parameterized generation is one of the reported shortcomings of traditional HDLs, so I don't see the merit in resorting to hermeneutic/philosophical/legal arguments to justify the current behavior of ghdl.
2. This makes the behavior of ghdl very different from all other simulation/synthesis tools out there and against the status-quo. This will only prevent people like me from adapting `ghdl` for their work/education/research.

tmeissner, 2020-04-17:
Non-LRM features should only be added under relaxedrules option. I and many others rely on GHDL as one of the few standard-complying tools.

Xiretza, 2020-04-17:
ghdl is traditionally very strict with its standards compliance, which is good - this might be a possible "feature" for `-frelaxed-rules` though.

eine, 2020-04-17:
> I don't think this should be _all_ about whether this issue makes ghdl legally "compliant" and approaching the issue like a lawyer :)

As a rule of thumb, GHDL is pedantic with the LRM, like a lawyer. Exceptions can be made, and are made, specially to make it compatible with vendor tools. However, that falls in the "relaxed" option. See https://ghdl.readthedocs.io/en/latest/using/ImplementationOfVHDL.html and https://ghdl.readthedocs.io/en/latest/using/ImplementationOfVHDL.html#using-vendor-libraries.

kammoh, 2020-04-17:
From the argument provided by @diogratia, can `ghdl` can still be still %100 LRM compliant while letting this code go through analysis?

tgingold, 2020-04-18:
This is now fixed.  A warning is generated at analysis time and the error is generated at runtime.
That's a trade-off.

closed 2020-04-18 by tgingold