ghdl/ghdl #1238: "multiple assignments" complaint when synthesizing a MUX with tristate buffers

rodrigomelo9, 2020-04-17:
First of all, the previously reported issues are now working :-D

Again, it is an example from Xilinx (an ISE example). It is synthesized with ISE and Vivado ([here the logs](https://github.com/ghdl/ghdl/files/4492830/issueMux.zip)).

I guess it is related to unsupported tri-state buffers? Anyway, I am reporting it FYI.

```
library ieee;
use ieee.std_logic_1164.all;

entity multiplexers_3 is

    port (di  : in  std_logic_vector(7 downto 0);
          sel : in  std_logic_vector(7 downto 0);
          do  : out std_logic);
	  
end multiplexers_3;

architecture archi of multiplexers_3 is
begin
    do <= di(0) when sel(0)='0' else 'Z';
    do <= di(1) when sel(1)='0' else 'Z';
    do <= di(2) when sel(2)='0' else 'Z';
    do <= di(3) when sel(3)='0' else 'Z';
    do <= di(4) when sel(4)='0' else 'Z';
    do <= di(5) when sel(5)='0' else 'Z';
    do <= di(6) when sel(6)='0' else 'Z';
    do <= di(7) when sel(7)='0' else 'Z';
end archi;
```

```
$ docker run --rm -t -v $(pwd):/src -w /src ghdl/synth:beta ghdl --synth examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd -e
ghdl:note: top entity is "multiplexers_3"
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:0
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:1
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:2
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:3
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:4
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:5
examples/ise/HDL_Coding_Techniques/multiplexers/multiplexers_3.vhd:14:11:error: multiple assignments for offsets 0:6
```

eine, 2020-04-17:
Is it possible to use tri-state buffers inside Xilinx's FPGAs? I thought it was limited to I/O.

daveshah1, 2020-04-17:
Some of the very early Xilinx parts did have internal tristates (something like pre Virtex 2), these days they convert internal tristates in HDL into equivalent non-tristate logic (i.e. muxes).

commit c03407bc (2020-04-23): synth: insert resolver to handle multiple drivers to a signal.  Fix #1238
closed 2020-04-23 by tgingold