ghdl/ghdl #1239: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-aggr.adb:371

vijaykumarvg1, 2020-04-17:
**Description**
I don't see any errors or warning I fixed all, still, I am getting this error which I am unaware

**How to reproduce?**

```vhd :file: 
library ieee;
use ieee.std_logic_1164.all;
use IEEE.std_logic_arith.all;
-------------------------------------------------------------------------------------
--
--
-- Definition of Ports
-- ACLK           : Synchronous clock
-- ARESETN        : System reset, active low
-- S_AXIS_TREADY  : Ready to accept data in
-- S_AXIS_TDATA   : Data in 
-- S_AXIS_TLAST   : Optional data in qualifier
-- S_AXIS_TVALID  : Data in is valid
-- M_AXIS_TVALID  : Data out is valid
-- M_AXIS_TDATA   : Data Out
-- M_AXIS_TLAST   : Optional data out qualifier
-- M_AXIS_TREADY  : Connected slave device is ready to accept data out
--
-------------------------------------------------------------------------------

------------------------------------------------------------------------------
-- Entity Section 
------------------------------------------------------------------------------
entity accelerator is
GENERIC( 
  	constant DATA_WIDTH 		: positive := 8;
	constant IMAGE_WIDTH 		: positive := 13;
	constant IMAGE_SIZE 		: positive := 169;
	constant DOUT_WIDTH		: positive := 5 -- TO BE CALCULATED
		); 

	port(
		-- DO NOT EDIT BELOW THIS LINE ---------------------
		-- Bus protocol ports, do not add or delete. 
		ACLK            : in	std_logic;
		ARESETN         : in	std_logic;
		S_AXIS_TREADY	: out	std_logic;
		S_AXIS_TDATA	: in	std_logic_vector(31 downto 0);
		S_AXIS_TLAST	: in	std_logic;
		S_AXIS_TVALID	: in	std_logic;
		M_AXIS_TVALID	: out	std_logic;
		M_AXIS_TDATA	: out	std_logic_vector(15 downto 0);
		--M_AXIS_TLAST	: out	std_logic;
		M_AXIS_TREADY	: in	std_logic;
		EN_LOC_STREAM_1: in std_logic
		-- DO NOT EDIT ABOVE THIS LINE ---------------------
	);

end accelerator;

------------------------------------------------------------------------------
-- Architecture Section
------------------------------------------------------------------------------
architecture Behavior of accelerator is
signal DOUT_1_1          : std_logic_vector(DOUT_WIDTH-1 downto 0);
signal DOUT_2_1          : std_logic_vector(DOUT_WIDTH-1 downto 0);
signal EN_STREAM_OUT_1	 : std_logic;
signal VALID_OUT_1       : std_logic;
signal INTERNAL_RST : std_logic;
---------------------------------- MAP NEXT LAYER - COMPONENTS START----------------------------------
COMPONENT CONV_LAYER_1
   port(
	DIN                 :IN std_logic_vector(DATA_WIDTH-1 downto 0);
	CLK,RST             :IN std_logic;
	EN_STREAM           :IN std_logic; 				-- S_AXIS_TREADY  : Ready to accept data in 
	EN_STREAM_OUT_1     :OUT std_logic; 				-- M_AXIS_TREADY  : Connected slave device is ready to accept data out/ Internal Enable
	VALID_OUT_1         :OUT std_logic;  				-- M_AXIS_TVALID  : Data out is valid
	EN_LOC_STREAM_1     :IN std_logic;
	DOUT_1_1            :OUT std_logic_vector(DOUT_WIDTH-1 downto 0);
	DOUT_2_1            :OUT std_logic_vector(DOUT_WIDTH-1 downto 0);
	INTERNAL_RST        :OUT std_logic
      			);
END COMPONENT CONV_LAYER_1;

begin

CONV_LYR_1 : CONV_LAYER_1 
          port map(
           CLK               => ACLK,
           RST               => ARESETN,
           DIN               => S_AXIS_TDATA(7 downto 0),
           EN_STREAM         => M_AXIS_TREADY,		
           DOUT_1_1               => DOUT_1_1,
           DOUT_2_1               => DOUT_2_1,
           EN_LOC_STREAM_1 => EN_LOC_STREAM_1,
           EN_STREAM_OUT_1        => EN_STREAM_OUT_1,
		   VALID_OUT_1            => VALID_OUT_1,
		   INTERNAL_RST => INTERNAL_RST
          );

 M_AXIS_TDATA(4 downto 0) <= DOUT_1_1;
 M_AXIS_TDATA(9 downto 5)<= DOUT_2_1;
 M_AXIS_TDATA (15 downto 10) <= (others => '0');
 S_AXIS_TREADY<= EN_STREAM_OUT_1;
 M_AXIS_TVALID<= VALID_OUT_1;
 
 
end architecture Behavior;
------------------------------ ARCHITECTURE DECLARATION - END---------------------------------------------

```

```sh :image: ghdl/ghdl:buster-mcode
-- Running command `ghdl -fsynopsys accelerator.vhd LAYER_1.vhd LAYER_2.vhd LAYER_3.vhd LAYER_4.vhd LAYER_5.vhd LAYER_6.vhd -e accelerator; synth -lut 4 ; write_blif accelerator.blif' --

```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-311-g430ca119) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/src/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-aggr.adb:371
******************************************************************
ERROR: vhdl import failed.

...
******************************************************************
```

tgingold, 2020-04-18:
I am not able to reproduce the same crash.  I will fix the one I get.

On the command line, you provided several files, but only one for the reproducer.

vijaykumarvg1, 2020-04-18:
[https://drive.google.com/drive/folders/1tOuV0m1zBrDGqqEo6YLBFixcR3c_spCM?usp=sharing](url)

Please access the files via above link. Thanks!

tgingold, 2020-04-18:
Thanks, I can reproduce the crash.

vijaykumarvg1, 2020-04-18:
For synthesis: 

Command line:
ghdl -a accelerator.vhd LAYER_1.vhd LAYER_2.vhd LAYER_3.vhd LAYER_4.vhd LAYER_5.vhd LAYER_6.vhd

ghdl -e -fsynopsys accelerator

ghdl --synth -fsynopsys accelerator

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-356-g628c01a3) [Dunoon edition]
Compiled with GNAT Version: 8.4.0
Target: x86_64-linux-gnu
/home/v/Desktop/tlut_dnn/tlut_dnn/VHDL_DNN_1/test/
Command line:
ghdl --synth -fsynopsys accelerator
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-aggr.adb:371
Call stack traceback locations:
0x7faea7fd8701 0x564f0f0723c2 0x564f0f072b30 0x564f0f06ac38 0x564f0f070091 0x564f0f070d8f 0x564f0f0702b3 0x564f0f070d8f 0x564f0f072083 0x564f0f072b30 0x564f0f06ac38 0x564f0f040e66 0x564f0f043c48 0x564f0f04e32f 0x564f0f0443b5 0x564f0f04e312 0x564f0f0443b5 0x564f0f04e312 0x564f0f04ec30 0x564f0f050933 0x564f0f050aed 0x564f0f04043e 0x564f0f040590 0x564f0f039662 0x564f0f0a069f 0x564f0efcf099 0x564f0f14b29b 0x564f0ed9b9e1 0x7faea759eb95 0x564f0ed9a1b8 0xfffffffffffffffe

tgingold, 2020-04-19:
There are many incorrect aggregates.  I will soon push a fix to improve diagnostics.

commit 7a2c53cd (2020-04-19): synth-aggr: check bound errors.  Fix #1239
closed 2020-04-19 by tgingold