ghdl/ghdl #1241: synth: Incorrect handling of indexed std_logic_vector in indexed array of records

daveshah1, 2020-04-17:
**Description**
This has been a slightly tricky one to isolate and minimise, so hopefully this is all correct.

It seems like the below code is being missynthesised, visualising the Yosys netlist and also by inspection of the ghdl --synth VHDL output, there is a path from constant "1" to output which isn't correct.

[image]

**Expected behaviour**
Synthesises correctly - should be selecting one bit of `data` and ignoring the constant 1s initialising `x` which is never accessed.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: index.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity top is
	port(
		sel : in unsigned(1 downto 0);
		data : in std_logic_vector(3 downto 0);
		q : out std_logic
	);
end entity;

architecture arch of top is
	type record_t is record
		x : std_logic_vector(1 downto 0);
		y : std_logic_vector(1 downto 0);
	end record;

	type array_t is array (0 to 1) of record_t;
	signal a : array_t;
begin
	a <= (("11", data(1 downto 0)), ("11", data(3 downto 2)));
	q <= a(to_integer(sel(1 downto 1))).y(to_integer(sel(0 downto 0)));
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth index.vhd -e top
```

**Additional context**
This was causing errors during SNES graphics scrolling, the code this was minimised from is here: https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/src/PPU.vhd#L1736

eine, 2020-04-17:
So, the `1'1` going into  `A` and `B` of `$6$mux` should be, respectively `0:0 - 0:0` and `1:1 - 0:0` coming from `data`?

What about `13 $sub`? Is that expected (to match the latency of `$5$mux` and `$6$mux`?

daveshah1, 2020-04-17:
> So, the 1'1 going into A and B of $6$mux should be, respectively 0:0 - 0:0 and 1:1 - 0:0 coming from data?

Yes

> What about 13 $sub? Is that expected (to match the latency of $5$mux and $6$mux)?

The subtraction is fine, it is because  "0 to 1" is considered "backwards" compared to downto indexing, note that the inputs to the $7 mux are also reversed accordingly.

eine, 2020-04-17:
Thanks. This is a very interesting/elegant example. Both very simple and quite complex at the same time.

tgingold, 2020-04-18:
Thank you for the issue.  SNES_NiSTer is a very interesting example!

commit c261402a (2020-04-18): netlists-expands: fix handling of non-consective extractions.  Fix #1241
closed 2020-04-18 by tgingold